lvgl/lvgl #6491: Images converted using lvgl's online image converter exhibit color differences.

makun666, 2024-07-09:
### LVGL version

v8.3.10

### What happened?

I need to convert a solid-color background image into a C array using LVGL's online converter. I've selected the CF_TRUE_COLOR_ALPHA color format, and I'm using a 16-bit color depth.

However, I have noticed a slight difference between the C array generated by the online converter and the C array I obtain by reading the same file directly using the png decoder.

Here are the details:
When I use the online converter, the converted value for each pixel in ARGB565 format is 0xE7 0x00 0xFF.
But when I read the file directly, the corresponding pixel value is 0xE6 0x00 0xFF.

For reference, the original ARGB888 value for that pixel in the C array is 0x36 0x1c 0x00 0xff. Converting this to ARGB565 results in the value 0xE6 0x00 0xFF.

I need to use the solid-color background image along with another image. When viewed using an image viewer software on Windows, there is no color difference between this image and the background.

Below is the solid-color background image.
[image]

However, after converting the background images using the online Image Converter, there is a significant color difference.
[image]

When the background image is stored in Flash and displayed using a PNG decoder, the color difference issue does not arise. Alternatively, manually modifying the c-array after conversion in Image Converter to 0xE6 00 also eliminates the color difference.

Therefore, I would like to inquire if there is a potential issue with the online Image Converter.

kisvegabor, 2024-07-10:
Is it an option for you to update the LVGL v9? It would be much easier for us to fix this issue in v9.

makun666, 2024-07-11:
Since our project is basically completed, it would be quite challenging to switch to using the lvgl v9 version.

Based on my understanding, there seems to be an issue with the online image converter, right?

Furthermore, I have tried selecting the v9 version on the online image converter, but the converted array is still problematic.

liamHowatt, 2024-07-16:
Could you please send the PNG file for us to try to reproduce the off-by-one color value?

makun666, 2024-07-17:
[bg.zip](https://github.com/user-attachments/files/16258061/bg.zip)
The one above is the solid color background image. 
You can try it out and will find that there are differences between the arrays obtained using the LVGL online image converter and directly converting a local file with a PNG decoder.

xinsuinizhuan, 2024-07-19:
so we need the C++ version tools or functions, convert image and font, not online to convert.

makun666, 2024-07-23:
> Could you please send the PNG file for us to try to reproduce the off-by-one color value?

Have you reproduced this problem using the pictures I sent?

liamHowatt, 2024-07-23:
Yes, I found some flawed logic while using a debugger and confirmed a fix. Both v8 and v9 of the online converter have the issue. I opened a PR for each. Thanks for reporting it.
https://github.com/lvgl/lv_img_conv/pull/40
https://github.com/lvgl/lv_img_conv/pull/41

liamHowatt, 2024-07-23:
I hope it doesn't ruin the dithering feature of the v8 converter.

kisvegabor, 2024-07-23:
Actually the original solution should produce better result. What if we fix in the v8's PNG decoder [here](https://github.com/lvgl/lvgl/blob/release/v8.4/src/extra/libs/png/lv_png.c#L249C13-L249C26)? I wouldn't update `lv_color_make` in v8 as it might produce unexpected issues on other places, but we can replace this function with a more accurate conversation.

liamHowatt, 2024-07-24:
My initial thought is that the tool is deviating from a standard way of converting between color spaces and it would be a big deal to change conversions everywhere to match the tool's unique conversion method. I can imagine an issue being opened where someone says there is a difference between the same image as PNG vs BMP, but upon investigation, the v8 BMP decoder only supports BMPs of the same color format so no conversion would happen there. Also, it seems to me that C array images will never have the color format converted at runtime.

Are there any other places in LVGL where color formats of images are converted using `lv_color_make` other than the PNG decoder? It seems like there isn't, so this change seems safe :+1:

liamHowatt, 2024-07-24:
I see GIF converts color formats. It feels less likely to me that someone would want to match a GIF to a background color, but they very well could. It's something to consider for the change too.

Palettes should also be considered.

kisvegabor, 2024-07-24:
The [JPEG decoder](https://github.com/lvgl/lvgl/blob/release/v8.4/src/extra/libs/sjpg/lv_sjpg.c#L759) can be also an issue.  However, as this the first time this issue came up, I think we are good with fixing the PNG decoder only.

liamHowatt, 2024-08-02:
16 bit is working in this PR: #6626. I'm facing some challenges with 8 bit. Will revisit it soon.

liamHowatt, 2024-08-19:
Not stale

kisvegabor, 2024-08-29:
Should be fixed by #6626

PR #6626 (2024-08-29): fix(png): use color rounding like img conv tool
### Description of the feature or fix

Fixes #6491

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig).
- Run `s [... cut]
closed 2024-08-29 by kisvegabor