lvgl/lvgl #6492: fs why not support create file or dir?

Antecer, 2024-07-09:
### Introduce the problem

I just see this code in lvgl:
```
/**
 * Open a file
 * @param drv       pointer to a driver where this function belongs
 * @param path      path to the file beginning with the driver letter (e.g. S:/folder/file.txt)
 * @param mode      read: FS_MODE_RD, write: FS_MODE_WR, both: FS_MODE_RD | FS_MODE_WR
 * @return          a file descriptor or NULL on error
 */
static void * fs_open(lv_fs_drv_t * drv, const char * path, lv_fs_mode_t mode)
{
    int flags = 0;
    if(mode == LV_FS_MODE_WR)
        flags = LFS_O_WRONLY;
    else if(mode == LV_FS_MODE_RD)
        flags = LFS_O_RDONLY;
    else if(mode == (LV_FS_MODE_WR | LV_FS_MODE_RD))
        flags = LFS_O_RDWR;

    LittleFile * lf = lv_malloc(sizeof(LittleFile));
    LV_ASSERT_MALLOC(lf);

    lfs_t * lfs = drv->user_data;
    int err = lfs_file_open(lfs, &lf->file, path, flags);
    if(err) {
        return NULL;
    }

    return (void *)lf;
}
```
here didn't support `LFS_O_CREAT`, because if it set to `mode`, the flags will set to 0.

If I need to create a file or path, then I need to call the original file system API, so I will switch back and forth between two sets of file system APIs. If so, why not use the original file system API directly instead of using the lvgl proxy API?

kisvegabor, 2024-07-10:
>  If so, why not use the original file system API directly instead of using the lvgl proxy API?

LVGL can be used in many environments and platform (PC, embedded, many RTOS, many FS libs) and internally LVGL needs to have an abstract way to to open and read images and other assets.  

There is no file/directory creation API because the FS API was intended to be used internally by LVGL where just read the assets. 

Did you use LVGL's FS in the application code or was it required for the UI itself?

Antecer, 2024-07-11:
maybe i want input some text or get some data, i need show it in lvgl, then confirmation, create the file and save it.

Could you tell me a little bit about where lv_fs is mainly used?
I would prefer to use the same file system interface throughout the program, rather than mixing multiple interfaces, which is prone to errors.

XuNeo, 2024-07-11:
With mode `LV_FS_MODE_WR `, it will write to file, or create if not exist. 

See

https://github.com/lvgl/lvgl/blob/730a3bd76f691c8490e170e7cdfd2d54142ddc0e/src/libs/fsdrv/lv_fs_posix.c#L111

kisvegabor, 2024-07-12:
It's [already in  LVGL](https://github.com/lvgl/lvgl/blob/730a3bd76f691c8490e170e7cdfd2d54142ddc0e/src/libs/fsdrv/lv_fs_posix.c#L111). You can use open a file with the `LV_FS_MODE_WR | LV_FS_MODE_RD` flags.

Antecer, 2024-07-13:
It is a bit field variable, so I think the following solution should be better?
```c
int flags = 0;
if(mode & LV_FS_MODE_RD) flags |= O_RDONLY;
if(mode & LV_FS_MODE_WR) flags |= O_WRONLY | O_CREAT;
```

kisvegabor, 2024-07-17:
They are defines like this for me:
```
#define O_RDONLY	     00
#define O_WRONLY	     01
#define O_RDWR		     02
```

So `O_RDONLY | O_WRONLY != O_RDWR`

Antecer, 2024-07-17:
> They are defines like this for me:
> 
> ```
> #define O_RDONLY	     00
> #define O_WRONLY	     01
> #define O_RDWR		     02
> ```
> 
> So `O_RDONLY | O_WRONLY != O_RDWR`

Well, OK. 
But if there are not too many flags, I think we should use bitmaps instead of the original define, which is more efficient. 
Such changes are not destructive.
As variables used as flags, there should be no user passing in original values.

closed 2024-08-10 by lvgl-bot