lvgl/lvgl #6508: Add a little dimension to the LVGL logo

kdschlosser, 2024-07-14:
### Introduce the problem

Not a problem at all. I git a little hair across my ass to give the LVGL logo some dimension.

### Proposal

This is the kind of thing companies pay some decent money for... You get it for the going rate of FREE..... I hope you like these and decide to use one or both. I can also make this animated but it would take me some time to do as I am not up to snuff with using blender. These are 2D images obviously but the actual work is a rendered 3D object. I have it saved as an STL, 3MF and OBJ currently. 

Tell me what you think

[image]

and even more depth

[image]

@kisvegabor

kisvegabor, 2024-07-15:
Thanks Kevin! I think our designer has a solid vision about the design, however we could use it very well for a 3D demo later. Can you attach the STL and 3MF files?

kdschlosser, 2024-07-16:
[lvgl_logo.zip](https://github.com/user-attachments/files/16243493/lvgl_logo.zip)

kdschlosser, 2024-07-16:
The STL file is a little booger'd up for some reason. not sure why or maybe it's just the preview that is messed up.

**update**
It looks like it is the preview that is messed up and not the actual STL file. When I open it in Blender it looks OK. 

Here is another one with more file formats.
[lvgl_logo2.zip](https://github.com/user-attachments/files/16243631/lvgl_logo2.zip)

kisvegabor, 2024-07-16:
Thank you!
It will be needed a little bit later (in a few months). I close this issue until then.

closed 2024-07-16 by kisvegabor