lvgl/lvgl #6512: There are a lot of macros that don't do anything.

kdschlosser, 2024-07-16:
### Introduce the problem

I wrote a script that is tasked with reading all of the header files, and source files to locate all of the macro's that are defined in LVGL. There is quite a few of them I will say that. in total there are 2159 macros defined. This doesn't include the macros used for include guarding or any macros that start and ends with dunders. 

of the 2159  macros there are 517 of them that are not used anywhere in LVGL at all other than being defined. Now I know that some of these are for use by the user and others might be to change settings in an external library. That would be for the people that know the code better than myself to sift through.

I wrote this script because of the sheer number of macros that are getting processed by the documentation build system so I wanted to know the number. and when it came out to the 2159 of them I was thinking there is no way that all of these are actually being used... As it turns out there is quite a few I am sure are no longer in use and can be removed.

### Proposal

Here is the list of macros that are not in use along with the files that they are defined in. There are sometimes the same macro defined in more than a single file.

<details>
  <summary>Click me for "The List"</summary>

```

AF_CONFIG_OPTION_INDIC
  lvgl\src\libs\freetype\ftoption.h

CIRCLE_CACHE_LIFE_MAX
  lvgl\src\draw\sw\lv_draw_sw_mask.c

CMD_DISSET5
  lvgl\src\drivers\display\st7735\lv_st7735.c

CMD_ETMOD
  lvgl\src\drivers\display\st7796\lv_st7796.c
  lvgl\src\drivers\display\st7789\lv_st7789.c

CMD_FRCTR2
  lvgl\src\drivers\display\st7789\lv_st7789.c

CMD_FRMCTR2
  lvgl\src\drivers\display\st7796\lv_st7796.c
  lvgl\src\drivers\display\st7789\lv_st7789.c
  lvgl\src\drivers\display\ili9341\lv_ili9341.c
  lvgl\src\drivers\display\st7735\lv_st7735.c

CMD_FRMCTR3
  lvgl\src\drivers\display\st7796\lv_st7796.c
  lvgl\src\drivers\display\st7789\lv_st7789.c
  lvgl\src\drivers\display\ili9341\lv_ili9341.c
  lvgl\src\drivers\display\st7735\lv_st7735.c

CMD_GAMSET
  lvgl\src\drivers\display\st7735\lv_st7735.c

CMD_IDXRD
  lvgl\src\drivers\display\ili9341\lv_ili9341.c

CMD_LCMCTRL
  lvgl\src\drivers\display\st7789\lv_st7789.c

CMD_NVFCTR1
  lvgl\src\drivers\display\st7735\lv_st7735.c

CMD_PORCTRL
  lvgl\src\drivers\display\st7789\lv_st7789.c

CMD_PWCTR4
  lvgl\src\drivers\display\st7789\lv_st7789.c
  lvgl\src\drivers\display\st7796\lv_st7796.c
  lvgl\src\drivers\display\st7735\lv_st7735.c

CMD_PWCTR5
  lvgl\src\drivers\display\st7789\lv_st7789.c
  lvgl\src\drivers\display\st7796\lv_st7796.c
  lvgl\src\drivers\display\st7735\lv_st7735.c

[... 1491 lines cut ...]
  lvgl\src\others\vg_lite_tvg\vg_lite.h

vg_lite_set_image_global_alpha
  lvgl\src\others\vg_lite_tvg\vg_lite.h

vg_lite_set_mask_layer
  lvgl\src\others\vg_lite_tvg\vg_lite.h

vg_lite_set_rad_grad
  lvgl\src\others\vg_lite_tvg\vg_lite.h

vg_lite_set_ts_buffer
  lvgl\src\others\vg_lite_tvg\vg_lite.h

vg_lite_set_update_stroke
  lvgl\src\others\vg_lite_tvg\vg_lite.h

vg_lite_update_rad_grad
  lvgl\src\others\vg_lite_tvg\vg_lite.h

```

</details>

If someone would spend the time to go through these and clean up the ones that are no longer in use that would be really helpful on the documentation end of things.

kdschlosser, 2024-07-25:
@kisvegabor 

I don't know if you saw this or not...

kisvegabor, 2024-07-29:
I've really missed it, thank you for pinging me.

I see 3 type of unused macros:
- The ones in 3rd party libs (FreeType, ThorVG): It doesn't make sense to fix on our end
- The ones that can be used by the user: e.g. `LV_STYLE_CONST_BG_IMAGE_OPA`
- The actual mistakes:  mostly display driver commands (makes sense to keep them for future usage) and a few others here and there.

kdschlosser, 2024-07-29:
I was just spitting them out. 

If things like Thor VG are included with LVGL then it would make sense to clean them up. It would be different if Thor VG was connected as a submodule. But you are in control of the code that is included with LVGL...

I am not even 100% sure why the Thor VG source is added to LVGL and why it's not a submodule.

kdschlosser, 2024-07-29:
The other thing is do you think it would be better to place all of the user "helper" macros into a single file. Something the user would be able to include, or perhaps a config setting to turn them on and off?? The reason why having the ability to turn them off would be nice is because for starters doxygen doesn't play nice nice with function like macros and they are of no value to have when generating code for a binding and can actually complicate things...

because of the bindings there really shouldn't be any function like macros for the purposes of being a helper for the user. If it's for the user then it should be a function and not a macro. macros are next to impossible to add to the binding code programmatically because there is no typing that is done for returned values or the parameters. no way to know what is expected.

kisvegabor, 2024-08-01:
The 113 `LV_STYLE_CONST_` needs to be macros to allow creating styles in compile time. So these are part of the API which might or might not be used by the user. 

> I am not even 100% sure why the Thor VG source is added to LVGL and why it's not a submodule.

We are not using submodules to
- keep things simpler
- make LVGL play nicely with all component managers
- allow customization on the 3rd party lib

kdschlosser, 2024-08-01:
the `LV_STYLE_CONST_` macros can be replaced by functions. The structure would need to be passed to the function along with the value instead.. 

macros in general are impossible to use in a binding that has the code programmatically generated. This is because there are no types that are used. If you have a macro that defines `1` is the `1` considered to be a `uint8_t`? could be but it could also be `uint16_t`, `uint32_t`, `int`, `int8_t`, `int16_t` or `int32_t`.. There is no way of knowing. typing is a rather important thing. When dealing with function like macros you have no clue what needs to be passed to the macro....  Then you have macros that work as simple code replacement and has names of things that don't exist in the macro itself but are names used in the code prior to where the macros code gets inserted. Once again, impossible to use in a binding.

kisvegabor, 2024-08-17:
> the LV_STYLE_CONST_ macros can be replaced by functions. The structure would need to be passed to the function along with the value instead.

It can't, because functions are not evaluated at compile time, therefore the result can't be stored in flash.

> macros in general are impossible to use in a binding that has the code programmatically generated.

It's not a big issue if constant styles are not available in the bindings. These are low level optimization which might not work well in high level language anyway.

kdschlosser, 2024-08-17:
OK...

vwheeler63, 2024-08-17:
> doxygen doesn't play nice nice with function like macros

Hi, Kevin!  ( @kdschlosser )

Can you help me understand the trouble you have run into with Doxygen and function-like macros?  I remembered a place in my library where I use these extensively, and it looks really good to me....  Here is an example in a .H file:  [[web link](http://crystal-clear-research.com/code_documentation/examples/m32/lib/wga/wg_ordered_circular_two_way_linked_list_8h.html#wgoctwll_in-lined_list-item_field_access)].

What am I missing?

kdschlosser, 2024-08-18:
> SKIP_FUNCTION_MACROS

> If the SKIP_FUNCTION_MACROS tag is set to YES then Doxygen's preprocessor will remove all references to function-like macros that are alone on a line, have an all uppercase name, and do not end with a semicolon. Such function macros are typically used for boiler-plate code, and will confuse the parser if not removed.

specifically `and will confuse the parser if not removed.`

https://www.doxygen.nl/manual/config.html

kdschlosser, 2024-08-18:
the default value for `SKIP_FUNCTION_MACROS`  is `yes`

vwheeler63, 2024-08-18:
> If the SKIP_FUNCTION_MACROS tag is set to YES then Doxygen's preprocessor will remove all references to function-like macros

SKIP_FUNCTION_MACROS   = YES

this is in the Doxygen config that I used to generate the documentation linked to above... [[link](http://crystal-clear-research.com/code_documentation/examples/m32/lib/wga/wg_ordered_circular_two_way_linked_list_8h.html#ad609cabf2cb000d61c1c3dfdc3d3639d)]

I seem to be missing something about what happens during generating of the documentation that causes a problem....

kdschlosser, 2024-08-18:
IDK if it causes a problem. I have never checked to see what the config file is set to for it. I do know that there are single line function macros in LVGL. Not sure if it causes any confusion with the preprocessor or not. I would imagine you would have to set that config setting to `no` and try it out and see what happens.

kdschlosser, 2024-08-18:
But. as ma point to make concerning the docstrings... IF I do the wormk to be able to parse doxygen docstrings and also ReST/Sphinx docstrings we will be able to properly type the macros parameters using ReST syntax. I don't think that is possible using doxygen docstrings.

in order to properly type using ReST/Sphinx directives you would have a doctsring keyed up like this....

```c
/**------------------------------------------------------------------------
 * \brief  Start -- set cursor to first item.
 *
 * :param apList:    List being traversed.
 * :type apList:       type name
 *
 * This macro is meant to be used in code like this (to avoid call/return
 * CPU overhead):
 * \code
 *	OCTWLList_t            * lpLst;
 *	OCTWLLItem_t           * lpItem;
 *	wgObserver_t           * lpObserver;
 *	wgObserverEventInfo_t    lEvtInfo;
 *
 *	require(val_ptr, apEvent != NULL);
 *	require(val_sit, Observer_qboolEventInitialized(apEvent));
 *
 *	lpLst = &apEvent->iObserverList;
 *	lEvtInfo.iuiEventID = auiEventID;
 *	lEvtInfo.iuiptrEventData = auiptrEventData;
 *
 *	for (mOCTWLL_Start(lpLst);  !mOCTWLL_qboolAfter(lpLst);  mOCTWLL_Forth(lpLst)) {
 *		lpItem = lpLst->ipCursor;
 *		lpObserver = lpItem->ivpOwner;
 *
 *		if (lpObserver->ifxnpUpdate) {
 *			lEvtInfo.iuiptrContext = lpObserver->iuiptrContext;
 *			lpObserver->ifxnpUpdate(&lEvtInfo);
 *		}
 *	}
 * \endcode
 *
 * mOCTWLL_Finish(), mOCTWLL_qboolBefore() and mOCTWLL_Back() respectively
 * will make the loop iterate in the reverse direction.
 *//*----------------------------------------------------------------------*/
#define mOCTWLL_Start(apList)                              ((apList)->ipCursor = (apList)->iListHub.ipRight)
```

That is something you are not able to do with Doxygen. Doxygen figures out the types based on the declaration/definition and since macros don't have the type names listed there is no way for Doxygen to know what they are unless you are able to manually add it.

I am going to have to use some coding Kung Fu to handle that because in the xml output from Doxygen there is no separation between a function-like macro and a traditional macro... It would be pretty easily done as all I have to do is look for a parenthesis in the macro definition. Then I could add it to the Macros section of the documentation for that file but I would add it so it comes out looking like a function.

The hard part would be having the c domain reference work properly because to add it so it documents like a function would mean the c directive used would need to be for a function but I would also nee to add it as a macro and that would cause a duplicate name issue. I might have to create a new c domain directive for those kinds of macros.

vwheeler63, 2024-08-18:
> I do know that there are single line function macros in LVGL.

Without semicolons after them?

kdschlosser, 2024-08-18:
That I do not know.

vwheeler63, 2024-08-18:
Okay.  Did you run into some kind of problems with Doxygen's evaluation of function-like macros?  I'd be happy to make Doxygen generate XML off my library if it would help you to dig into it, if it helps.  The wgObserver (I'm going to call it a "class" because that's how it is used) makes use of those macros, so you could see the source documentation _and_ where it is used, if that helps.  Let me know.

kdschlosser, 2024-08-18:
I have to take off for  some of the day. I will be back early evening and I can put some time into messing about with this.

I do have some interesting information.. The Doxygen XML files are not unique to Doxygen, the  XML API is what is called Docbook format. Now knowing that is going to make it easier to bring Doxygen and Sphinx together.

vwheeler63, 2024-08-18:
> Docbook format.

Hot!!  I had heard of Docbook, but hadn't looked into it....  Nice find, sir!!

kdschlosser, 2024-08-19:
I have to do some more looking but just with a quick glance it looks like the docbook XML output is going to be easier to deal with than the doxygen XML output is. The Doxygen XML is a mangled form of docbook that has a whole lot of added complexity to it that is not really needed when using it to feed information into sphinx.

vwheeler63, 2024-08-19:
> is going to be easier to deal with than the doxygen XML output is.

I'm concerned.  It sounds like you are thinking of replacing Doxygen as the initial "data provider".  Is that what I am hearing?

kdschlosser, 2024-08-19:
doxygen would still be needed to output no matter what.

I wish that doxygen was written so it could be accessed without writing anything to disk. If it had some kind of exported API...

vwheeler63, 2024-08-19:
> I am just saying that the docbook format that it outputs is easier to parse than the doxygen xml output

Ah!  Now I feel enlightened.  So you're thinking of turning off the XML output and setting

```
GENERATE_DOCBOOK       = YES
```

Interesting....

kdschlosser, 2024-08-19:
Possibly. It appears like the XML is far less complex. It's basically just the raw documentation. It's not trying to replay any formatting information, or at least very little of it.

vwheeler63, 2024-08-19:
> Possibly. It appears like the XML is far less complex. It's basically just the raw documentation. It's not trying to replay any formatting information, or at least very little of it.

Ouch.  I see.  From what I have seen of the XML, it makes the structure -- what _part_ of the documentation is being presented in each node -- explicit, so it would seem like the correct place to take input, given that we're generating a structured output that can have many components....

Are you back to using `breathe` again?  Or what is it that you're up to at the moment?

kdschlosser, 2024-08-20:
I am looking at all different aspects of it. I am "learning" how all of this stuff works.

The problem with breathe is it's not complete for one.. For 2 there are things that are not done correctly in it like placing C structures into CPP stricture directives. This makes it confusing when having to write the documentation due to having to keep on checking what uses the C directives and what uses the CPP directives. It's rather annoying actually. It doesn't handle some of the docbook markup tags properly which causes the documentation to look like someone that speaks broken English wrote it because it ends up omitting words that have some types of markup applied to them. While I am working around the issue of how breathe is handling reading the information how it normally works is it would run doxygen against one file at a time. This causes a problem when using macros to define config settings because not every single header file that needs to be read includes lv_conf.h so the config settings would get all jumbled up if we used it the way it is designed to be used. 

The biggest issue with doxygen is it seem to like to add all kinds of extra white space and in some cases it doesn't clean off the extra white space it should be scrubbing when it parses the documentation. It is leaving that up to the HTML compiler/renderer in a browser to handle. bytes = money when sending anything over the internet. if the white space doesn't need to be there then why send it?

I also have things going on here at home that I have to take care of so the amount of time I am spending is very limited right now. I wish I had more to invest into it but I don't. The couple of PR's needed to correct the current problems with the documentation have not been merged yet and I am at a standstill until that happens. 

The PR that Gabor made that added the `_` back to the structure names is failing the CI. It is failing because parts of the API got moved into being private and the code generator for the MicroPython binding is not able to see those pieces so it is not writing the correct code which is causing it to generate incorrect code and that is why it's failing. It compiles without error but it doesn't run correctly. So until that gets fixed and the PR merged I am not able to move forward.

vwheeler63, 2024-08-20:
> I am "learning" how all of this stuff works.

Nice.

> the amount of time I am spending is very limited right now.

Understood.

> I am at a standstill until that happens.

Ouch.  Got it.

closed 2024-09-05 by kdschlosser