lvgl/lvgl #6519: Evdev driver doesn't handle LV_INDEV_TYPE_BUTTON input device type correctly if input event code is BTN_MOUSE or BTN_TOUCH

tmilkovic51, 2024-07-17:
### LVGL version

v9.1.0

### What happened?

I want to emulate a touch event at a specific coordinate using a physical button on Linux. I am using `evdev` as an LVGL input driver for `/dev/input/event0` event (a physical button with `BTN_TOUCH` event code is connected to this event) and configured the input device type in LVGL as `LV_INDEV_TYPE_BUTTON`.

Further, I assigned the point coordinates to the input device where touch should be emulated (as described [here](https://docs.lvgl.io/master/porting/indev.html#button)) and placed an LVGL button widget at those coordinates.

However, when I press the physical button, button widget's callback function is never called.

By debugging further, I noticed that `evdev` input driver in LVGL never sets the `btn_id` member of `lv_indev_data_t` structure, so I implemented a little fix in [this commit](https://github.com/lvgl/lvgl/commit/290c9469cee7bac1da361a1d7ffed5bf316f78da). 
I'm not sure if this is the correct way to use the evdev driver. If it is, I can open a pull request.

### How to reproduce?

Configuration:
- Linux side:
  - One physical button connected to GPIO on MPU - Configured by adding `gpio-keys`-compatible node in device tree with `linux,code` property set either to `BTN_MOUSE` (`0x110`) or `BTN_TOUCH` (`0x14A`)
    - `evtest` utility shows that events are correctly emitted on `/dev/input/event0` device when pressing and releasing the physical button
- LVGL side:
  - Used [lv_port_linux](https://github.com/lvgl/lv_port_linux) project as a starting sample
  - Linux framebuffer device used as display (`LV_USE_LINUX_FBDEV` is set to `1` in `lv_conf.h`)
  - Linux `evdev` is used as an input device (`LV_USE_EVDEV` is set to `1` in `lv_conf.h`)
  - In `main` function, `evdev` input device with  `LV_INDEV_TYPE_BUTTON`  type is created by calling:
    -  `indev = lv_evdev_create(LV_INDEV_TYPE_BUTTON, "/dev/input/event0");`
  - Touch points to be emulated are configured for the created `evdev` input device by calling:
    - `lv_indev_set_button_points(indev, points_array);`
  - A button widget is placed at the coordinates where touch event should take place, and `LV_EVENT_PRESSED` event callback is added to the button

Expected:
- Button event callback function is called with `LV_EVENT_PRESSED` and `LV_EVENT_RELEASED` as event codes

Result: 
- Button event callback function is never called with `LV_EVENT_PRESSED` and `LV_EVENT_RELEASED` event codes

kisvegabor, 2024-07-22:
Hi,

Thanks for pointing it out! 

It seems like a straightforward update! Could you add it an send a Pull request?

tmilkovic51, 2024-07-23:
Opened #6558.

After it is merged, I'll open backport PRs for v9.1 and v9.0 release branches. 

On releases <= v8.4 (where drivers were in a [separate git repo](https://github.com/lvgl/lv_drivers/blob/release/v8.3/indev/evdev.c)), I see that evdev handled those events completely differently, so this fix is not applicable to those releases.

closed 2024-08-14 by lvgl-bot