lvgl/lvgl #6524: Significant performance difference between V8.3 and V9 when using SDL

darko31, 2024-07-17:
### LVGL version

v9.0.0

### What happened?

Hello,

first of all thank you very much for this awesome library. I am sure that everybody appreciates this kind of project.

While testing lv_port_pc_eclipse on my PC and also on the target hardware that has Weston running (IMX8ULP-EVK board
with 1280x720 screen attached to it), I noticed a significantly higher CPU usage on V9 version of the demo in comparison to the v8.3.

This is not such a big issue on the PC, but running the V9 demo on the IMX8ULP yields almost 100% CPU usage on a single core and somewhere around 10 FPS when the Analytics tab in the demo is showed with all the graphs.

The IMX8ULP is no powerhouse, it's a dual core ARM A35 running at 800MHz, but still, running the v8.3 version of the demo the FPS is stable at 33 FPS and CPU usage is around 50%.

This can be seen also with the natively compiled demos on the PC. Here's a comparison
[image]

the left window is v9 version while the right one is v8.3.
I've set the resolution to 480x800, the CMake release is set to debug and LV_USE_ASSERT_MEM_INTEGRITY and 
LV_USE_ASSERT_OBJ are set to 0.

Is this difference because of the slightly different contents displayed on the screen or is something else at play?
The first Profile tab of the demo looks pretty much the same between v8.3 and v9 and v9 again has significantly higher CPU usage (5% vs 30%).

I ran both apps with with perf and loaded debug symbols for the libSDL2 and here are the captures
[image]
[image]

v8.3 and v9 seem to use a different way of rendering things, BlitRGBtoRGBPixelAlpha vs BlitNtoNPixelAlpha.

Excuse my lack of understanding, I'm still learning about the LVGL and SDL.

If you have some suggestions on how to optimize the V9 version, please share.

Thank you!
Darko

kisvegabor, 2024-07-18:
Hi,

Thank you reporting this issue! An important questions: do use pure software rendering or do you use SDL rendering (`LV_USE_DRAW_SDL   1` in v9)

darko31, 2024-07-18:
> Hi,
> 
> Thank you reporting this issue! An important questions: do use pure software rendering or do you use SDL rendering (`LV_USE_DRAW_SDL 1` in v9)

Ah, it is not enabled because if I enable it, it doesn't show any pictures at all and prints this error

`could not load image: SDL_RWFromFile(): No file or no mode specified`

Screenshoots below
[image]
[image]

This is with release/v9.0 and release/v9.1 tags.

If I switch to the latest master (0a1e411038306b3fb4ee846dd6999907ef6222f2), the Profile page looks good
[image]

but Analytics page is still looking weird
[image]

I'm looking into the updates in the lv_sdl_window.c but let me know if I should look at something else.
Thanks!

kisvegabor, 2024-07-19:
On paper the v9 SDL implementation should be faster because it used XRGB8888 format instead of ARGB888 and it doesn't copy the rendered area to SDL's buffer, but renders directly there.

Maybe your devices likes ARGB888 more. Can you try changing [this line](https://github.com/lvgl/lvgl/blob/master/src/drivers/sdl/lv_sdl_window.c#L420) to `SDL_PIXELFORMAT_ARGB8888`?

Please keep using the software rendering. The SDL based rendering is being updated [here](https://github.com/lvgl/lvgl/pull/6437). I just wanted to be sure what you are using.

darko31, 2024-07-19:
Okay, I'll try it out. Let me know if I can do some testing and/or debugging related to this although it seems that it can be consistently reproduced on a PC, it doesn't require some specific setup.

darko31, 2024-07-22:
> On paper the v9 SDL implementation should be faster because it used XRGB8888 format instead of ARGB888 and it doesn't copy the rendered area to SDL's buffer, but renders directly there.
>
> Maybe your [...]

The default lv_conf.h in the lv_pc_port_eclipse has LV_COLOR_DEPTH set to 16, but after changing it to 32 and making the change you suggested, the CPU usage on the IMX8ULP drops down to around 52% to 60% and FPS is pretty stable at ~28 FPS when Analytics tab is open and scrolled down to the network speed widget. This is slightly higher than V8.3.

Thanks for the suggestion!

I'm interested to try out the https://github.com/lvgl/lvgl/pull/6437 to see how it works.

kisvegabor, 2024-08-12:
Some performance issues are reported in https://github.com/lvgl/lvgl/pull/6437 too. I'll look into it this week. Let's continue in the PR.

closed 2024-08-12 by kisvegabor