lvgl/lvgl #6528: No functions to set long_press_time and other indev properties

clydebarrow, 2024-07-18:
### LVGL version

V9.1

### What happened?

V8 allowed `long_press_time` and `long_press_repeat_time` to be set for input devices. This is no longer possible in V9 as those properties are defined in `lv_indev_private.h` and are thus not accessible to user code, and `lv_indev.h` provides no setter functions.

This likely also applies to other indev properties such as `scroll_limit`.

liamHowatt, 2024-07-26:
I'm not sure why they were removed but sensible defaults are sometimes nicer than too many configuration options. The long press durations have come up a few times so it might be a good candidate for having a setter. @kisvegabor?

kisvegabor, 2024-07-26:
That's true! I agree that we should have public both setters and getters for these properties. @clydebarrow do you have time and interest to send a Pull request?

kisvegabor, 2024-08-15:
Fixed by #6633

WebDust21, 2024-12-28:
Not _totally_ fixed in #6633; that only added a getter/setter for "indev->long_press_time".

There's still (in LVGL v9.2.0) no getters/setters for:

- indev->long_press_repeat_time
- indev->gesture_limit
- indev->gesture_min_velocity
- indev->rotary_sensitivity

and as noted elsewhere, the default #define values are also not in "lv_conf.h"; as a result, LVGL library source has to be modified to change them.  (In other words, changes will be lost when updating the LVGL library.)

ran into this trying to adjust the gesture thresholds (default threshold of 50 pixels is too sensitive for me on a 7" 1024x600 panel), only to find that I have to modify LVGL source to change them.  IMO without getter/setter functions, ideally the defaults would either be located in "lv_conf.h".

It used to be possible to "hack" into the internal LVGL structures from the returned pointers--but in v9, this solution has been completely eliminated by "private" types, i.e. this type of simple workaround code is no longer possible from userspace:
```
lv_indev_t *indev = lv_indev_active(); 
indev->gesture_limit = 200;
```

WebDust21, 2024-12-28:
EDIT: Found a very important entry in "lv_conf.h"...

```
/*Include `lvgl_private.h` in `lvgl.h` to access internal data and functions by default*/
#define LV_USE_PRIVATE_API		0
```

If that's enabled, then the above little snippit works to perfection...well, that is, as long as you're not trying to call it immediately after "lv_x11_inputs_create", which _doesn't_ return an "indev_t*" pointer.  (Both "lv_libinput_create" and "lv_evdev_create" return an "indev_t" pointer.)

In the case of X11, trying to get the indev handle with "lv_indev_active()" is also problematic, as it returns NULL until _after_ "lv_indev_read" has been called at least once.  This happens internally based on a timer--making it quite impractical for adjusting the touch setup functionality.

_Getting_ the "indev" handle from X11 is ridiculously painful, but with "LV_USE_PRIVATE_API" enabled, it at least is possible.  Note that I had to duplicate the "_x11_inp_data" structure, [as it's located in a C source file](https://github.com/lvgl/lvgl/blob/c903c1dc0fdfae20bce69d0a48894bdebae278ff/src/drivers/x11/lv_x11_input.c#L28-L42), not a C header. 
Yes, I know that dumping that struct into inline source is messy--but the whole _job_ is messy!
```
	// Initialize the touch driver.
	lv_indev_t *indev = NULL;

#if LV_USE_LIBINPUT
	indev = lv_libinput_create(LV_INDEV_TYPE_POINTER, "/dev/input/event0");
	lv_indev_set_display(indev, disp);
#elif LV_USE_EVDEV
	indev = lv_evdev_create(LV_INDEV_TYPE_POINTER, "/dev/input/event0");
	lv_indev_set_display(indev, disp);
#elif LV_USE_X11
#include "mouse_cursor_icon.h"
	lv_x11_inputs_create(disp, &mouse_cursor_icon);       /* initialize X11 input drivers (for keyboard, mouse & mousewheel) */

	// Painfully extract the touchscreen "indev_t" pointer from X11...
typedef struct _x11_inp_data {
    /* LVGL related information */
    lv_group_t * inp_group;      /**< input group for X input elements */
    lv_indev_t * keyboard;       /**< keyboard input device object */
    lv_indev_t * mousepointer;   /**< mouse input device object */
    lv_indev_t * mousewheel;     /**< encoder input device object */
    lv_timer_t * timer;          /**< timer object for @ref x11_event_handler */
    /* user input related information */
    char         kb_buffer[32];   /**< keyboard buffer for X keyboard inputs */
    lv_point_t   mouse_pos;       /**< current reported mouse position */
    bool         left_mouse_btn;  /**< current state of left mouse button */
    bool         right_mouse_btn; /**< current state of right mouse button */
    bool         wheel_mouse_btn; /**< current state of wheel (=middle) mouse button */
    int16_t      wheel_cnt;       /**< mouse wheel increments */
} x11_inp_data_t;

	_x11_user_hdr_t *disp_hdr = lv_display_get_driver_data(disp);
	LV_ASSERT_NULL(disp_hdr);
	x11_inp_data_t ** inp_data = &disp_hdr->inp_data;
	indev = (*inp_data)->mousepointer;
#endif

	// Increase the minimum "gesture" threshold so it's not so sensitive
	if (indev != NULL) {
		indev->gesture_limit = 200;
	} else {
		fprintf(stderr, "Could not get INDEV pointer.\n");
	}
```

Yes, it works.

kisvegabor, 2025-01-03:
> indev->long_press_repeat_time
indev->gesture_limit
indev->gesture_min_velocity
indev->rotary_sensitivity

We really need API for these as well. If you have some free time, a PR would be highly appreciated. :pray: 

> Getting the "indev" handle from X11 is ridiculously painful

Functions like `lv_indev_t * lv_x11_get_mouse/mousewheel/keypad()` seems like a useful addition. What do you think?

WebDust21, 2025-01-04:
> If you have some free time, a PR would be highly appreciated. 🙏

Well, can't say I have "free time", as I've been spending all my available time the past week on updating/completing @XuNeo 's Lua LVGL bindings.  There's a LOT of work required there before it'll be ready as full LVGL bindings--but it's going to be worth it.

by the time I'm done with that, I should have a pretty good list of all the LVGL properties that don't have getters...and can take care of all of those with one PR.
(currently have adapted 27 of LVGL's 97 widget examples to LuaVGL.)

kisvegabor, 2025-01-09:
> Well, can't say I have "free time", as I've been spending all my available time the past week on updating/completing @XuNeo 's Lua LVGL bindings. There's a LOT of work required there before it'll be ready as full LVGL bindings--but it's going to be worth it.

Oh, that's amazing! 

> by the time I'm done with that, I should have a pretty good list of all the LVGL properties that don't have getters...and can take care of all of those with one PR.

It's great too. An API unification is always very welcome :slightly_smiling_face: