lvgl/lvgl #6529: v_screen_load_anim will retain the previous interface

IAMMX, 2024-07-18:
### LVGL version

master

### What happened?

lv_screen_load_anim with auto_del set to true should remove the previous interface, but it is not being cleared. Please fix this issue and subsequently, add a CI test to prevent regression in the future.
cc @liamHowatt  @kisvegabor 

### How to reproduce?

```c
lv_obj_t *screen_1,*screen_2;
lv_obj_t *screen_1_btn;
lv_obj_t *screen_2_label;

static void btn_event(lv_event_t * e)
{
  lv_event_code_t code = lv_event_get_code(e);
  if(code == LV_EVENT_CLICKED)
  {
    screen_2=lv_obj_create(NULL);
    lv_obj_set_style_bg_opa(screen_2,LV_OPA_50,0);
    screen_2_label=lv_label_create(screen_2);
    lv_label_set_text(screen_2_label, "this is screen_2");
    lv_obj_center(screen_2_label);
    lv_screen_load_anim(screen_2,LV_SCR_LOAD_ANIM_NONE, 0, 0, true);
  }
}

void test_scr_load_anim(void)
{
  screen_1=lv_obj_create(NULL);

  screen_1_btn=lv_button_create(screen_1);
  lv_obj_center(screen_1_btn);
  lv_obj_add_event_cb(screen_1_btn, btn_event, LV_EVENT_CLICKED, NULL);

  lv_screen_load(screen_1);
}
```

liamHowatt, 2024-07-23:
Thanks. It seems to not be related to loading animations. I confirmed `screen_1` does get deleted. Decreasing the bg opacity of a loaded screen object seems to be poorly defined behavior. Do you have a use case for setting the screen opacity to less than 100%? Maybe some logic could be added in LVGL to ignore the bg opacity style of screen objects.

IAMMX, 2024-07-24:
Thanks. It's indeed an issue of transparency. I'm looking to implement a screen transition while keeping the background image stationary. I've already achieved this by manipulating the `lv_layer_bottom`. Thank you for your assistance.

closed 2024-07-24 by IAMMX