lvgl/lvgl #6532: in v8.x are there any apis to help to implement undo and redo funtion?

docterling, 2024-07-19:
### Introduce the problem

just like the android's whiteboard app do.

liamHowatt, 2024-07-26:
For sure! What kind of state changes would be undone and redone, in your application? I can try to give an example of how it could be done.

docterling, 2024-07-27:
[image]
for example,i've drawn 1-5 numbers(or 1-5 lines) in canvas,if i press undone btn it would show 1-4(or 1-4 lines),press again show 1-3(or 1-3 lines),and now press redone btn it would show 1-4(or 1-4 lines),press again show 1-5(or 1-5 lines)
can this be done in undone and redone callbacks in lvgl v8.4?

liamHowatt, 2024-07-29:
There are no direct APIs for undo-redo since it's pretty niche. The use case is fairly application-specific and users would probably want to implement it differently each time. It's not too difficult to implement the logic for your own application.

Are use using `lv_canvas_set_px`? Or `lv_canvas_draw_line`? The tricky part is implementing undo-redo efficiently.

You can use this as a starting point. Let me know if you have any questions or if it doesn't work for some reason.

```c
#define CANV_W   300
#define CANV_H   300
#define HIST_MAX 100

static struct {
  lv_point_t points[HIST_MAX];
  int len;
  int full_len;
} hist = {
  .points = { {CANV_W / 2, CANV_H / 2} },
  .len = 1,
  .full_len = 1
};

static void redraw_canv(lv_obj_t * canv)
{
  lv_draw_line_dsc_t line_dsc;
  lv_draw_line_dsc_init(&line_dsc);
  line_dsc.color = lv_color_white();
  line_dsc.opa = LV_OPA_COVER;
  line_dsc.width = 3;

  lv_canvas_fill_bg(canv, lv_color_black(), LV_OPA_COVER);
  lv_canvas_draw_line(canv, hist.points, hist.len, &line_dsc);
}

static void click_cb(lv_event_t * e) {
  if(hist.len >= HIST_MAX) {
    return;
  }
  hist.len++;
  hist.full_len = hist.len;

  lv_indev_t * indev = lv_event_get_indev(e);
  lv_obj_t * canv = lv_event_get_target(e);
  lv_point_t point;
  lv_indev_get_point(indev, &point);
  lv_area_t canv_coords;
  lv_obj_get_coords(canv, &canv_coords);
  point.x -= canv_coords.x1;
  point.y -= canv_coords.y1;

  hist.points[hist.len - 1] = point;

  redraw_canv(canv);
}

static void undo_cb(lv_event_t * e) {
  if(hist.len <= 0) {
    return;
  }
  hist.len--;
  if(hist.len > 0) {
    lv_obj_t * canv = lv_event_get_user_data(e);
    redraw_canv(canv);
  }
}

static void redo_cb(lv_event_t * e) {
  if(hist.len < hist.full_len) {
    hist.len++;
    lv_obj_t * canv = lv_event_get_user_data(e);
    redraw_canv(canv);
  }
}

static void undo_redo(void) {
  lv_obj_t * canv = lv_canvas_create(lv_scr_act());
  lv_obj_center(canv);
  lv_obj_add_flag(canv, LV_OBJ_FLAG_CLICKABLE);
  static lv_color_t cbuf[LV_CANVAS_BUF_SIZE_TRUE_COLOR(CANV_W, CANV_H)];
  lv_canvas_set_buffer(canv, cbuf, CANV_W, CANV_H, LV_IMG_CF_TRUE_COLOR);
  redraw_canv(canv);
  lv_obj_add_event_cb(canv, click_cb, LV_EVENT_CLICKED, NULL);

  lv_obj_t * undo_btn = lv_btn_create(lv_scr_act());
  lv_obj_align(undo_btn, LV_ALIGN_LEFT_MID, 0, 0);
  lv_label_set_text(lv_label_create(undo_btn), "undo");
  lv_obj_add_event_cb(undo_btn, undo_cb, LV_EVENT_CLICKED, canv);

  lv_obj_t * redo_btn = lv_btn_create(lv_scr_act());
  lv_obj_align(redo_btn, LV_ALIGN_RIGHT_MID, 0, 0);
  lv_label_set_text(lv_label_create(redo_btn), "redo");
  lv_obj_add_event_cb(redo_btn, redo_cb, LV_EVENT_CLICKED, canv);
}
```

[image]

docterling, 2024-08-02:
i've tried your idea
it can basicly work,thank you.
and what if i want to implement erase function in a transparent backgroud
a red line or something when i click on an eraser btn,i swipe the eraser over the red line,the swipe over part would become transparent

liamHowatt, 2024-08-06:
Sounds cool. Are you able to try adding it and let me know if you get stuck? Is there a part you are unsure about?

closed 2024-08-30 by lvgl-bot