lvgl/lvgl #6533: could provide lvgl_image_convert and lvgl_font_convert tools or functions by C/C++ in project，not to convert online, need to convert Dynamic in project.

xinsuinizhuan, 2024-07-19:
### Introduce the problem

need to convert image or font Dynamic in project，but now have no lvgl_image_convert and lvgl_font_convert interface，image and font need to convert by other language tool online or static。

xinsuinizhuan, 2024-07-19:
https://gitee.com/Jumping99/lv_font_conv this is a tool by C++，but english is ok, some icon and chinese is error:
[image]
[image]

liamHowatt, 2024-07-26:
Do you know about `lv_tiny_ttf`? You can load fonts dynamically with it. It has has been improved a lot recently.

For images, the PNG decoder essentially converts the image and caches it the way you describe. Does that meet your needs?

xinsuinizhuan, 2024-08-19:
need lvgl_font_convert tool by c++

liamHowatt, 2024-08-19:
Just to clarify, you want a desktop C++ application that does what https://github.com/lvgl/lv_font_conv does? What is https://github.com/lvgl/lv_font_conv not capable of, that you need?

xinsuinizhuan, 2024-08-20:
yes，i want the lv_font_conv add to the lvgl， lvgl is C++, but the  https://github.com/lvgl/lv_font_conv is not c++， i could not add them together。

liamHowatt, 2024-08-21:
Font conversion does not typically happen at runtime in an LVGL application. Are you trying to convert TFT fonts? Can you use TinyTFT? It's built into LVGL and intended to be used at runtime.

I know you use LVGL on Windows so if you really wanted you could invoke the JavaScript font converter from your Windows C++ application.

Would you mind describing what you're trying to create? Are you trying to create a custom frontend for the font converter in LVGL?

I want to understand the use case so please excuse my many questions.

closed 2024-09-16 by lvgl-bot