lvgl/lvgl #6534: Using evaluate to decide on a specific draw unit

cristian-stoica, 2024-07-19:
### Introduce the problem

The current behavior of LVGL is to call the evaluate function of each draw unit for them to bid on execution of a particular draw task.
This evaluation is done at runtime, during frame rendering. One advantage of this is that it is possible (at least in principle) to arbitrarily change the draw unit at runtime, during application lifetime. The disadvantage is that the performance of the rendering is impacted by this evaluation. An alternative design could decide on the draw unit dispatch for all tasks once, during initialization. That would introduce (again theoretical) inflexibility in the dispatch decision but with the advantage of reduced CPU utilization.

Considering that LVGl applications are targeting embedded devices, do we find useful to review an alternative approach to when task evaluation is being made?

kisvegabor, 2024-08-03:
We could map e.g `LV_DRAW_TASK_TYPE_IMAGE` to a draw unit, but we couldn't make more specific decision based on particular settings. E.g. what if GPU doesn't support `LV_COLOR_FORMAT_RGB565A8` images?

cristian-stoica, 2024-08-05:
Do the drawing units have more information on the color capabilities of the GPUs during scene rendering than during initialization? If a GPU doesn't support some color format is it inherent with the HW or is it a capability that can be arbitrarily disabled at runtime?

If the code must traverse a tree of predicates to decide what drawing unit is capable isn't the result independent of when the question is being asked?

(Dispatching to a particular capable drawing unit can be decided based on a policy on the availability of the unit but I think this is a different matter than finding out what units are capable)

kisvegabor, 2024-08-07:
It seems I don't understand your proposal.

> An alternative design could decide on the draw unit dispatch for all tasks once, during initialization.

During the initialization of the draw task or the draw unit? Can you draft a rough flow about the proposed behavior? Just to be sure that we are on the same page.

cristian-stoica, 2024-08-07:
During initialization of the application, of the draw units.
For each task type, it can be decided once, at app initialization, what draw unit will render the task. This can be recorded in some data structure of function pointers that records all task types.
Then, at dispatch, that structure will be indexed by the task type to obtain and execute the draw function for that task type.

What's currently being done is setting preference scores per task type **during task creation**:
```
       /*Let the draw units set their preference score*/
        t->preference_score = 100;
        t->preferred_draw_unit_id = 0;
        lv_draw_unit_t * u = info->unit_head;
        while(u) {
            if(u->evaluate_cb) u->evaluate_cb(u, t);
            u = u->next;
        }
```

and then dispatching to all available draw units until one picks it up:
```
        /*Let all draw units to pick draw tasks*/
        lv_draw_unit_t * u = _draw_info.unit_head;
        while(u) {
            int32_t taken_cnt = u->dispatch_cb(u, layer);
            if(taken_cnt != LV_DRAW_UNIT_IDLE) task_dispatched = true;  /// -> shouldn't this also break here?
            u = u->next;  
        }
```

kisvegabor, 2024-08-17:
I feel like there are too many options even in a single draw task type. Speaking of a simple `FILL`:
- has radius?
- no gradient?
- horizontal vertical gradient?
- more complex gradient?
- small fill might be faster with SW?

Could measure the execution time of the `evaluate_cb`? If so did it really need significant performance? If so I'm absolutely open to optimize it, however it's execution time is negligible I'd focus on something where we can gain more.

cristian-stoica, 2024-08-26:
I appreciate it. We already have some lower hanging fruits we work on so this would be a best-effort project.
I would also like to have a feel for return on investment before making too many changes.
Thanks

closed 2024-09-19 by lvgl-bot