lvgl/lvgl #6536: about tests CMakeLists path configurations

IAMMX, 2024-07-20:
### Introduce the problem

Why is an absolute path being used here, and can it be changed to a relative path? This is because lvgl is located in a directory with Chinese characters in its path, and using an absolute path can result in errors.
I am using the Windows system.
[image]

kisvegabor, 2024-08-03:
I have no idea to be honest. Do you know how to change it relative path?

IAMMX, 2024-08-05:
Sorry, I don't know either. 
@liamHowatt Can you take a look?

liamHowatt, 2024-08-06:
Fyi, all CI builds use CMake+Ninja. Is that what you're using on Windows? [these two commands](https://github.com/lvgl/lvgl/blob/7e3be59dda1a0929744b0b4fcacf5d1734169199/tests/main.py#L98-L101), for reference.

I found [`CMAKE_NINJA_OUTPUT_PATH_PREFIX`](https://cmake.org/cmake/help/latest/variable/CMAKE_NINJA_OUTPUT_PATH_PREFIX.html) which adds to the start of the path, but cannot remove anything from the path. Also there are many ways to add custom messages, yet no way to format Ninja's build messages that I could find.

ChatGPT suggested filtering the output with a script. Since you're on Windows, the easiest way would be a simple Python script.

```
<your build command> | python3 filter_output.py
```

```py
import sys
for line in sys.stdin:
    print(line.replace("C:\...\...", ""), end='')
```

closed 2024-08-19 by IAMMX

liamHowatt, 2024-08-20:
Sorry I didn't have a more satisfying answer. I hope you're able to do what you need.