lvgl/lvgl #6538: LVGL 9.1.1 hangs in lv_display_set_buffers - one solution found

Westcott1, 2024-07-20:
### LVGL version

9.1.1

### What happened?

This works OK in LVGL 8
However in v9 my code would hang in lv_display_set_buffers.
I found that it was caused by using a global char buf[] in an sprintf statement.
Changing the buf to a local one fixes the problem.

### How to reproduce?

THIS HANGS IN lv_display_set_buffers()
-------------------------------------------
char globalBuf[50];

void LVGLinit() {
  Log("LVGL init...", INFO);

  lv_init();

  disp = lv_display_create(TFT_HOR_RES, TFT_VER_RES);
  lv_display_set_flush_cb(disp, DisplayFlush);
  lv_display_set_buffers(disp, buf[0], buf[1], sizeof(buf[0]), LV_DISPLAY_RENDER_MODE_PARTIAL);

  indev = lv_indev_create();
  lv_indev_set_type(indev, LV_INDEV_TYPE_POINTER);
  lv_indev_set_read_cb(indev, TouchRead);

  sprintf(globalBuf, "LVGL version: %d.%d.%d\n", lv_version_major(), lv_version_minor(), lv_version_patch());
  Log(globalBuf, INFO);

THIS WORKS CORRECTLY
---------------------------
void LVGLinit() {
  Log("LVGL init...", INFO);

  lv_init();

  disp = lv_display_create(TFT_HOR_RES, TFT_VER_RES);
  lv_display_set_flush_cb(disp, DisplayFlush);
  lv_display_set_buffers(disp, buf[0], buf[1], sizeof(buf[0]), LV_DISPLAY_RENDER_MODE_PARTIAL);

  indev = lv_indev_create();
  lv_indev_set_type(indev, LV_INDEV_TYPE_POINTER);
  lv_indev_set_read_cb(indev, TouchRead);

  char localBuf[50];
  sprintf(localBuf, "LVGL version: %d.%d.%d\n", lv_version_major(), lv_version_minor(), lv_version_patch());
  Log(localBuf, INFO);

mr-sven, 2024-07-22:
where do you define `buf[0]` `buf[1]` ?

kisvegabor, 2024-07-22:
It interesting because when you do something with `globalBuf` or `localBuf`, `lv_display_set_buffers` already returned. Only `char globalBuf[50];` is before `lv_display_set_buffers`. The declaration itself changes the memory map, and probably just hides some other error (e.g. write out of bounds).

Westcott1, 2024-07-22:
Hi Kisvegabor,
Thanks, yes, I was thinking the same things.
It certainly never gets as far as actually executing sprintf.
However, that code works perfectly well in in LVGL 8.

Westcott1, 2024-07-22:
Hi Mr-Sven,
`static uint16_t buf[2][TFT_HOR_RES * TFT_VER_RES / 10];`
Just before globalBuf.

mr-sven, 2024-07-22:
Ahm, as recommended you should use `buf1 = heap_caps_malloc(buffer_size, MALLOC_CAP_SPIRAM);`.
The ESP32 has only a few kilobytes of internal RAM, so if you hard define the buffers global it may cause out of memory. The ESP IDF sample points to that and says you should use the external SPIRAM for that.

Westcott1, 2024-07-22:
Thanks, I'll look at the example.
However, the double buffer method I'm using works really well!

closed 2024-08-13 by lvgl-bot