lvgl/lvgl #6545: ESP-IDF LVGL 9.x Direct rendering double buffer glitches

mr-sven, 2024-07-22:
### LVGL version

v9.x

### What happened?

Using the current LVGL master or stable, as soon as the double buffering and direct render mode is on, the bar animation is glitchy 

https://github.com/user-attachments/assets/fbb248ae-5cf2-4454-9285-72ef633a9863

I don't know exactly why this happens. [Test Branch](https://github.com/mr-sven/esp32-8048S050C/tree/lvgl-test)

### How to reproduce?

```c
static void lvgl_disp_flush(lv_display_t * disp, const lv_area_t * area, uint8_t * px_map)
{
    esp_lcd_panel_handle_t panel_handle = (esp_lcd_panel_handle_t)lv_display_get_user_data(disp);
    esp_lcd_panel_draw_bitmap(panel_handle, area->x1, area->y1, area->x2 + 1, area->y2 + 1, px_map);
    lv_display_flush_ready(disp);
}

void lvgl_lcd_init(lv_display_t *disp)
{
    void *buf1 = NULL;
    void *buf2 = NULL;
    //....

    // create lcd panel
    ESP_ERROR_CHECK(esp_lcd_new_rgb_panel(&st7262_panel_config, &panel_handle));
    ESP_ERROR_CHECK(esp_lcd_panel_reset(panel_handle));
    ESP_ERROR_CHECK(esp_lcd_panel_init(panel_handle));

    // assign callback and handle
    lv_display_set_user_data(disp, panel_handle);
    lv_display_set_flush_cb(disp, lvgl_disp_flush);

    buffer_size = LCD_WIDTH * LCD_HEIGHT * 2; // 2 = 16bit color data
    esp_lcd_rgb_panel_get_frame_buffer(panel_handle, 2, &buf1, &buf2);
    lv_display_set_buffers(disp, buf1, buf2, buffer_size, LV_DISPLAY_RENDER_MODE_DIRECT);
}
```

mr-sven, 2024-07-22:
For me it looks like the buffers gets updated while DMA transfer is in progress.

kisvegabor, 2024-07-24:
Is `esp_lcd_panel_draw_bitmap` blocking? If not you should wait its competition to call `lv_display_flush_ready`-

mr-sven, 2024-07-25:
I checked the code, I think it's not blocking. The ESP-IDF checks the pointer if it matches one of the frame buffers and then it informs the DMA to use the next buffer. I tried to use the mutex sample from the ESP-IDF but it's very slow

mr-sven, 2024-08-22:
I checked the source of `esp_lcd_panel_draw_bitmap` -> `rgb_panel_draw_bitmap` and in double FB mode it is not blocking.
The code moves the DMA pointer to the other frame buffer, so it is not waiting for vsync.

The ESP IDF RGB Panel example uses a semaphore to wait for vsync, but its incredibly slow.

I've searched the ESP-BSP package and I extracted the tearing fix and now I use xTaskNotify and ulTaskNotifyTake which is now very fast. I put all in my current test branch https://github.com/mr-sven/esp32-8048S050C/tree/lvgl-test as ESP IDF Component so it can be easily extend to other sunton displays.

paaspaas00, 2024-09-09:
@mr-sven is it fixed now?

mr-sven, 2024-09-09:
I merged now to main so the issue is fixed in https://github.com/mr-sven/esp32-8048S050C.

closed 2024-09-09 by mr-sven