lvgl/lvgl #6547: Two animation in same time

fatihozen3, 2024-07-22:
### Introduce the problem

Hi I use IMXRT 1176 MCU and LVGL v9.1.0 .

I want to move 2 objects in same screen in nearly same time. But first objects flickers when I start to move 2. object. How can I prevent this flickering ?

I use different functions and different animations for different movable objects.

static void v_SCREEN_Window_RightAnimation(uint16_t b_stop_point){
    lv_anim_t a;
    lv_anim_init(&a);
    lv_anim_set_var(&a, object_move_1);
    lv_anim_set_duration(&a, 5000); //moves slower when duration increased
    lv_anim_set_exec_cb(&a, f_Move_Object1);
    lv_anim_set_values(&a, lv_obj_get_y(object_move_1), b_stop_point);
    lv_anim_start(&a);
    }

liamHowatt, 2024-07-31:
Could you send a larger code snippet with the callbacks too, please?

fatiihyildizz, 2024-08-20:
Hello,

Actually we are using timeline for this. Callback function is :
[image]

With the a button event, we simulate two parallel objects together. Event is:
[image]

But after this timeline start, both of them moving but one of them flicker. What could this problem be related to?

liamHowatt, 2024-08-21:
Are you using the same object for both animations by mistake?

This works for me:

```c
static void anim_cb(void * var, int32_t val)
{
    lv_obj_set_y(var, val);
}

static void two_anim(void)
{
    for(int i = 0; i < 2; i++) {
        lv_obj_t * obj = lv_obj_create(lv_screen_active());
        lv_obj_set_style_bg_color(obj, lv_color_black(), 0);

        lv_anim_t a;
        lv_anim_init(&a);
        lv_anim_set_var(&a, obj);
        lv_anim_set_duration(&a, 5000); //moves slower when duration increased
        lv_anim_set_exec_cb(&a, anim_cb);
        lv_anim_set_values(&a, lv_obj_get_y(obj), 1000);
        if(i == 0) lv_anim_set_delay(&a, 500);
        lv_anim_start(&a);
    }
}
```

fatiihyildizz, 2024-08-22:
This is not work for me.I think, this problem is related to our frame buffer and driver mechanism. But i dont found an answer. Because When I want to change the slider value by dragging and dropping while a scrolling text is active, I encounter a flicker on the slider. But if the scrolling text not active, there is no problem for slider. When creating the LVGL structure, I used a double frame buffer and selected DIRECT mode as the rendering method.
[image]
Frame buffer size is equal to screen size. And i dont use a gpu. Could the problem be that I am not getting enough speed? @liamHowatt

fatiihyildizz, 2024-08-22:
I tested and even when the gpu(pxp) is active, current problem doesnt resolved

liamHowatt, 2024-09-09:
Sorry for the delay, fatiihyildizz. It may be a speed issue, unfortunately. I cannot be certain though.

closed 2024-10-03 by lvgl-bot