lvgl/lvgl #6550: lvgl on ARM 64 lv_task_handler () deadlock with alsa and/or SDL_audio

marriusco, 2024-07-22:
### LVGL version

9.1.1.dev

### What happened?

Reading audio stream and trying to update a realtime graph hangs in lv_task_handler ()

Works fine on x86. 
After 3..10 seconds or so hangs on ARM 64.

[image]

### How to reproduce?

```
/*
   errors removed to shorten the code
*/

#include <signal.h>
#include <unistd.h>
#include <stdio.h>
#include <stdlib.h>
#include <arpa/inet.h>
#include <string>
#include <iostream>
#include "lvgl.h"
#include "SDL2/SDL.h" # built form sources.
#include "SDL2/SDL_audio.h"

#define NSAMPLES        512         // all below tweaked for realtime feedback
#define POINTS          NSAMPLES
#define NPOINTS         64
#define AVERAGE         32
#define SAMPLE_RATE     44100
#define FRAMES_PER_BUFFER POINTS
#define NUM_SECONDS     5
#define NUM_CHANNELS    2
#define DITHER_FLAG     0
#define WRITE_TO_FILE   0
#define SAMPLES (NSAMPLES * NUM_CHANNELS)

static lv_obj_t* chart;
lv_chart_series_t* ser;
int II=0;

static void init_ui();
void process(int val);

void audioRecordingCallback( void* userdata, Uint8* stream, int len )
{
    if(len==0 || stream==nullptr)return;
    int v;
    int j=0;
    for(int i=0;i<len;)
    {
        v += (int)(*((uint16_t*)(stream+i))); // get one channel
        i += NUM_CHANNELS * sizeof(uint16_t); //skip second channel
        if(j++>AVERAGE)
        {
            v/=AVERAGE;
            v-=32767;
            if(v<0)v=0;
            j=0;
            process(v);
        }
    }
}

int main (int argc, char *argv[])
{
    int err = SDL_Init( SDL_INIT_AUDIO );
    int i_ndex = 0;
    int gRecordingDeviceCount = SDL_GetNumAudioDevices( SDL_TRUE );
    for( int i = 0; i < gRecordingDeviceCount; ++i )
    {
        std::cout << SDL_GetAudioDeviceName( i, SDL_TRUE ) << "\n";
    }

    SDL_AudioSpec desiredRecordingSpec;
    SDL_AudioSpec gReceivedRecordingSpec;
    SDL_zero(desiredRecordingSpec);
    desiredRecordingSpec.freq = SAMPLE_RATE;
    desiredRecordingSpec.format =AUDIO_U16; // AUDIO_F32MSB;//AUDIO_F32; //
    desiredRecordingSpec.channels = NUM_CHANNELS;
    desiredRecordingSpec.samples = SAMPLES;
    desiredRecordingSpec.callback = audioRecordingCallback;

    int recordingDeviceId = SDL_OpenAudioDevice( SDL_GetAudioDeviceName( i_ndex, SDL_TRUE ),
        SDL_TRUE, &desiredRecordingSpec,
        &gReceivedRecordingSpec,
        0 );

    int bytesPerSample = gReceivedRecordingSpec.channels *
             ( SDL_AUDIO_BITSIZE( gReceivedRecordingSpec.format ) /   8 );
    int bytesPerSecond = gReceivedRecordingSpec.freq * bytesPerSample;
    int bcTime = (desiredRecordingSpec.samples*bytesPerSample*1000)/bytesPerSecond;

    lv_init();
#ifndef  __x86_64__
    // ON IMX8MM HANGS IN main loop lv_task_handler();
    lv_display_t *display =  lv_glfw_window_create(600,400);
#else
    lv_display_t* display = lv_x11_window_create("DISPLAY", 600, 400);
    lv_x11_inputs_create(display, NULL);
#endif

    lv_obj_set_style_bg_color(lv_screen_active(), lv_color_hex(0x00CC00), LV_PART_MAIN);
    init_ui();
    SDL_PauseAudioDevice( recordingDeviceId, SDL_FALSE );
    while (Break==false && display)
    {
       	lv_task_handler();
    }
    SDL_PauseAudioDevice( recordingDeviceId, SDL_TRUE );
    SDL_Quit();
    lv_deinit();
    return (0);
}

void process(int val)
{
    if(ser) // wait for its creation
    {
      ser->y_points[II++]=val;
      if(II>=NPOINTS)
      {
          II=0;
          lv_chart_refresh(chart);
      }
    }
}

void init_ui()
{
    chart = lv_chart_create(lv_screen_active());
    lv_obj_set_size(chart, 600, 400);
    lv_obj_center(chart);
    lv_chart_set_type(chart, LV_CHART_TYPE_LINE);
    lv_obj_align(chart, LV_ALIGN_BOTTOM_MID, 10, 0);

    lv_chart_set_range(chart, LV_CHART_AXIS_PRIMARY_Y, -16384, 16384);
    lv_obj_set_style_pad_column(chart, 0, 0);
    lv_obj_set_style_bg_color(chart, lv_color_hex(0x540404), LV_PART_MAIN);
    lv_chart_set_point_count(chart, NPOINTS+1);
    ser = lv_chart_add_series(chart, lv_palette_main(LV_PALETTE_GREEN), LV_CHART_AXIS_PRIMARY_Y);
}

```
gdb BT

```
0x0000ffffbee9e3f4 in pthread_cond_wait@@GLIBC_2.17 () from /lib/aarch64-linux-gnu/libpthread.so.0
(gdb) br
Breakpoint 1 at 0xffffbee9e3f4
(gdb) bt
#0  0x0000ffffbee9e3f4 in pthread_cond_wait@@GLIBC_2.17 () from /lib/aarch64-linux-gnu/libpthread.so.0
#1  0x0000ffffb8b7ba2c in ?? () from /usr/lib/aarch64-linux-gnu/dri/swrast_dri.so
#2  0x0000ffffb8b8798c in ?? () from /usr/lib/aarch64-linux-gnu/dri/swrast_dri.so
#3  0x0000ffffb8b88454 in ?? () from /usr/lib/aarch64-linux-gnu/dri/swrast_dri.so
#4  0x0000ffffb8fd24ec in ?? () from /usr/lib/aarch64-linux-gnu/dri/swrast_dri.so
#5  0x0000ffffb99f47a4 in ?? () from /lib/aarch64-linux-gnu/libGLX_mesa.so.0
#6  0x0000aaaaaaab091c in window_update_handler ()
#7  0x0000aaaaaaae1d74 in lv_timer_exec ()
#8  0x0000aaaaaaae16e4 in lv_timer_handler ()
#9  0x0000aaaaaaaaedac in lv_task_handler ()
#10 0x0000aaaaaaaaf0bc in main ()
(gdb) 

```

XuNeo, 2024-07-23:
Is `process` called from different thread?

marriusco, 2024-07-23:
yes is called from a different thread. Ill address that

marriusco, 2024-07-23:
Resolved. Thank you

closed 2024-07-23 by marriusco