lvgl/lvgl #6551: Wrong property ID returned from LV_PROPERTY_ID_INDEX() for style properties.

sliang951753, 2024-07-23:
### LVGL version

v9.0.0

### What happened?

The macro #define LV_PROPERTY_ID(clz, name, type, index)    LV_PROPERTY_## clz ##_##name = (LV_PROPERTY_## clz ##_START + (index)) | ((type) << LV_PROPERTY_TYPE_SHIFT)
defines a property ID. But it does not store the real "index", it adds the property id start value to the index.
For example, the index of LV_STYLE_X is  8,  however the LV_PROPERTY_STYLE_X stores LV_STYLE_X + LV_PROPERTY_STYLE_START = 9. The problem will occur when set this property by using lv_obj_set_local_style_prop(), the second argument is the index of the style property.  
I think the code should be 
lv_obj_set_local_style_prop(obj, index - LV_PROPERTY_STYLE_START, value->style, value->selector);
instead of 
lv_obj_set_local_style_prop(obj, index, value->style, value->selector);

Anyone can help to check this ?

XuNeo, 2024-07-23:
I think I have covered the case here.

https://github.com/lvgl/lvgl/blob/daa7fefb3a4bd058f1c5b5166871085876d0ada4/tests/src/test_cases/widgets/test_obj_property.c#L38-L44

The index value for style property, is the value from `enum _lv_style_prop_t`

See lv_style_properties.h

```c
enum {
    LV_PROPERTY_ID(STYLE, ALIGN,                    LV_PROPERTY_TYPE_INT,        LV_STYLE_ALIGN),
    LV_PROPERTY_ID(STYLE, ANIM,                     LV_PROPERTY_TYPE_INT,        LV_STYLE_ANIM),
...
```

XuNeo, 2024-07-23:
Sorry. After check the test code, the prop.id is wrong.

You are right. The style enum is offset by 1.

sliang951753, 2024-07-23:
> Sorry. After check the test code, the prop.id is wrong.
> 
> You are right. The style enum is offset by 1.

prop.id = LV_STYLE_X;  // this works. 
prop.id = LV_PROPERTY_STYLE_X; // this doesn't work.

It looks your test case is not correct.

the STYLE property already starts from 1, so when making the PROPERTY_STYLE_X, we don't need to add the STYLE PROPERTY START ID.  

If we need to add the STYLE PROPERTY START ID to identify it is a style property, we need to substract the STYLE PROPERTY START ID when using it. 

Since I'm not familiar with the style subsystem design, could you help to check which solution is the best for this case ?

XuNeo, 2024-07-23:
I drafted a fix, could you please check if it works for you?

sliang951753, 2024-07-23:
> I drafted a fix, could you please check if it works for you?

It works. Thank you.

PR #6552 (2024-07-26): fix(property): fix style property

### Description of the feature or fix

The test case is out of date, `prop.id` should not use LV_STYLE_xxx enum directly. 
Set builtin last property ID to 0xffff, leaving 0x10000 to 0xfffffff for user.

Fix #6551

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https [... cut]
closed 2024-07-26 by XuNeo