lvgl/lvgl #6554: tick_get_cb variable  overflow

anheiqq, 2024-07-23:
### LVGL version

v9.1.0

### What happened?

static uint32_t tick_get_cb(void)
{
    struct timeval tv_now;
    gettimeofday(&tv_now, NULL);
    uint64_t time_ms;
    time_ms = (tv_now.tv_sec * 1000000 + tv_now.tv_usec) / 1000;
    return time_ms;
}

The time_ms variable is defined as uint64_t, which can hold a much larger range of values compared to uint32_t. However, casting time_ms to uint32_t before returning can lead to an overflow if the system has been running for more than UINT32_MAX milliseconds.

### How to reproduce?

labe: LV_LABEL_LONG_SCROLL_CIRCULAR
The scrolling effect will become ineffective after running for a period of time。

I changed it to the following code：
uint32_t tick_get_cb(void)
{
    static struct timeval tv_start = {0};
    if(tv_start.tv_sec == 0 && tv_start.tv_usec == 0) {
        gettimeofday(&tv_start, NULL);
    }
    struct timeval tv_now;
    gettimeofday(&tv_now, NULL);
    uint64_t time_ms = (uint64_t)(tv_now.tv_sec - tv_start.tv_sec) * 1000 + (tv_now.tv_usec - tv_start.tv_usec) / 1000;
    return (uint32_t)time_ms;
}

anheiqq, 2024-07-25:
I found that the latest code has been modified to：
static uint32_t tick_get_cb(void)
{
    struct timespec t;
    clock_gettime(CLOCK_MONOTONIC, &t);
    uint64_t time_ms = t.tv_sec * 1000 + (t.tv_nsec / 1000000);
    return time_ms;
}
This is better than my code。

Is the following modification better?

static uint32_t tick_get_cb(void)
{
    struct timespec t;
    clock_gettime(CLOCK_MONOTONIC, &t);
    uint64_t time_ms = t.tv_sec * 1000 + (t.tv_nsec / 1000000);
    return (uint32_t)(time_ms & 0xFFFFFFFF); // 
}

liamHowatt, 2024-07-31:
> Is the following modification better?

That has the same effect as the current one. Casting from `uint64_t` to `uint32_t` implicitly truncates.

After how long approximately does the scrolling effect fail? Is it much sooner than 50 days? The `uint32_t` will overflow after ~50 days of milliseconds. If it fails sooner, then there might be something wrong with this expression: `uint64_t time_ms = t.tv_sec * 1000 + (t.tv_nsec / 1000000);`

Is your system 32 or 64 bits? If you're on a 32 bit system (i.e. a 32 bit Raspberry Pi), then `t.tv_sec` (of type `time_t`) is an `int32_t` and the arithmetic of `t.tv_sec * 1000` is performed as `int32_t`. Similar problem for `t.tv_nsec` which is `long`, which is 32 bits on 32 bit Linux.

We could add explicit casts to the expression and/or express the literals as `1000llu`.

anheiqq, 2024-08-05:
my system is 32bits.

static uint32_t tick_get_cb(void)
{
struct timeval tv_now;
gettimeofday(&tv_now, NULL);
uint64_t time_ms;
time_ms = (tv_now.tv_sec * 1000000 + tv_now.tv_usec) / 1000;
return time_ms;
}
Using this code, the scrolling effect will approximately fail after one to two hours.

static uint32_t tick_get_cb(void)
{
struct timespec t;
clock_gettime(CLOCK_MONOTONIC, &t);
uint64_t time_ms = t.tv_sec * 1000 + (t.tv_nsec / 1000000);
return time_ms;
}
The latest code for the scrolling effect shows no issues so far, but long-term testing hasn't been done yet.
It's already good to go for our needs.

closed 2024-08-07 by anheiqq