lvgl/lvgl #6556: Suggestions About to Pass Monkey Test

fatihozen3, 2024-07-23:
### Introduce the problem

Hi I use IMXRT1176 with LVGL v9.1.0 and FreeRTOS v10.5.1 . 

I have several pages includes images and texts and some events to change page etc.

I have used monkey test and I failed. I see UsageFault Handler sometimes, and sometimes MemManage Handler. What may cause this problem ?

I declared screen objects global as below:

`lv_obj_t *screen_1 = NULL;`

I declared another objects inside the screen as below, they are static for every screen:

`static lv_obj_t *img_background = NULL;`
`
#define LV_MEM_SIZE (350 * 1024U) ` for now.

I opened sys mon and mem mon watch memory usage and there is no overflow as far as I see:

```
#define LV_USE_SYSMON   1
#define LV_USE_MEM_MONITOR 1
```

I suspect screen change function, I am not sure about it:

```
    lv_indev_wait_release(lv_indev_get_act());
    lv_obj_clean(lv_scr_act());
    Screen_Main_Init();
    lv_screen_load_anim(screen_1, LV_SCR_LOAD_ANIM_NONE, 0, 0, true);
```

fatihozen3, 2024-07-25:
I have to give info about sth:

I change screens as below:
```

    lv_indev_wait_release(lv_indev_get_act());
    Screen_Main_Init();
    lv_screen_load_anim(screen_1, LV_SCR_LOAD_ANIM_NONE, 0, 0, true);
```
    
    
    and in another task of LVGL I want to update some parameters of active screen.
    
    I am trying to do that as below condition:
    
   ` if ((lv_obj_is_valid(screen_1) == true) && (lv_scr_act() == screen_1)) `
   
   But there is a fault on this line. screen load anim function does not delete the previous screen I guess. because when I am on screen 1, program enters screen 2 update function. So the point is either previous screen deletion or lv_obj_is_valid function does not work as expected.

liamHowatt, 2024-07-31:
Similar to #6571

`lv_obj_is_valid` can report an object as valid even after it has been deleted. It is not a reliable way to detect whether an object is still alive or not.

Can you try adding an event handler for `LV_EVENT_DELETE` and update a variable to keep track of the screen's deletion so you can detect the deletion more accurately? Also see #6599 which should be used for this type of situation.

closed 2024-08-26 by lvgl-bot