lvgl/lvgl #6559: lv_img_set_angle doesn't work in v9

codegrue, 2024-07-23:
### LVGL version

v9.1.0

### What happened?

After upgrading to v9, ```lv_img_set_angle``` no longer has an effect on images.

### How to reproduce?

```
  lv_img_set_angle(ui_Image, 450);

  auto r = lv_img_get_angle(ui_Image);
  printf("Image angle = %d", r); // returns zero
```

liamHowatt, 2024-07-31:
Are you using `LV_IMAGE_ALIGN_STRETCH` or `LV_IMAGE_ALIGN_TILE` alignment on the image? Looking at the source code, images with that alignment can't be rotated.

codegrue, 2024-08-05:
I am not using those, but apparently rotations have changed to be done with transforms now, so the new code looks like this:

'''
lv_obj_set_style_transform_rotation(ui_Image, 450, LV_PART_MAIN | LV_STATE_DEFAULT);
'''

closed 2024-08-05 by codegrue

liamHowatt, 2024-08-06:
Okay, nice! It is strange that `lv_img_get_angle` didn't return the new angle, though.

kisvegabor, 2024-08-12:
Note that `lv_obj_set_style_transform_rotation` is slower and uses a lot of memory as it first creates a snapshot about the widget (image in this case) and rotates the snapshot as an image.