lvgl/lvgl #6564: My display rotate continuously 90-180-270-360-90 so on

vishvanathdoijad, 2024-07-24:
### Introduce the problem

I want to rotate my display as a portrait  view
My settings are,	
cpu0_globals->gui.disp_drv.sw_rotate = pdTRUE;
cpu0_globals->gui.disp_drv.rotated = 1;
cpu0_globals->gui.disp_drv.full_refresh = pdFALSE;				// full_refrech not work when sw_rotate = pdTRUE
cpu0_globals->gui.disp_drv.direct_mode = pdTRUE;
cpu0_globals->gui.disp_drv.flush_cb = my_flush_cb;

But my display contineously rotate in all direction
Please help.

vishvanathdoijad, 2024-07-24:
/*
 * vga.c
 *
 *  Created on: 1 Jun 2018
 *      Author: peterb
 */

#include "vga.h"

#include "xscugic.h"
#include "support.h"
#include "gui.h"
#include "../../lvgl/src/misc/lv_color.h"
struct vga_prams 					*vga_data;
extern XScuGic 						xInterruptController;
extern cpu0_globals_t				*cpu0_globals;
struct vga_creg_map					*vga = (void*)VGA_CTRL_BASE;

static uint64_t get_nco_value( unsigned int px_clk );

struct vga_prams vga_table[] = {
//tot_h_px, h_px, h_syn_len, h_fpch, h_bpch, h_pol,    tot_v_ln,v_ln, v_syn_len, v_fpch, v_bpch, v_pol, 	px_clk
  {800,     640,  96,  		 16, 	 48,  	SYNC_NEG, 525, 		480, 	2,  	 10, 	 33, 	 SYNC_NEG, 25175000 },	// 640 x 480 @ 60 Hz Standard VGA
  {1056,    800,  128, 		 40, 	 88,  	SYNC_POS, 628, 		600, 	4,  	 1,  	 23, 	 SYNC_POS, 40000000 },	// 800 x 600 @ 60 Hz
  {1344,    1024, 136, 		 24, 	 160, 	SYNC_NEG, 806, 		768, 	6,  	 3,  	 29, 	 SYNC_NEG, 65000000 }, 	// 1024 x 768 @ 60 Hz
  {1328,    1024, 136, 		 24, 	 144, 	SYNC_NEG, 806, 		768, 	6,  	 3,  	 29, 	 SYNC_NEG, 75000000 }, 	// 1024 x 768 @ 70 Hz
  {1904,    1440, 152, 		 80, 	 232, 	SYNC_NEG, 934, 		900, 	6,  	 3,  	 25,	 SYNC_POS, 106500000},	// 1440 x 900 @ 60 Hz CVT Standard
  {1600,    1440, 32,  		 48, 	 80,  	SYNC_NEG, 926, 		900, 	6,  	 3,  	 17,	 SYNC_POS, 88750000 },	// *1440 x 900 @ 60 Hz CVT Reduced Blanking
  {1600,    1440, 32,  		 48, 	 80,  	SYNC_NEG, 942, 		900, 	6,  	 3,  	 33,	 SYNC_POS, 136750000},	// 1440 x 900 @ 75 Hz CVT Standard
  {1600,    1440, 32,  		 48, 	 80,  	SYNC_NEG, 948, 		900, 	6,  	 3,  	 39,	 SYNC_POS, 157000000},	// 1440 x 900 @ 85 Hz CVT Standard
  {1600,    1440, 32,		 48, 	 80,  	SYNC_NEG, 953, 		900, 	6,  	 3,  	 44,	 SYNC_POS, 182750000},	// 1440 x 900 @ 120 Hz CVT Reduced Blanking
  {1344,    1024, 136,		 24, 	 160,  	SYNC_NEG, 628, 		600, 	4,  	 1,  	 23,	 SYNC_POS, 40000000}	// 1024 x 600 @ 50 Hz CVT Reduced Blanking 40000000
};

void vga_irq_handler( void *p ) {

	vga->vga_fbuf_addr = cpu0_globals->gui.dma_src;
	cpu0_globals->gui.buf_switched = pdTRUE;
	XScuGic_Disable(&xInterruptController, XPAR_FABRIC_ZEDBOARD_AXI_VGA_0_FRM_CPT_IRQ_INTR);
	lv_disp_flush_ready((lv_disp_drv_t*)&cpu0_globals->gui.disp_drv);
}

void set_vga_prams( uint8_t table_idx ) {

	uint64_t	nco_val;
	int			Status;

	vga_data = &vga_table[table_idx];
	vga->h_ctrl1 = SET_THP( vga_data->tot_h_px );
	vga->h_ctrl1 |= SET_HP( vga_data->h_px );
	vga->h_ctrl1 |= SET_HS( vga_data->h_sync_len );
	vga->h_ctrl2 = SET_HFP( vga_data->h_fporch );
	vga->h_ctrl2 |= SET_BP( vga_data->h_bporch );
	vga->h_ctrl2 |= SET_POL( vga_data->h_pol );
	vga->v_ctrl1 = SET_TVL( vga_data->tot_v_ln );
	vga->v_ctrl1 |= SET_VL( vga_data->v_ln );
	vga->v_ctrl1 |= SET_VS( vga_data->v_sync_len );
	vga->v_ctrl2 = SET_VFP( vga_data->v_fporch );
	vga->v_ctrl2 |= SET_BP( vga_data->v_bporch );
	vga->v_ctrl2 |= SET_POL( vga_data->v_pol );
	nco_val = get_nco_value( vga_data->px_clk );
	vga->px_nco_lsbs = nco_val & 0xFFFFFFFF;
	vga->px_nco_msbs = ((nco_val & 0xFF00000000) >> 32);
	vga->vga_fbuf_addr = VGA_DDR_DMA_BASE;
	vga->total_pixels = (vga_data->h_px * vga_data->v_ln) | DMA_FIFO_RST;
	/*
	 * Connect the interrupt handler that will be called when an
	 * interrupt occurs for the device.
	 */
	Status = XScuGic_Connect(&xInterruptController, XPAR_FABRIC_ZEDBOARD_AXI_VGA_0_FRM_CPT_IRQ_INTR, \
			(Xil_ExceptionHandler)&vga_irq_handler, NULL);
	if (Status != XST_SUCCESS) {
		// TODO: Some error report...
	}
	XScuGic_SetPriorityTriggerType(&xInterruptController, XPAR_FABRIC_ZEDBOARD_AXI_VGA_0_FRM_CPT_IRQ_INTR, \
			0xB8, 0x3);
	return;
}

static void update_dual_buf( lv_disp_drv_t *disp_drv, const lv_area_t *area, lv_color_t *colour_p ) {

	lv_disp_t*	disp = _lv_refr_get_disp_refreshing();
	lv_coord_t	y, hres = lv_disp_get_hor_res(disp);
    uint16_t	a;
    lv_color_t	*buf_cpy;

    if( colour_p == disp_drv->draw_buf->buf1)
    {

        buf_cpy = disp_drv->draw_buf->buf2;
        xil_printf("cpy_buf2\n\r");
    }
    else
    {
        buf_cpy = disp_drv->draw_buf->buf1;
        xil_printf("cpy_buf1\n\r");
    }
     for(a = 0; a < disp->inv_p; a++)
     {
    	if(disp->inv_area_joined[a]) continue;
        lv_coord_t w = lv_area_get_width(&disp->inv_areas[a]);
        xil_printf("area get width %d \n\r",w);
        for(y = disp->inv_areas[a].y1; y <= disp->inv_areas[a].y2 && y < disp_drv->ver_res; y++)
        {
            memcpy(buf_cpy+(y * hres + disp->inv_areas[a].x1), colour_p+(y * hres + disp->inv_areas[a].x1), w * sizeof(lv_color_t));
        }
        xil_printf("mem cpy completed\n\r");
    }
}

void vga_disp_flush(lv_disp_drv_t *disp_drv, const lv_area_t *area, lv_color_t *colour_p) {

	static uint8_t	first_call = 1;

	if( first_call ) {
		cpu0_globals->gui.dma_src = (uint32_t)colour_p;
		first_call =  0;
		vga->total_pixels &= ~DMA_FIFO_RST; 	// Release Reset
		vga->total_pixels |= DMA_FRAME_READY;	// Start Proceedings
		vga->irq_reg = VGA_IRQ_EN;				// Enable DMA complete interrupt
		xil_printf("First call\n\r");
	}
	if( lv_disp_flush_is_last( disp_drv ) ) {
		cpu0_globals->gui.dma_src = (uint32_t)colour_p;
		cpu0_globals->gui.buf_switched = pdFALSE;
		XScuGic_Enable(&xInterruptController, XPAR_FABRIC_ZEDBOARD_AXI_VGA_0_FRM_CPT_IRQ_INTR);
		while(!cpu0_globals->gui.buf_switched);	// vTaskDelay(1);
		update_dual_buf(disp_drv, area, colour_p);
	    lv_obj_invalidate(lv_scr_act());
		xil_printf("Last call \t Call updated dual buf function\n\r");
	} else
	{
		lv_disp_flush_ready( disp_drv );
		xil_printf("Flush ready for next frame\n\r");
	}
}

#define NCO_2POW36 68719476736LL
#define NCO_CLK 200000000L
uint64_t get_nco_value( unsigned int px_clk ) {

	uint64_t	nco;

	nco = (uint64_t)(((double)px_clk / (double)NCO_CLK) * (double)NCO_2POW36);
	return nco;
	/*  NCO36 configuraton
	    n = (b / c) * 2^36
	    b = (n / 2^36) * c

	    Where n = cntrl value.
	          b = desired clock rate
	          c = system clock rate
	*/
}

kisvegabor, 2024-07-24:
It seems like a tricky issue, so sorry for not responding directly at first. Is it an option for you to update to LVGL v9? Screen rotation is handled better in this new version.

vishvanathdoijad, 2024-07-25:
Thanx @kisvegabor for quick reply,
1.   I am using Squareline Stdio 1.4.1 for development, As per blog "https://blog.lvgl.io/" LVGL ends its collaboration with SquareLine Studio. So far Which tool you suggest for GUI development which support LVGL v9...?
2.   Please provide me link to download LVGL v9 library.
3.   Please provide me link to download PDF version of document.

kisvegabor, 2024-07-25:
> I am using Squareline Stdio 1.4.1 for development, As per blog "https://blog.lvgl.io/" LVGL ends its collaboration with SquareLine Studio. So far Which tool you suggest for GUI development which support LVGL v9...?

Despite that we are not collaborating actively you can still use SquareLine Studio. A free an open source option is EEZ Studio. 

> Please provide me link to download LVGL v9 library.

I recommend cloning LVGL from GitHub, using the `master` branch.

> Please provide me link to download PDF version of document.

Unfortunately it's not available right now, due to some issues in our docs build process. :slightly_frowning_face: 
However I was always curious why people need the pdf docs. Can you tell why do you prefer it over the online version?

vishvanathdoijad, 2024-07-25:
**> However` I was always curious why people need the pdf docs. Can you tell why do you prefer it over the online version?**
Ans- PDFs can be accessed offline, which is handy when you don't have a stable internet connection (I am facing internet issues). And PDF allows for annotations, highlighting (I preferred), and comments directly on the document, which can be useful for studying, reviewing.

"I have integrated LVGL v9.1 into my project following the documentation. Initially, I encountered path errors similar to those in \lvgl\tests\test_images\xxx\xxxx_align64.c, as shown in the first image. To resolve these, I updated the paths, for example, changing #include "lvgl/lvgl.h" to #include "../../../../lvgl.h", which resolved all path-related errors.

However, I'm still encountering build errors, as shown in the second image."

[image]
[image]

kisvegabor, 2024-07-29:
Thank you, noted! 

@kdschlosser do you know something about the status of PDF docs build?

vishvanathdoijad, 2024-07-29:
Hi @kisvegabor 
Please resolve my Issue, Waiting for your suggestions.

vishvanathdoijad, 2024-07-29:
@kisvegabor 
Can you please portrait this LVGL sponsored project..?
https://github.com/lvgl/lv_port_xilinx_zedboard_vitis

kdschlosser, 2024-07-29:
the unicode causes the latex build to fail I have not had the chance to dive into it to straighten out the unicode errors. It has something to do with the font that is used by the latex build and the inability to change it. I don't remember 100% what it was but it is something along those lines.

vishvanathdoijad, 2024-07-30:
@kdschlosser @kisvegabor @kumekay @gmomchilov

Can you please rotate 90/portrait this LVGL sponsored project..?
https://github.com/lvgl/lv_port_xilinx_zedboard_vitis

kdschlosser, 2024-07-30:
@vishvanathdoijad 

You are running into compilation errors because you have 2 folders that are "src" under each other and you have not added the lvgl folder to the includes when calling the compiler. If you are launching the test program using the relative path to lvgl from your project folder the test is going to be adding `.` to the include paths which is going to be the wrong directory.

vishvanathdoijad, 2024-07-31:
Thanx @kdschlosser 
I will heck and back to you

vishvanathdoijad, 2024-07-31:
[image]

vishvanathdoijad, 2024-07-31:
When I replace the folder  lvgl (v9) to lvgl (v8.3) it works.

kisvegabor, 2024-08-01:
I see, thank you @kdschlosser.

Probably it's because the CJK and Arabic characters in the examples.

vishvanathdoijad, 2024-08-01:
@kdschlosser 
What should be the workaround to mitigate this problem?"

kdschlosser, 2024-08-01:
I have no idea.. If you plave version 8's code into the same src path it works but if you place version 9 it fails.

I would need to see the exact output of the failure. This is before you make any changes to any of the files or include paths.

kdschlosser, 2024-08-01:
from what I am seeing it sounds like you need to pass `-I ./src` to the compiler in order for it to work properly. It is hard to say without seeing all of the output. 

sphinx and latex is a fickle beast when it comes to unicode characters. The default setup works fine with Greek and Latin characters but not Asian character sets. This is where it becomes hard because the documentation has characters that are used from different character sets and there is no "one size fits all" font for these use cases. That causes the build to fail.

vishvanathdoijad, 2024-08-05:
> @kdschlosser @kisvegabor @kumekay @gmomchilov
> 
> Can you please rotate 90/portrait this LVGL sponsored project..? https://github.com/lvgl/lv_port_xilinx_zedboard_vitis

1.   Could you please rotate the LVGL (8.3) sponsored project to 90 degrees (portrait mode)? The project is available at: https://github.com/lvgl/lv_port_xilinx_zedboard_vitis.
2.   Alternatively, could you update the project for LVGL 9, where rotation is simpler (Tested on Code block https://github.com/lvgl/lv_port_win_codeblocks/tree/release/v9.0)? I’ve tried to update it myself, but I’m encountering a makefile error. Please find the attached scree shot.

[image]

@pete-pjb

vishvanathdoijad, 2024-08-08:
> **> However` I was always curious why people need the pdf docs. Can you tell why do you prefer it over the online version?** Ans- PDFs can be accessed offline, which is handy when you don't have a stable internet connection (I am facing internet issues). And PDF allows for annotations, highlighting (I preferred), and comments directly on the document, which can be useful for studying, reviewing.
> 
> "I have integrated LVGL v9.1 into my project following the documentation. Initially, I encountered path errors similar to those in \lvgl\tests\test_images\xxx\xxxx_align64.c, as shown in the first image. To resolve these, I updated the paths, for example, changing #include "lvgl/lvgl.h" to #include "../../../../lvgl.h", which resolved all path-related errors.
> 
> However, I'm still encountering build errors, as shown in the second image."
> 
> ![1](https://private-user-images.githubusercontent.com/45351039/352113218-37c39c8c-8ce3-486a-82ab-4b23c4f8ab5c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjMwOTUwMzEsIm5iZiI6MTcyMzA5NDczMSwicGF0aCI6Ii80NTM1MTAzOS8zNTIxMTMyMTgtMzdjMzljOGMtOGNlMy00ODZhLTgyYWItNGIyM2M0ZjhhYjVjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODA4VDA1MjUzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU3ZDM1ZDFiNWU3ZmExZmM3YmFhMmQ4M2YzOTk5ODZlOTk1ZmEzZTE0NzJiNTEzNGZiYjRmNjgyYTUyZTc3M2UmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.87N5Rl4lDZxkyWMiU3NwX4QtEbEMsCbV-NMuP7RIJuo) ![2](https://private-user-images.githubusercontent.com/45351039/352113302-ebce8ccb-44c9-4ae6-bc94-3f236ca5a492.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjMwOTUwMzEsIm5iZiI6MTcyMzA5NDczMSwicGF0aCI6Ii80NTM1MTAzOS8zNTIxMTMzMDItZWJjZThjY2ItNDRjOS00YWU2LWJjOTQtM2YyMzZjYTVhNDkyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODA4VDA1MjUzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk4OGY0NWNjODQzYjgzMDJiNmQwNWY1MGI1YmM2NGJhMDhlOWEwNzdmMzBkNGM4ZjhkMGIyOGVjYTI0MDk4OWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.F4uGJKIJuAKwMHK1K99DceKelKGIXflokNPnrI-ZQYs)

> **> However` I was always curious why people need the pdf docs. Can you tell why do you prefer it over the online version?** Ans- PDFs can be accessed offline, which is handy when you don't have a stable internet connection (I am facing internet issues). And PDF allows for annotations, highlighting (I preferred), and comments directly on the document, which can be useful for studying, reviewing.
> 
> "I have integrated LVGL v9.1 into my project following the documentation. Initially, I encountered path errors similar to those in \lvgl\tests\test_images\xxx\xxxx_align64.c, as shown in the first image. To resolve these, I updated the paths, for example, changing #include "lvgl/lvgl.h" to #include "../../../../lvgl.h", which resolved all path-related errors.
> 
> However, I'm still encountering build errors, as shown in the second image."
> 
> ![1](https://private-user-images.githubusercontent.com/45351039/352113218-37c39c8c-8ce3-486a-82ab-4b23c4f8ab5c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjMwOTk3MDYsIm5iZiI6MTcyMzA5OTQwNiwicGF0aCI6Ii80NTM1MTAzOS8zNTIxMTMyMTgtMzdjMzljOGMtOGNlMy00ODZhLTgyYWItNGIyM2M0ZjhhYjVjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODA4VDA2NDMyNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU1ZjYzMTUyMjZmNTNjZmVmMTdiNTNlM2I4NWJjMDNiZDI0NGNkYzMwMzk5Yjc1Nzc0YWRjNWRmYmU5YmM1YWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.eU65TDkmJJXTEfHtF6VrGwpXkG44IRqeNf4ZUcMOYyQ) ![2](https://private-user-images.githubusercontent.com/45351039/352113302-ebce8ccb-44c9-4ae6-bc94-3f236ca5a492.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjMwOTk3MDYsIm5iZiI6MTcyMzA5OTQwNiwicGF0aCI6Ii80NTM1MTAzOS8zNTIxMTMzMDItZWJjZThjY2ItNDRjOS00YWU2LWJjOTQtM2YyMzZjYTVhNDkyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODA4VDA2NDMyNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk4YTgyMTA2NGE5MmFmNmEwMGEyYWM2MDNiMzg4YmFhMjc0ZmZlNjNhZTc5OGIxMjdiNTBkZDViOWRmNmJiNGMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.mIEaU8CriHc5Vts8W3FAb4mD8xS23yzzpYFo_TB8KtQ)

LVGL 9.1.1 Library is compiled in LINUX machine, Still not in windows (Vitis IDE)

vishvanathdoijad, 2024-08-08:
> LVGL 9.1.1 Library is compiled in LINUX machine, Still not in windows (Vitis IDE)

/

vishvanathdoijad, 2024-08-09:
> LVGL 9.1.1 Library is compiled in LINUX machine, Still not in windows (Vitis IDE)

Now I am able to compile LVGL 9.1.1 on Linux machine (VItis IDE)

My question is, for 90 rotation in LVGL 9.1.1 does we required to set the "sw_rotate" which is define in "lvgl/scr/display/lv_display_private.h" along with " lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_90)"?
If Yes, Please tell me the steps to set that bit in main code.

vishvanathdoijad, 2024-08-09:
"Below is my Vitis IDE code for LVGL 9.1.1. Please review it. I also need suggestions for buffer switching, as I am using double buffering in the final code."
1. Landscape mode working cool
2. getting garbage on screen when use 90 degree rotation
[image]

[image]

/*
 * vdmaTest.c
 *
 *  Created on: Apr 9, 2020
 *      Author: VIPIN
 */
#include "xparameters.h"
#include "xaxivdma.h"
#include "xscugic.h"
#include "sleep.h"
#include <stdlib.h>
#include "xil_cache.h"
#include "xil_cache.h"
#include "xvtc.h"
#include "lvgl/lvgl.h"
#include "xscutimer.h"

#include "lvgl/demos/lv_demos.h"
#include "lvgl/src/display/lv_display.h"
#include "lvgl/src/display/lv_display_private.h"
#include "lvgl/examples/lv_examples.h"
#include "lvgl/demos/benchmark/lv_demo_benchmark.h"
#include "lvgl/src/drivers/display/fb/lv_linux_fbdev.h"

#define time_load_value 333334

#define HSize 1024
#define VSize 600
#define FrameSize (HSize * VSize * (LV_COLOR_DEPTH/8))

static XScuGic Intc;

static int SetupIntrSystem(XAxiVdma *AxiVdmaPtr, u16 ReadIntrId);
uint8_t Buffer[1024*600*3];
//lv_color_t *buf_1 = (lv_color_t *) Buffer;  // First buffer
//lv_color_t *buf_2 = NULL;  // Second buffer, or NULL if not used
uint8_t buf1[FrameSize];
uint8_t buf2[FrameSize];
uint8_t temp_buffer[FrameSize];

uint8_t *current_buffer = temp_buffer;

void fill_screen(uint32_t color);

static u8 next_fram_is_ready_to_display = 0;

int buffer_flushing_completed;

uint32_t hsv_to_rgb(float h, float s, float v) {
    float r, g, b;
    int i = (int)(h * 6);
    float f = h * 6 - i;
    float p = v * (1 - s);
    float q = v * (1 - f * s);
    float t = v * (1 - (1 - f) * s);

    switch (i % 6) {
        case 0: r = v, g = t, b = p; break;
        case 1: r = q, g = v, b = p; break;
        case 2: r = p, g = v, b = t; break;
        case 3: r = p, g = q, b = v; break;
        case 4: r = t, g = p, b = v; break;
        case 5: r = v, g = p, b = q; break;
[... 342 lines cut ...]

	Status = XScuGic_Connect(IntcInstancePtr,XPS_SCU_TMR_INT_ID,(Xil_InterruptHandler)Timer_ISR,(void *)&Scu_timer);
	if (Status != XST_SUCCESS) {
		xil_printf("Failed to connect timer intc %d\r\n", Status);
		return XST_FAILURE;
	}

	XScuGic_Enable(IntcInstancePtr,ReadIntrId);

	XScuGic_Enable(IntcInstancePtr,XPS_SCU_TMR_INT_ID);

	XScuTimer_EnableInterrupt(&Scu_timer);

	Xil_ExceptionInit();
	Xil_ExceptionRegisterHandler(XIL_EXCEPTION_ID_INT,(Xil_ExceptionHandler)XScuGic_InterruptHandler,(void *)IntcInstancePtr);
	Xil_ExceptionEnable();

	/* Register call-back functions
	 */
	XAxiVdma_SetCallBack(AxiVdmaPtr, XAXIVDMA_HANDLER_GENERAL, ReadCallBack, (void *)AxiVdmaPtr, XAXIVDMA_READ);

	XAxiVdma_SetCallBack(AxiVdmaPtr, XAXIVDMA_HANDLER_ERROR, ReadErrorCallBack, (void *)AxiVdmaPtr, XAXIVDMA_READ);

	return XST_SUCCESS;
}