lvgl/lvgl #6565: Runtime theme switching

Kernald, 2024-07-25:
### Problem to solve

Having a way to change the theme at runtime, without re-creating objects. This would allow for e.g. a light and a dark variant of a theme, swapped at runtime depending on ambient luminosity/time of day/...

### Success criteria

Having a way to change theme at runtime

### Solution outline

#395 had a solution in lvgl 6.5, but the solution implemented back then seems to have disappeared at some point.

### Rabbit holes

n/a

### Testing

Unsure - I'm assuming this could be tested fairly similarly to how themes are tested today.

### Teaching

https://docs.lvgl.io/master/overview/style.html#themes seems like a pretty straightforward place to mention it.

Kernald, 2024-08-09:
This is still relevant.

kisvegabor, 2024-08-17:
We have really sacrificed run time theme switching for more flexibility and simplicity.

IMO the best solution right now is to use observers, as shown [here](https://docs.lvgl.io/master/others/observer.html#modular-style-update-on-theme-change). With this patter you have much more freedom to change anything on theme switching.

closed 2024-09-11 by lvgl-bot