lvgl/lvgl #6570: Schedule for release 9,2

cristian-stoica, 2024-07-25:
### Introduce the problem

@kisvegabor When is the release 9.2 scheduled?
If the release is not imminent we would like to include several recent patches we've made if there is enough time for review.
Thank you

kisvegabor, 2024-07-25:
I'd like to merge only one more PR: https://github.com/lvgl/lvgl/pull/6068

Can you send the patches next week?

kisvegabor, 2024-07-26:
Once #6068 is merged I'd like to have 1 week of feature freeze. After that we are good to go! 

cc @XuNeo @FASTSHIFT @liamHowatt

cristian-stoica, 2024-07-26:
yes - thank you

kisvegabor, 2024-08-13:
Let's really set up a deadline otherwise we will never have a release 😀

Let's merge/add what we can this week and have a feature stop next week. What do you think?

kisvegabor, 2024-08-14:
cc @XuNeo @FASTSHIFT @liamHowatt

vwheeler63, 2024-08-14:
If you don't mind me asking -- is it okay to get the doc-builds fixes into this release?

kisvegabor, 2024-08-15:
@vwheeler63 It would be great, but I need to take a deeper look at its current status.

vwheeler63, 2024-08-15:
@kisvegabor Makes sense.

kisvegabor, 2024-08-21:
Yes, we will merge what we can this week, and make release on Monday (August 26)

kisvegabor, 2024-08-26:
@GorgonMeducer could you prepare a PR to update the CMSIS packs for the release?

GorgonMeducer, 2024-08-26:
No problem. @kisvegabor

kisvegabor, 2024-08-29:
v9.2 is released so closing this issue.

closed 2024-08-29 by kisvegabor