lvgl/lvgl #6571: lv_obj_is_valid() and lv_scr_act() doesnt work

fatiihyildizz, 2024-07-25:
### LVGL version

v9.0.1 / master

### What happened?

Hello,
I have a touchscreen project with multiple displays. I enable the transition between screens with events on the buttons. In the relevant event, I use the lv_screen_load_anim() function with the 'auto_del' parameter set to 'true'. To check which screen I'm on in a different function of the project, I do the following check:
`f (lv_obj_is_valid(ps_screen_audio) == true && lv_scr_act() == ps_screen_audio) {`

Normally, it is expected to enter here only when the active screen is ps_screen_audio, but a different screen that is active 3-4 screens ago and then deleted with 'auto_del' can also pass here.

How can we prevent this problem?

### How to reproduce?

To enter new screen with touch event:
```
static void v_TOUCHEVENT_AudioBack(lv_event_t *e)
{
    lv_indev_wait_release(lv_indev_get_act());
    v_SCREEN_MainInit();
    lv_obj_clean(lv_scr_act());
    lv_screen_load_anim(ps_screen_main, LV_SCR_LOAD_ANIM_NONE, 0, 0, true);
}
```
To check the active screen:
` if (lv_obj_is_valid(ps_screen_main) == true && lv_scr_act() == ps_screen_main) {}`

liamHowatt, 2024-07-26:
Please see https://github.com/lvgl/lvgl/issues/6395

`lv_obj_is_valid` is good to use for debugging but can report objects as valid after they've been deleted.

We could implement `lv_obj_null_on_delete` as was mentioned in the discussion.

fatiihyildizz, 2024-07-26:
As far as I can see, the lv_obj_null_on_delete function has not been released yet. Is there a date for when it will be released?

liamHowatt, 2024-07-29:
I opened a PR for it.  #6599

Is it clear how to use that function? Send a code sample if you want and I can show you how to use `lv_obj_null_on_delete` for your case.

fatiihyildizz, 2024-07-30:
We can proceed through this code snippet:

`static void v_TOUCHEVENT_AudioBack(lv_event_t *e)
{
    lv_indev_wait_release(lv_indev_get_act());
    v_SCREEN_MainInit();
    lv_obj_delete(lv_scr_act());
    lv_screen_load_anim(ps_screen_main, LV_SCR_LOAD_ANIM_NONE, 0, 0, true);
}`

liamHowatt, 2024-08-13:
I'm sorry, could you please send a bit more? In particular the part where you have this

> ` if (lv_obj_is_valid(ps_screen_main) == true && lv_scr_act() == ps_screen_main) {}`

Here's an example of how to use `lv_obj_null_on_delete` from the [PR](https://github.com/lvgl/lvgl/pull/6599)

```c
void some_timer_callback(lv_timer_t * t)
{
   static lv_obj_t * my_label;
   if(my_label == NULL) {
      my_label = lv_label_create(lv_screen_active());
      lv_obj_delete_delayed(my_label, 1000);
      lv_obj_null_on_delete(&my_label);
   }
   else {
      lv_obj_set_x(my_label, lv_obj_get_x(my_label) + 1);
}
```

PR #6599 (2024-08-13): feat(obj): add lv_obj_null_on_delete
### Description of the feature or fix

Fixes #6571
Related to #6395

Set an object pointer to `NULL` when it gets deleted.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig [... cut]
closed 2024-09-08 by lvgl-bot