lvgl/lvgl #6574: Update our ESP docs

kisvegabor, 2024-07-25:
### Introduce the problem

Our current [ESP32 docs](https://docs.lvgl.io/master/integration/chip/espressif.html) are quite outdated and shall be updated

### Proposal

I think we shall remove the current content and add these sections:
- Overview: 
   - general introduction abut Espressif/ESP32
   - Overview of the models highlighting the GUI related capabilities
- Get started in ESP-IDF
   - What is it?
   - Install ESP-IDF: link to the official guide
- ESP-BSP
  - What is it? Basic structure? 
  - How to try out a ready to use esp-bsp project? How to choose an other demo?
  - How to create a custom project?
      - [esp_lvgl_port](https://github.com/espressif/esp-bsp/tree/master/components/esp_lvgl_port/src/lvgl9) from `esp-bsp`
         - What is it? 
         - How to use it? Link to examples.   
      - How to add a custom driver? 
        - Why? How? Generic info about using SPI, etc  
 - Optimize speed 
   - What to configure to make LVGL faster
- Notes: Generic good to know notes  
  - E.g. how to use LVGL master as anon-managed component

kisvegabor, 2024-08-21:
Thanks to @tore-espressif  the docs are updated. At some point I think we should add [these points](https://github.com/lvgl/lvgl/pull/6658#issuecomment-2285767796).

henkjannl, 2024-08-24:
I recently bought an Elecrow 3.5" touch screen device as vehicle to learn LVGL. It was quite challenging to get started, since PlatformIO installed LVGL 9.0 by default, and I was unaware documentation is not yet up-to-date. It is working now, I posted my findings in:

[](https://forum.elecrow.com/discussion/474/lovyangfx-settings-for-3-5-parallel-rgb-display)

Perhaps the code snippets can be used as inspiration for the updated documentation.

I'm now struggling to create a GUI using SquareLine_Studio since this not yet supports V9. Is there any idea when SquareLine_Studio will support V9 as well?

kisvegabor, 2024-08-30:
I'm sorry for the trouble, but happy to hear that the new docs were useful. I can't comment much about SquareLine, but I recommend asking around at https://forum.squareline.io/ if you stuck somewhere.

kisvegabor, 2024-10-30:
Closing it for now. We can add new content on demand.