lvgl/lvgl #6576: Add method to detect not synced Kconfig, specially for NuttX

XuNeo, 2024-07-26:
>              OK - it's a NuttX thing. It seems that NuttX takes a copy of LVGLs Kconfig file and uses it locally rather than using the LVGL one directly. Vaguely remember some NuttX discussion about this - so the solution is imperfect but I understand it now.

_Originally posted by @TimJTi in https://github.com/lvgl/lvgl/issues/6313#issuecomment-2245290554_

Since the build system of NuttX has limitation of using third-party Kconfig that downloaded during build. NuttX copies the Kconfig file to work around it. But when migrating between different LVGL version, it's likely happen the Kconfig is out of sync with LVGL's Kconfig. We can use version number to detect it and report error if so.

TimJTi, 2024-08-13:
Is this something more related to NuttX, so we can let this auto-close here?

kisvegabor, 2024-08-17:
I think so, however @XuNeo knows more about it.

PR #6789 (2024-09-20): chore(Kconfig): add version info to Kconfig file to check mismatch

For NuttX, lvgl/Kconfig file is copied out, add version info in Kconfig file to detect if the Kconfig is out of sync with source code.

Fix #6576

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/maste [... cut]
closed 2024-09-20 by kisvegabor