lvgl/lvgl #6586: use lv_task_handler is dump

xiaohai1998, 2024-07-27:
### Introduce the problem

I'm using v8. Why do I get dump when I call lv_task_handler? Has anyone else experienced this problem

xiaohai1998, 2024-07-27:
this is my lvgl init code
`
static void mf_lv_flush_cb(lv_disp_drv_t* disp_drv, const lv_area_t* area, lv_color_t* color_p)
{
    ddi_log(0,"lvgl brush\r\n");
    lv_start_lock(1);
    dev_lcd_brush_screen();
    lv_start_lock(0);
    lv_disp_flush_ready(disp_drv);

}

static void mf_lv_display_driver_rounder_callback(lv_disp_drv_t* disp_drv, lv_area_t* area)
{
    area->x1 = 0;
    area->x2 = disp_drv->hor_res - 1;
    area->y1 = 0;
    area->y2 = disp_drv->ver_res - 1;
}

int lv_drv_init(void)
{
    int hor_res = 240;
    int ver_res = 320;

    lv_log_register_print_cb(ddi_log);

    lv_init();
    //lv_png_init();//png解码
    //lv_port_fs_init();
    g_p_dispmem = LCD_Getframe_buffer();

    
    lv_disp_draw_buf_init(
        &disp_buf,
        g_p_dispmem,
        NULL,
        hor_res * ver_res);

    lv_disp_drv_t disp_drv;
    lv_disp_drv_init(&disp_drv);
    disp_drv.hor_res = hor_res;
    disp_drv.ver_res = ver_res;
    disp_drv.flush_cb = mf_lv_flush_cb;
    disp_drv.draw_buf = &disp_buf;
    //disp_drv.rounder_cb = mf_lv_display_driver_rounder_callback;
    lv_disp_t* g_p_disp = lv_disp_drv_register(&disp_drv);
    if (g_p_disp == NULL)
    {
        ddi_log(0,"DBG====Failed to register disp drv!\n");
        return -1;
    }
       return 0;
}
`
main.c
`
int main(){
        lv_drv_init();
	for (;;)
	{
		LVGL_MUTEX_LOCK;
		lv_task_handler();
		LVGL_MUTEX_UNLOCK;
		ddi_sys_msleep(10);

	}
}

`
I ported lvgl8.0.1, but I didn't experience this problem with lvgl7

xiaohai1998, 2024-07-27:
this is my lvgl init code
`
static void mf_lv_flush_cb(lv_disp_drv_t* disp_drv, const lv_area_t* area, lv_color_t* color_p)
{
    ddi_log(0,"lvgl brush\r\n");
    lv_start_lock(1);
    dev_lcd_brush_screen();
    lv_start_lock(0);
    lv_disp_flush_ready(disp_drv);

}

static void mf_lv_display_driver_rounder_callback(lv_disp_drv_t* disp_drv, lv_area_t* area)
{
    area->x1 = 0;
    area->x2 = disp_drv->hor_res - 1;
    area->y1 = 0;
    area->y2 = disp_drv->ver_res - 1;
}

int lv_drv_init(void)
{
    int hor_res = 240;
    int ver_res = 320;

    lv_log_register_print_cb(ddi_log);

    lv_init();
    //lv_png_init();//png解码
    //lv_port_fs_init();
    g_p_dispmem = LCD_Getframe_buffer();

    
    lv_disp_draw_buf_init(
        &disp_buf,
        g_p_dispmem,
        NULL,
        hor_res * ver_res);

    lv_disp_drv_t disp_drv;
    lv_disp_drv_init(&disp_drv);
    disp_drv.hor_res = hor_res;
    disp_drv.ver_res = ver_res;
    disp_drv.flush_cb = mf_lv_flush_cb;
    disp_drv.draw_buf = &disp_buf;
    //disp_drv.rounder_cb = mf_lv_display_driver_rounder_callback;
    lv_disp_t* g_p_disp = lv_disp_drv_register(&disp_drv);
    if (g_p_disp == NULL)
    {
        ddi_log(0,"DBG====Failed to register disp drv!\n");
        return -1;
    }
       return 0;
}
`
main.c
`
int main(){
        lv_drv_init();
	for (;;)
	{
		LVGL_MUTEX_LOCK;
		lv_task_handler();
		LVGL_MUTEX_UNLOCK;
		ddi_sys_msleep(10);

	}
}

`
I ported lvgl8.0.1, but I didn't experience this problem with lvgl7,The error says draw_buf_flush in this function
[image]

xiaohai1998, 2024-07-27:
[image]

xiaohai1998, 2024-07-27:
[image]

Staok, 2024-07-29:
都是说中文的就不扯英文了。我注意到你的 mf_lv_flush_cb() 里面有个lv_start_lock(1);这个是加锁吗？是和 lv_task_handler() 前面的一样的锁吗，如果是就死锁了，flush_cb 回调函数里面不需要加互斥锁。
要保证 lvgl 在多线程环境稳定运行，要达到两个基本：

- lv_ 开头的 API 在调用前均要加互斥锁；
- 保证 lv_ 的 API 的正确的执行顺序，就是 如果顺序乱了就会出现 已经 删除的 obj 还被调用 就出现 访问空指针错误。

还有 lv_obj_is_valid()函数应该全局不能被调用。
等等

xiaohai1998, 2024-07-29:
这是来自QQ邮箱的假期自动回复邮件。你好，我最近正在休假中，无法亲自回复你的邮件。我将在假期结束后，尽快给你回复。

xiaohai1998, 2024-07-29:
我试过把锁注释掉也不行，我只是单纯移植lvgl v8版本的接口，然后只要调用那个lv_task_handler就会死机，方便加个qq私聊一下么 

小心輕放的光陰~~
***@***.***

xiaohai1998, 2024-07-29:
> 都是说中文的就不扯英文了。我注意到你的 mf_lv_flush_cb() 里面有个lv_start_lock(1);这个是加锁吗？是和 lv_task_handler() 前面的一样的锁吗，如果是就死锁了，flush_cb 回调函数里面不需要加互斥锁。 要保证 lvgl 在多线程环境稳定运行，要达到两个基本：
> 
> * lv_ 开头的 API 在调用前均要加互斥锁；
> * 保证 lv_ 的 API 的正确的执行顺序，就是 如果顺序乱了就会出现 已经 删除的 obj 还被调用 就出现 访问空指针错误。
> 
> 还有 lv_obj_is_valid()函数应该全局不能被调用。 等等

xiaohai1998, 2024-07-29:
> 都是说中文的就不扯英文了。我注意到你的 mf_lv_flush_cb() 里面有个lv_start_lock(1);这个是加锁吗？是和 lv_task_handler() 前面的一样的锁吗，如果是就死锁了，flush_cb 回调函数里面不需要加互斥锁。 要保证 lvgl 在多线程环境稳定运行，要达到两个基本：
> 
> * lv_ 开头的 API 在调用前均要加互斥锁；
> * 保证 lv_ 的 API 的正确的执行顺序，就是 如果顺序乱了就会出现 已经 删除的 obj 还被调用 就出现 访问空指针错误。
> 
> 还有 lv_obj_is_valid()函数应该全局不能被调用。 等等

我那个锁不是同一个

Staok, 2024-07-29:
那就可能是你的 dev_lcd_brush_screen() 里面挂了，可以编译程序 不适用任何优化 并且加上 -g 符号导出符号表，然后用 gdb 查看 coredump，gbd 里面用 list 等命令 查看具体挂在哪里了，因人而异，这得一点点查了

xiaohai1998, 2024-07-29:
> 那就可能是你的 dev_lcd_brush_screen() 里面挂了，可以编译程序 不使用任何优化 并且加上 -g 符号导出符号表，然后用 gdb 查看 coredump，gdb 里面用 list 等命令 查看具体挂在哪里了，因人而异，这得一点点查了

那我一会儿把那个接口注释掉试试，主要是我用v7版本的没这个问题

xiaohai1998, 2024-07-29:
> 那就可能是你的 dev_lcd_brush_screen() 里面挂了，可以编译程序 不使用任何优化 并且加上 -g 符号导出符号表，然后用 gdb 查看 coredump，gdb 里面用 list 等命令 查看具体挂在哪里了，因人而异，这得一点点查了

刚刚试了不行，还是死在那个lvgl事件循环里面

kmtlan, 2024-10-16:
我也遇到这个问题了，加不加锁都会卡在lv_task_handler()，到底怎么解决😭

xiaohai1998, 2024-10-16:
好兄弟，我解决了，太难了哇

小心輕放的光陰~~
***@***.***

Staok, 2024-11-21:
@xiaohai1998  you shall just close this issue, this keeping sending emails...

closed 2024-11-28 by kisvegabor