lvgl/lvgl #6587: Image Alignment Issue

MCervi42, 2024-07-27:
### LVGL version

9.1.0

### What happened?

Good day to everyone. First, I want to say a huge thanks to the developers and those who support LVGL.

**Issue:** The image displayed outputs misaligned (this is the closest I can describe it; pictures and video attached).

I found a couple of posts about this issue for older versions (V.7 and V.8) and tried to fix it, but with no success. I've spent 5 days trying to resolve this issue, but no luck. However, I am pretty confident that it has something to do with timings and buffer (e.g., `lv_task_handler();`, `lv_tick_inc();`). When I play with `DRAW_BUF_SIZE` (changing it to 4, 2, etc.), the lower `DRAW_BUF_SIZE` is set, the fewer misalignments there are. Also, I want to add that this effect is always different; sometimes the image can be almost good, and after a reboot, the image is totally misaligned.

I've tried to see what's going on with logging, but there are no logs at all. It's also worth noting that TFT_eSPI works great on its own, with no issues at all.

I am relatively new to this topic, so I'm sorry if this question is stupid and easily fixable. Once again, thanks for your work and advice.

Uploading IMG_6760.MOV…

I use VSCODE with Platformio.

3.2" TFT SPI 240x320px (IL9341)

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 115200
lib_deps = 
	lvgl/lvgl@^9.1.0
	bodmer/TFT_eSPI@^2.5.43
	

Uploading IMG_6760.MOV…

### How to reproduce?

#include <Arduino.h>
#include <lvgl.h>
#include <TFT_eSPI.h>
#include <demos/lv_demos.h>

#define TFT_HOR_RES 320
#define TFT_VER_RES 240

/*LVGL draw into this buffer, 1/10 screen size usually works well. The size is in bytes*/
void *draw_buf_1;
unsigned long lastTickMillis = 0;

static lv_display_t *disp;

#define DRAW_BUF_SIZE (TFT_HOR_RES * TFT_VER_RES / 10 * (LV_COLOR_DEPTH / 8))

void my_touchpad_read(lv_indev_t *indev, lv_indev_data_t *data) {

}

void setup() {
  Serial.begin(115200);
  lv_init();

  draw_buf_1 = heap_caps_malloc(DRAW_BUF_SIZE, MALLOC_CAP_DMA | MALLOC_CAP_INTERNAL);
  disp = lv_tft_espi_create(TFT_HOR_RES, TFT_VER_RES, draw_buf_1, DRAW_BUF_SIZE);

  lv_demo_benchmark();
}

void loop() {
  unsigned int tickPeriod = millis() - lastTickMillis;
  lv_tick_inc(tickPeriod);
  lastTickMillis = millis();
[... 968 lines cut ...]
    #define LV_DEMO_MUSIC_LANDSCAPE 0
    #define LV_DEMO_MUSIC_ROUND     0
    #define LV_DEMO_MUSIC_LARGE     0
    #define LV_DEMO_MUSIC_AUTO_PLAY 0
#endif

/*Flex layout demo*/
#define LV_USE_DEMO_FLEX_LAYOUT     0

/*Smart-phone like multi-language demo*/
#define LV_USE_DEMO_MULTILANG       0

/*Widget transformation demo*/
#define LV_USE_DEMO_TRANSFORM       0

/*Demonstrate scroll settings*/
#define LV_USE_DEMO_SCROLL          0

/*Vector graphic demo*/
#define LV_USE_DEMO_VECTOR_GRAPHIC  0
/*--END OF LV_CONF_H--*/

#endif /*LV_CONF_H*/

#endif /*End of "Content enable"*/

MCervi42, 2024-07-27:
https://github.com/user-attachments/assets/f2af8c10-8cdf-4ec0-90ba-9b0b92689a61

closed 2024-07-27 by MCervi42

kisvegabor, 2024-08-08:
Hi,

Have you found the solution? If so could you describe it here so that the next generation learn from it.

MCervi42, 2024-08-08:
**SOLUTION:**

I can't say what exactly helped to solve this issue, as i was trying a list of actions to solve this issue. However, I'm pretty sure those two steps helped me.

**STEPS:**
1. Downgrade to TFT_eSPI V. 2.5.0 (Checked with later releases, didn't work)
2. Set "SPI_FREQUENCY" in "User_Setup.h / _Section 4. Other options_" (For TFT_eSPI) to  **27000000**

**platformio.ini**
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 115200
lib_deps = 
	lvgl/lvgl@^9.1.0
	bodmer/TFT_eSPI@2.5.0