lvgl/lvgl #6588: PXP: undeclared function DEMO_CleanInvalidateByRange

faxe1008, 2024-07-27:
### LVGL version

22f1e03afbeaab46cdb14fbf8e5acc84f68140a5

### What happened?

When compiling I get this error here:

`
/src/draw/nxp/pxp/lv_draw_buf_pxp.c:71:9: warning: implicit declaration of function 'DEMO_CleanInvalidateCacheByAddr'; did you mean 'SCB_CleanInvalidateDCache_by_Addr'? [-Wimplicit-function-declaration]
   71 |         DEMO_CleanInvalidateCacheByAddr((void *)draw_buf->data, size);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         SCB_CleanInvalidateDCache_by_Addr
`
This function is not provided anywhere - is this something that is supposed to come from `lvgl_support.h`

kisvegabor, 2024-08-12:
cc @nicusorcitu @anaGrad

nicusorcitu, 2024-08-19:
cc @cosmindanielradu19

nicusorcitu, 2024-08-19:
@faxe1008 
`DEMO_CleanInvalidateCacheByAddr()` is implemented in `lvgl_support.c` and yes, it is a wrapper to clear and invalidate cache by range.
It is platform dependent, it can call `SCB_CleanInvalidateDCache_by_Addr` or `XCACHE_CleanInvalidateCacheByRange` or ...

faxe1008, 2024-08-19:
Alright, thanks for the clarification. The naming is a bit confusing. Would it be feasable to rename it or not worth the effort?

nicusorcitu, 2024-09-04:
> Alright, thanks for the clarification. The naming is a bit confusing. Would it be feasable to rename it or not worth the effort?

Sorry for my late response, I missed your reply. I would leave it as is. All the platform code is prefixed with `DEMO`, maybe that was the confusion.
If this satisfies your answer, please close the ticket.
Thanks,
Nicu.

closed 2024-09-04 by faxe1008

MickMad, 2024-10-24:
My two cents as an LVGL user on NXP platforms: I think this is a really "dirty" solution; just the fact that the function name starts with DEMO should raise a red flag.
It would be much cleaner to add a define in lv_conf.h to reference a platform specific cache invalidation instruction, instead of referencing in LVGL a very specific function name that exists only in very specific NXP evaulation projects.
Please note that this issue also affects VGLite code.

nicusorcitu, 2024-10-24:
Hi @MickMad 
I can agree with you that "DEMO" prefix is not the best naming for a function called from inside lvgl library. This is a legacy behavior from lvgl 8.x where the invalidate cache function was called _only_ from SDK enablement code.
Once with lvgl 9.x and the introduce of "invalidate_cache_cb", we kept the same naming - I will revise it for the next 9.3 release. @cosmindanielradu19

But I can not agree with a define in lv_conf.h to replace that functionality. The purpose of this API is to invalidate data cache from a specific address by a specific size. And this can apply only to certain draw buffers. E.G. LVGL allows you to have a different implementation for font handlers and a different one for other generic buffers (like layers).

There is no issue I can see neither on PXP or VGLite.

MickMad, 2024-10-24:
Hi @nicusorcitu , thanks for the quick reply.
I agree that the "invalidate_cache_cb" is the best way to handle this. No need to have a define then.

> There is no issue I can see neither on PXP or VGLite.
Perhaps I used the wrong words; what I meant is that the function DEMO_CleanInvalidateCacheByAddr is also called in VGLite code (lv_draw_buf_vglite.c, lines 72 and 107). Any change here will have to be reflected there as well.

nicusorcitu, 2024-10-24:
Yes, both PXP and VGLite (on rt1170 for example) share the same API function (as they are on the same platform).

andrewleech, 2024-11-13:
> @faxe1008 `DEMO_CleanInvalidateCacheByAddr()` is implemented in `lvgl_support.c` and yes, it is a wrapper to clear and invalidate cache by range. It is platform dependent, it can call `SCB_CleanInvalidateDCache_by_Addr` or `XCACHE_CleanInvalidateCacheByRange` or ...

Could you please provide a reference to a example project that implements this? I'm running into this failure and it's not defined in the MCU SDKs /  examples I've got for my 1176 EVK

nicusorcitu, 2024-11-13:
When you generate the SDK from NXP website please make sure you have the fallowing components:
FreeRTOS, LVGL, VG-Lite GPU Library

andrewleech, 2024-11-13:
Could those files please be added to a NXP GitHub repo? At work I need to show traceability of code as we work in a regulated space. Having to manually download packs like that requires a lot of extra paperwork to describe where/how it was obtained.

nicusorcitu, 2024-11-14:
Yes, starting with next generation SDK 24.12, all the releases will be done on NXP github as source code.