lvgl/lvgl #6591: Is there a low-level API for rotation that facilitates using an external accelerator?

Zmmfly, 2024-07-29:
### Introduce the problem

After purchasing the ESP32-P4 development board, I wanted to use its PPA peripheral to improve the LVGL frame rate.

In RGB888 color mode, the frame rate improvement brought by using PPA is huge for fill operations, but not so obvious for blend operations;

On RGB565, I found that there is basically no difference between using PPA and not using PPA. I don’t know what went wrong. I found that the frame rate of image rotation in benchmarks is very low. If it can be accelerated with PPA, I think the benefits will be huge;

However, I didn’t find a convenient way to override the default operation like `LV_DRAW_SW_RGB888_BLEND_NORMAL_TO_RGB888`.

faxe1008, 2024-07-30:
From my understanding of LVGL you would write your custom draw unit implementation for this. In there you can expose which draw operations it should handle and also a preference score. You can take a peek at the existing accelerators for inspiration. The API they use is afaik all public.

kisvegabor, 2024-08-17:
I can confirm what @faxe1008 said. 

I can also add that usually simple 2D GPUs are not faster than SW rendering, partly because some of the operations are limited by the memory bandwidth. However GPUs are working in the background and at least the CPU can do something else in the meantime.

closed 2024-10-03 by lvgl-bot