lvgl/lvgl #6598: Linux DRM - rotation does not appear to work

jbd1986, 2024-07-29:
### LVGL version

v9.1.0

### What happened?

I'm attempting to port our v8.3-based application to LVGL 9.1

We're using a ILI9341 240x320 screen physically rotated to be 320x240, and we're relying on sw_rotate=1 combined with 270 degree rotation to get the desired output.

So far, I've replaced our original DRM initialization code with the code shown in the reproduce section.

This brings up our initial splash screen with the correct bg color and text color, but the screen has not been rotated (looks like it's still in portrait mode), and the text is completely unreadable.  If I remove the lv_display_set_rotation line, the text will become readable as expected, but obviously the screen is still in portrait mode.

### How to reproduce?

const char *device = "/dev/dri/card0";
lv_display_t * disp = lv_linux_drm_create();

// Configure display parameters
lv_display_set_resolution(disp, 240, 320);
lv_display_set_color_format(disp, LV_COLOR_FORMAT_ARGB8888);
lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_270);
lv_display_set_render_mode(disp, LV_DISPLAY_RENDER_MODE_PARTIAL);

// Fully initialize DRM
lv_linux_drm_set_file(disp, device, -1);

jbd1986, 2024-08-02:
Other users are having similar issues, see this forum post:

https://forum.lvgl.io/t/how-to-rotate-screen-with-drm-backend/16704

jbd1986, 2024-08-17:
Issue still exists. 

Would recommend to change the default render method to FULL or PARTIAL, and add support for rotation in drm_flush.

kisvegabor, 2024-08-21:
I think we should rather add support for both. However aren't there any features in DRM to support rotation?

jbd1986, 2024-08-21:
Each DRM implementation is different.  IMX8 platform for example does not have a HW rotation implementation.  

If LVGL is handling both buffers in the double buffering scheme and responsible for triggering the buffer swap, then that is probably the easiest, and most efficient place to perform a SW-based rotation.  

If performing this on the IMX8 linux DRM driver side, I would think you would have to create an additional set of framebuffers, implement a secondary double buffering scheme, and then copy the data over as soon as you receive a frame, so that you don't run the risk of the underlying data changing while you are manipulating it.  This would also add frame latency.  Maybe I'm overthinking it, but I'm fairly sure DRM expects the frames to be 'complete' already.

jbd1986, 2024-08-26:
BUMP, still an issue with code at HEAD

kisvegabor, 2024-08-29:
It'd be really nice to add this feature. 

@marex, @MiSimon, @timstableford, @ErichStyger do you have time and interest to add rotation feature to DRM, similar to [this](https://github.com/lvgl/lvgl/blob/master/src/drivers/sdl/lv_sdl_window.c#L215-L251). (It should work only with PARTIAL mode)

jbd1986, 2024-09-30:
Issue should be re-opened

vwheeler63, 2025-08-06:
Probably also related to #8636

jdefields, 2025-09-15:
Any movement on this?

ino-josh, 2025-10-07:
@AndreCostaaa I was just looking into this... looks like the iMX8M Mini is the _only_ processor in the iMX series that is ES 2.0. Even the Nano supports 3.1.
That's a bummer... is there an advantage to targeting 3.x vs 2.0, when 2.0 would offer the widest compatibility? Is there something in the rendering that requires 3.x?

closed 2026-02-23 by lvgl-bot