lvgl/lvgl #6602: Roller Not Sending Value Change When Setting The Index From Code

NobodyMore, 2024-07-30:
### LVGL version

V9.1.0

### What happened?

problem happened when i discovered when changing the roller value with keypad, it doesnt send "LV_EVENT_VALUE_CHANGED"
it works perfectly upon scrolling using mouse. 
enter button can be used to force this event.
problem is that when we change the current position of a roller from the code, it doesnt send this event.
it can be fixed by sending the event in lv_roller.c in the end of method "lv_roller_set_selected"

liamHowatt, 2024-07-31:
@kisvegabor any reason not to add `lv_obj_send_event(obj, LV_EVENT_VALUE_CHANGED, &roller->sel_opt_id)` [here](https://github.com/lvgl/lvgl/blob/a798963ee89164ba2758c7fdb5fabea33e22a805/src/widgets/roller/lv_roller.c#L372)?

kisvegabor, 2024-08-01:
Normally the value is changed when your hit enter. At least this is how it work on PC in e.g. in drop downs.  

As a workaround you can add an event to `LV_EVENT_KEY` and get `lv_roller_get selected()`. Could it work in your cas?

NobodyMore, 2024-08-03:
thanks for your response.
in can be because im using the rollers diffrently than their original usage. im using rollers with one visible row to make something like a state selector. 
i will use for solution with KEY event.
but i think an event for this changes can be useful. like on value change on inputFields.

closed 2024-08-03 by NobodyMore