lvgl/lvgl #6610: Lottie animation freezes when size changed

utiq, 2024-08-01:
### LVGL version

v9.1.1-dev

### What happened?

I'm trying to play a lottie animation from the example. It works fine if the size is 64, when I increase it to 96, it freezes. I'm using ESP-IDF 5.1.4 and running on a ESP32-S3-N32R8

### How to reproduce?

This is my code:
```
    // Test lottie animation
    extern const uint8_t lv_example_lottie_approve[];
    extern const size_t lv_example_lottie_approve_size;

    lv_obj_t *lottie = lv_lottie_create(lv_screen_active());

    lv_lottie_set_src_data(lottie, lv_example_lottie_approve, lv_example_lottie_approve_size);

    // Check free heap size after allocation
    free_heap_size = heap_caps_get_free_size(MALLOC_CAP_8BIT);
    ESP_LOGI(TAG, "Free heap size after buffer allocation: %d bytes", free_heap_size);

    static uint8_t buf[96 * 96 * 4];
    lv_lottie_set_buffer(lottie, 96, 96, buf);

    lv_obj_center(lottie);
```

this is part of the logs, no error is shown, see available memory:
```
I (1278) gc9a01: Configure LCD backlight pin
I (1278) gpio: GPIO[40]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1281) gc9a01: Turn on LCD backlight
I (1286) gc9a01: Backlight level set to  1
I (1390) gc9a01: Free heap size before buffer allocation: 8576019 bytes
I (1395) gc9a01: Install LVGL tick timer
I (1396) gc9a01: Print some elements on the display
I (1406) gc9a01: Free heap size after buffer allocation: 8564579 bytes
I (1407) main_task: Returned from app_main()
```

This is how I'm setting the update task
```
// Initialize the queue before creating the task
  system_queue = xQueueCreate(10, sizeof(system_packet));
  if (system_queue == NULL)
  {
    ESP_LOGE(TAG, "Failed to create system queue");
    return;
  }

  xTaskCreatePinnedToCore(lvgl_time_task, "lvgl_time_task", 100 * 1024, NULL, 5, NULL, 0);
```

kisvegabor, 2024-08-01:
Can you make it smaller? E.g. 32x32?

kisvegabor, 2024-08-01:
Lottie allocates memories from a lot of places:
- stack
- normal malloc
- LVGL's malloc

Sometimes it's tricky to find the limiting one. What if you increse `LV_MEM_SIZE`?

utiq, 2024-08-01:
@kisvegabor  `LV_MEM_SIZE` was 128, I increased it a little more, no change.
I also noticed that there is a `LV_DRAW_THREAD_STACK_SIZE`, I increased it to 128, no change

kisvegabor, 2024-08-17:
Lottie also uses normal heap (`malloc`) do you have large enough system heap?

closed 2024-09-14 by lvgl-bot