lvgl/lvgl #6611: Use Arm2D with LLVM

kisvegabor, 2024-08-01:
### LVGL version

master

### What happened?

I added Arm2D to [this repo](https://github.com/lvgl/lv_port_renesas_ek-ra8d1) added `arm_2d_cfg.h` and CMSIS-DSP, disabled LTO (to avoid some link time error), added `-fms-extensions` but finally got this error:
```
In file included from ../src/Arm-2D/Library/Source/arm_2d_helium.c:196:
../src/Arm-2D/Library/Source/__arm_2d_tile_copy_with_source_mask_and_opacity_helium.c:479:14: error: operand must be an even-numbered register
  479 |             "  vidup.u16       gathOfs, %[incr], #0x4             \n"
      |              ^
<inline asm>:3:28: note: instantiated into assembly here
    3 |   vidup.u16       gathOfs, r11, #0x4             
      |                            ^
In file included from ../src/Arm-2D/Library/Source/arm_2d_helium.c:196:
../src/Arm-2D/Library/Source/__arm_2d_tile_copy_with_source_mask_and_opacity_helium.c:523:14: error: operand must be an even-numbered register
  523 |             "  vidup.u16       gathOfs, %[incr], #0x4             \n"
      |              ^
<inline asm>:29:28: note: instantiated into assembly here
   29 |   vidup.u16       gathOfs, r11, #0x4             
      |                            ^
In file included from ../src/Arm-2D/Library/Source/arm_2d_helium.c:196:
../src/Arm-2D/Library/Source/__arm_2d_tile_copy_with_source_mask_and_opacity_helium.c:758:14: error: operand must be an even-numbered register
  758 |             "  vidup.u16       gathOfs, %[incr], #0x4             \n"
      |              ^
<inline asm>:3:28: note: instantiated into assembly here
    3 |   vidup.u16       gathOfs, r3, #0x4             
      |                            ^
In file included from ../src/Arm-2D/Library/Source/arm_2d_helium.c:196:
../src/Arm-2D/Library/Source/__arm_2d_tile_copy_with_source_mask_and_opacity_helium.c:799:14: error: operand must be an even-numbered register
  799 |             "  vidup.u16       gathOfs, %[incr], #0x4             \n"
      |              ^
<inline asm>:27:28: note: instantiated into assembly here
   27 |   vidup.u16       gathOfs, r3, #0x4             
      |                            ^
4 errors generated.
make: *** [src/Arm-2D/Library/Source/subdir.mk:101: src/Arm-2D/Library/Source/arm_2d_helium.o] Error 1
Building file: ../src/Arm-2D/Library/Source/arm_2d_helium.c
clang --target=arm-none-eabi -mcpu=cortex-m85 -mthumb -O3 -ffunction-sections -fdata-sections -fdiagnostics-parseable-fixits -fms-extensions -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_port_renesas_ek-ra8d1_llvm/src" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_port_renesas_ek-ra8d1_llvm/src/Arm-2D/Helper/Include" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_port_renesas_ek-ra8d1_llvm/src/Arm-2D/Library/Include" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_port_renesas_ek-ra8d1_llvm/src/Arm-2D/Library/Include" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_port_renesas_ek-ra8d1_llvm/ra/fsp/inc" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_port_renesas_ek-ra8d1_llvm/ra/fsp/inc/api" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_port_renesas_ek-ra8d1_llvm/ra/fsp/inc/instances" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_port_renesas_ek-ra8d1_llvm/ra/fsp/src/rm_freertos_port" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_port_renesas_ek-ra8d1_llvm/ra/aws/FreeRTOS/FreeRTOS/Source/include" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_port_renesas_ek-ra8d1_llvm/ra/arm/CMSIS_5/CMSIS/Core/Include" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_port_renesas_ek-ra8d1_llvm/ra_gen" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_port_renesas_ek-ra8d1_llvm/ra_cfg/fsp_cfg/bsp" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_port_renesas_ek-ra8d1_llvm/ra_cfg/fsp_cfg" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_port_renesas_ek-ra8d1_llvm/ra_cfg/aws" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_port_renesas_ek-ra8d1_llvm/ra/tes/dave2d/inc" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_port_renesas_ek-ra8d1_llvm/src/lvgl" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_port_renesas_ek-ra8d1_llvm/src/lvgl/src" -I"." -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_port_renesas_ek-ra8d1_llvm/generate" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_port_renesas_ek-ra8d1_llvm/ra/arm/CMSIS_5/CMSIS/DSP/PrivateInclude" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_port_renesas_ek-ra8d1_llvm/ra/arm/CMSIS_5/CMSIS/DSP/Include" -D_RENESAS_RA_ -D_RA_CORE=CM85 -D_RA_ORDINAL=1 -MMD -MP -MF"src/Arm-2D/Library/Source/arm_2d_helium.d" -MT"src/Arm-2D/Library/Source/arm_2d_helium.o" "../src/Arm-2D/Library/Source/arm_2d_helium.c" -c -o "src/Arm-2D/Library/Source/arm_2d_helium.o"
"make -r --output-sync -j16 all" terminated with exit code 2. Build might be incomplete.

11:01:15 Build Failed. 10 errors, 0 warnings. (took 6s.260ms)
```

@GorgonMeducer could you take a look at it?

GorgonMeducer, 2024-08-01:
@kisvegabor 
We are looking at this issue.
Please define macro USE_MVE_INTRINSICS as a workaround.

FabKlein, 2024-08-01:
Error is caused by a missing inline asm odd/even register constraint ("Te" for even / "To" for odd)
Arm-v8.1M architecture enforces such constraints on some instructions like VIDUP. 
Fix pushed here : https://github.com/ARM-software/Arm-2D/commit/5cf69a5feeb46272d270799cb3ac2a4f09b6395a

GorgonMeducer, 2024-08-01:
Thank you for your prompt action @FabKlein 

@kisvegabor Could you please confirm the fixing? (please use the developing branch).

kisvegabor, 2024-08-01:
Thank @FabKlein! I could get further with compiling. Now it fails wit this:
```
>>> referenced by putchar.c:0 (/workspace/workspace/llvm-embedded/llvm-17/build-and-test/build/picolibc/armv7m_soft_fpv4_sp_d16/src/picolibc_armv7m_soft_fpv4_sp_d16-build/../../../../../src/picolibc/newlib/libc/tinystdio/putchar.c:0)
>>>               libc_tinystdio_putchar.c.o:(putchar) in archive /home/kisvegabor/LLVMEmbeddedToolchainForArm-17.0.1-Linux-x86_64/bin/../lib/clang-runtimes/arm-none-eabi/armv7m_soft_fpv4_sp_d16/lib/libc.a
>>> referenced by printf.c:0 (/workspace/workspace/llvm-embedded/llvm-17/build-and-test/build/picolibc/armv7m_soft_fpv4_sp_d16/src/picolibc_armv7m_soft_fpv4_sp_d16-build/../../../../../src/picolibc/newlib/libc/tinystdio/printf.c:0)
>>>               libc_tinystdio_printf.c.o:(printf) in archive /home/kisvegabor/LLVMEmbeddedToolchainForArm-17.0.1-Linux-x86_64/bin/../lib/clang-runtimes/arm-none-eabi/armv7m_soft_fpv4_sp_d16/lib/libc.a
clang++: error: ld.lld command failed with exit code 1 (use -v to see invocation)
make: *** [makefile:382: lv_port_renesas_ek-ra8d1_llvm.elf] Error 1
clang++  --target=arm-none-eabi -mcpu=cortex-m85 -mthumb -O3 -ffunction-sections -fdata-sections -fdiagnostics-parseable-fixits -o "lv_port_renesas_ek-ra8d1_llvm.elf" ./src/port/lv_port_indev.o  ./src/lvgl/src/widgets/win/lv_win.o  ./src/lvgl/src/widgets/tileview/lv_tileview.o  ./src/lvgl/src/widgets/textarea/lv_textarea.o  ./src/lvgl/src/widgets/tabview/lv_tabview.o  ./src/lvgl/src/widgets/table/lv_table.o  ./src/lvgl/src/widgets/switch/lv_switch.o  ./src/lvgl/src/widgets/spinner/lv_spinner.o  ./src/lvgl/src/widgets/spinbox/lv_spinbox.o  ./src/lvgl/src/widgets/span/lv_span.o  ./src/lvgl/src/widgets/slider/lv_slider.o  ./src/lvgl/src/widgets/scale/lv_scale.o  ./src/lvgl/src/widgets/roller/lv_roller.o  ./src/lvgl/src/widgets/property/lv_image_properties.o ./src/lvgl/src/widgets/property/lv_obj_properties.o ./src/lvgl/src/widgets/property/lv_style_properties.o  ./src/lvgl/src/widgets/objx_templ/lv_objx_templ.o  ./src/lvgl/src/widgets/msgbox/lv_msgbox.o  ./src/lvgl/src/widgets/menu/lv_menu.o  ./src/lvgl/src/widgets/lottie/lv_lottie.o  ./src/lvgl/src/widgets/list/lv_list.o  ./src/lvgl/src/widgets/line/lv_line.o  ./src/lvgl/src/widgets/led/lv_led.o  ./src/lvgl/src/widgets/label/lv_label.o  ./src/lvgl/src/widgets/keyboard/lv_keyboard.o  ./src/lvgl/src/widgets/imagebutton/lv_imagebutton.o  ./src/lvgl/src/widgets/image/lv_image.o  ./src/lvgl/src/widgets/dropdown/lv_dropdown.o  ./src/lvgl/src/widgets/checkbox/lv_checkbox.o  ./src/lvgl/src/widgets/chart/lv_chart.o  ./src/lvgl/src/widgets/canvas/lv_canvas.o  ./src/lvgl/src/widgets/calendar/lv_calendar.o ./src/lvgl/src/widgets/calendar/lv_calendar_chinese.o ./src/lvgl/src/widgets/calendar/lv_calendar_header_arrow.o ./src/lvgl/src/widgets/calendar/lv_calendar_header_dropdown.o  ./src/lvgl/src/widgets/buttonmatrix/lv_buttonmatrix.o  ./src/lvgl/src/widgets/button/lv_button.o  ./src/lvgl/src/widgets/bar/lv_bar.o  ./src/lvgl/src/widgets/arc/lv_arc.o  ./src/lvgl/src/widgets/animimage/lv_animimage.o  ./src/lvgl/src/tick/lv_tick.o  ./src/lvgl/src/themes/simple/lv_theme_simple.o  ./src/lvgl/src/themes/mono/lv_theme_mono.o  ./src/lvgl/src/themes/default/lv_theme_default.o  ./src/lvgl/src/themes/lv_theme.o  ./src/lvgl/src/stdlib/rtthread/lv_mem_core_rtthread.o ./src/lvgl/src/stdlib/rtthread/lv_sprintf_rtthread.o ./src/lvgl/src/stdlib/rtthread/lv_string_rtthread.o  ./src/lvgl/src/stdlib/micropython/lv_mem_core_micropython.o  ./src/lvgl/src/stdlib/clib/lv_mem_core_clib.o ./src/lvgl/src/stdlib/clib/lv_sprintf_clib.o ./src/lvgl/src/stdlib/clib/lv_string_clib.o  ./src/lvgl/src/stdlib/builtin/lv_mem_core_builtin.o ./src/lvgl/src/stdlib/builtin/lv_sprintf_builtin.o ./src/lvgl/src/stdlib/builtin/lv_string_builtin.o ./src/lvgl/src/stdlib/builtin/lv_tlsf.o  ./src/lvgl/src/stdlib/lv_mem.o  ./src/lvgl/src/others/vg_lite_tvg/vg_lite_matrix.o ./src/lvgl/src/others/vg_lite_tvg/vg_lite_tvg.o  ./src/lvgl/src/others/sysmon/lv_sysmon.o  ./src/lvgl/src/others/snapshot/lv_snapshot.o  ./src/lvgl/src/others/observer/lv_observer.o  ./src/lvgl/src/others/monkey/lv_monkey.o  ./src/lvgl/src/others/imgfont/lv_imgfont.o  ./src/lvgl/src/others/ime/lv_ime_pinyin.o  ./src/lvgl/src/others/gridnav/lv_gridnav.o  ./src/lvgl/src/others/fragment/lv_fragment.o ./src/lvgl/src/others/fragment/lv_fragment_manager.o  ./src/lvgl/src/others/file_explorer/lv_file_explorer.o  ./src/lvgl/src/osal/lv_cmsis_rtos2.o ./src/lvgl/src/osal/lv_freertos.o ./src/lvgl/src/osal/lv_mqx.o ./src/lvgl/src/osal/lv_os.o ./src/lvgl/src/osal/lv_os_none.o ./src/lvgl/src/osal/lv_pthread.o ./src/lvgl/src/osal/lv_rtthread.o ./src/lvgl/src/osal/lv_windows.o  ./src/lvgl/src/misc/cache/_lv_cache_lru_rb.o ./src/lvgl/src/misc/cache/lv_cache.o ./src/lvgl/src/misc/cache/lv_cache_entry.o ./src/lvgl/src/misc/cache/lv_image_cache.o ./src/lvgl/src/misc/cache/lv_image_header_cache.o  ./src/lvgl/src/misc/lv_anim.o ./src/lvgl/src/misc/lv_anim_timeline.o ./src/lvgl/src/misc/lv_area.o ./src/lvgl/src/misc/lv_array.o ./src/lvgl/src/misc/lv_async.o ./src/lvgl/src/misc/lv_bidi.o ./src/lvgl/src/misc/lv_color.o ./src/lvgl/src/misc/lv_color_op.o ./src/lvgl/src/misc/lv_event.o ./src/lvgl/src/misc/lv_fs.o ./src/lvgl/src/misc/lv_ll.o ./src/lvgl/src/misc/lv_log.o ./src/lvgl/src/misc/lv_lru.o ./src/lvgl/src/misc/lv_math.o ./src/lvgl/src/misc/lv_matrix.o ./src/lvgl/src/misc/lv_palette.o ./src/lvgl/src/misc/lv_profiler_builtin.o ./src/lvgl/src/misc/lv_rb.o ./src/lvgl/src/misc/lv_style. [... cut ...] r/projects/lvgl/e2_studio-workspace/lv_port_renesas_ek-ra8d1_llvm/script" -Wl,-Map=lv_port_renesas_ek-ra8d1_llvm.map -Wl,--start-group -Wl,--end-group -nostartfiles -Wl,-e_PowerON_Reset -Wl,--cref -Wl,--icf=none
"make -r --output-sync -j16 all" terminated with exit code 2. Build might be incomplete.
``` 

Almost all the Arm2D folsers are excluded from the build so I don't from where `printf` is referenced:
[image]

 Probably it's coming from Arm2D as it was working earlier. I keep investigating.

GorgonMeducer, 2024-08-01:
@kisvegabor For this linker issue, I have no idea. If it misses something, can we find why it is missing?

kisvegabor, 2024-08-01:
I've found that `__arm_2d_log_printf` is causing the issue. If I remove its content it compiles well. 

An other minor issue `__va_list` was not found, I needed to rename it to `va_list` in LLVM.

GorgonMeducer, 2024-08-01:
> I've found that __arm_2d_log_printf is causing the issue. If I remove its content it compiles well.

Have you disabled the arm-2d log in arm_2d_cfg.h?
`__ARM_2D_CFG_ENABLE_LOG__`, please define it as `0`

> An other minor issue __va_list was not found, I needed to rename it to va_list in LLVM.

OK, I will fix this.

kisvegabor, 2024-08-12:
It compiles well with the latest commit in `developing`. Thank you!

GorgonMeducer, 2024-08-13:
Great! We can close this issue.

kisvegabor, 2024-08-17:
Thank you @GorgonMeducer and @FabKlein!

closed 2024-08-17 by kisvegabor