lvgl/lvgl #6627: Can we add image pre caching function in lvgl

jinsc123654, 2024-08-03:
### Problem to solve

Although enabling LV_IMG_CACHE_DEF_SIZE allows images to be cached after use.

If my next interface will have an animation consisting of 10-20 images, using LV_IMG_CACHE_DEF_SIZE can cache these images, but there will be significant frame drops on the first interface image.

Is there a function provided to pre decode the image of the file system and enter it into the LV_IMG_CACHE_DEF_SIZE cache.

Alternatively, provide another macro definition for preloading images of the file system.

### Success criteria

When lvgl is started or the image control is not used, some file system files are preloaded. After preloading is completed, the image control is used for display.

### Solution outline

After preloading the image, there is no significant decrease in frame rate when using the file system path for the first time

### Rabbit holes

1: When to perform preloading

2: How to use img control to display images before preloading is completed

### Testing

On the MCU, pre load 20 or more images for animation playback, and there will be no significant frame rate drop during the first decoding of the images

### Teaching

Same as the display of testing methods

jinsc123654, 2024-08-05:
😊

XuNeo, 2024-08-05:
Manually call `lv_image_decoder_open` to preload the image to cache. After call `lv_image_decoder_close`, it will be automatically released when cache algorithm says so.

You'll have to choose a right time to preload image, to sustain normal user operation. A common case would be the the short time after animation transition, after which the user is not ready to interact again. The preload time should be in several mili-seconds.

Time line:
```
interaction(clickButton)---animation---easeIn(out)---preload---interactionAgain
```

Use another thread to read file if you could, to avoid such complication.

closed 2024-08-05 by jinsc123654

kisvegabor, 2024-08-07:
@XuNeo I've already mentioned it somewhere that having a simple `lv_image_preload(img_src, draw_buf)` function could be useful to have full control what to load, when, and which memory to use. This way we could preload image files to external RAM.