lvgl/lvgl #6631: How to use private headers

XuNeo, 2024-08-03:
### Introduce the problem

A common sense is not to include private headers from library.

The basic data structure like `lv_obj_class_t` is hidden. How the developers could develop custom widgets?
If developers are forced to include `lvgl_private.h`, then it makes no sense to separate them to private headers.

Another point is LVGL have many basic modules like list, text processing. Those should also be provided as public because there's real case to use them. It makes no sense to copy the existing sources in order to achieve same functionality.

### Proposal

What private is really private and should be used outside of LVGL. Event processing is one case that should be really private.  Basic obj's data structure is needed to extend LVGL's capability, thus should be public.

Make basic utilities as public.

lhdjply, 2024-08-03:
I think it's probably due to a poorly chosen filename. `lv_private.h` is meant to be private and generally not called upon. Would renaming it to `lvgl_pro.h` or `lvgl_ex.h` be better? That would give the impression of being for extensions or advanced features, and if users need to call upon it, they can do so themselves.

FASTSHIFT, 2024-08-03:
> I think it's probably due to a poorly chosen filename. `lvgl_private.h` is meant to be private and generally not called upon. Would renaming it to `lvgl_pro.h` or `lvgl_ex.h` be better? That would give the impression of being for extensions or advanced features, and if users need to call upon it, they can do so themselves.

My thinking is the same as yours, because the ultimate goal is to distinguish between plain apis and high-level apis, which are also essentially public apis. Calling it `priavte` is more likely to confuse users.

kisvegabor, 2024-08-03:
What about calling it lvgl_intetnal.h?

FASTSHIFT, 2024-08-04:
> What about calling it lvgl_internal.h?

I agree.
How about `lvgl_advanced.h`? It was suggested by ChatGPT.

XuNeo, 2024-08-04:
> I think it's probably due to a poorly chosen filename. lvgl_private.h is meant to be private and generally not called upon. Would renaming it to lvgl_pro.h or lvgl_ex.h be better? That would give [...]

That makes sense. I vote for `lvgl_ex.h` or `lvgl_extension.h`, as these names imply they are for developing extensions for LVGL. Application developers should use `lvgl.h`.

kisvegabor, 2024-08-05:
`extension` might mean that a public extension to LVGL. E.g. less commonly used widgets.

`internal` is already used in `lv_conf_internal.h`, meaning its for internal (i.e. developers) use.

cc @liamHowatt

XuNeo, 2024-08-05:
> meaning its for internal (i.e. developers) use.

But the people using LVGL are all developers. Sorry, but I still feel `internal` is same as `private` that should not be exposed(I.E `external`).

kisvegabor, 2024-08-05:
> But the people using LVGL are all developers. 

True! :smile: 
I meant developer of LVGL, which is different from the developer just using LVGL . I think it's normal to use private headers for developing the library, but not for developing an application with the library. 

The question is: if you write a new widget for LVGL, is it rather part of LVGL (the UI library) or part of the application? Technically it's not part of LVGL as you don't add these in the `lvgl` folder. However, it's not a typical part of the application either as it's just a building block similar to `lv_slider`. From the app's point of view it doesn't matter where it's placed. All the widgets are supplied by an underlying  UI engine from which the app builds up the screen.

liamHowatt, 2024-08-06:
It depends on how much we want to encourage/discourage the use of hidden features. If we call it `lvgl_advanced.h`, then an application developer will feel encouraged to use the advanced features without fear of compromising the stability of their app. The downside is that the "risks" might not be clear, because they are slightly compromising the longer-term compatibility of their app, and maybe that should be made clear by a cautionary name like `private` or `internal`. I could imagine `extension` (`ex` is too obfuscated maybe?) being a decent compromise, because if the user accepts the implications of "extending" LVGL, then they are prepared for the risks. The only issue is that "extensions" imply plugins or something else that it's not.

Whatever the decision, I think this doc could be filled in in the PR: https://docs.lvgl.io/master/overview/new_widget.html

ataweg, 2024-08-18:
I have to tell you very clearly: it was not a very good idea to move the definitions of the structs into the private header files. Everyone who has been using the last release (9.1) will be cursing at the transition to the next release, because they will either have to include the private header files they need, which will make the concept of private header files absurd. Or they will have to change their code massively, which they probably won't do. I implore you to reconsider this step. It would be a great help to many LVGL enthusiasts.

kisvegabor, 2024-08-20:
Please mention some examples of when users need to include private headers. 

I assume that the vast majority of the users are using the basic API of LVGL (available from `lvgl.h`) and only the power users use advanced features where `lvgl_private.h` is needed.

kisvegabor, 2024-08-21:
I can't see a good place to mention it. Any ideas?

We should mention it in the migration guide for sure though.

XuNeo, 2024-08-21:
What about adding a config option to include `lvgl_private.h` from `lvgl.h`, so not to break existing projects.

By the way, renaming of `_lv_xxx` still needs to update API map, to make it compatible with v9.1.

kisvegabor, 2024-08-21:
> What about adding a config option to include lvgl_private.h from lvgl.h, so not to break existing projects.

Such as `LV_USE_PRIVATE_API` (0 by default)? 

> By the way, renaming of _lv_xxx still needs to update API map, to make it compatible with v9.1.

Before this API change it wasn't clear which APIs are considered internal and which was considered public. I'd say if someone used `_lv*` API he was using an internal API which can be changed. However, we should add entries for to most common functions to minimize the rage. 😀

What are these functions in your opinion?

XuNeo, 2024-08-21:
> Such as LV_USE_PRIVATE_API (0 by default)?

Yes. In this way, all the existing projects can enable this option in order to use v9.2 seamlessly.

> What are these functions in your opinion?

There really are lots of functions could be used by application(the utils).

E.g.

```c
_LV_EVENT_LAST
_LV_FLEX_REVERSE
_LV_STYLE_LAST_BUILT_IN_PROP
_lv_ll_init
_lv_ll_ins_head
_lv_ll_clear

_lv_obj_get_ext_draw_size
_lv_obj_get_layer_type

_lv_area_is_in
_lv_area_intersect
_lv_inv_area
_lv_area_is_point_on

_lv_obj_scroll_by_raw
```

kisvegabor, 2024-08-23:
Can you open a PR for it? We can merge it at latest on Monday.

PR #6737 (2024-08-26): chore: add LV_USE_PRIVATE_API
Fixes #6631

cc @XuNeo 

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig).
- Run `scripts/code-format.py` (`astyle v3.4.12` need [... cut]
closed 2024-08-26 by kisvegabor