lvgl/lvgl #6633: Let the user change long_press_time easily

avilleret, 2024-08-04:
### Problem to solve

Currently, the only way to change `long_press_time` is to edit `LV_INDEV_DEF_LONG_PRESS_TIME` [here](https://github.com/lvgl/lvgl/blob/master/src/indev/lv_indev.c#L38) (because `lv_indev_t` structure is private) or to rewrite the time logic like @kisvegabor suggested in #5821 

### Success criteria

Be able to change `long_press_time` from code.

### Solution outline

As suggested in #5821, a `lv_obj_set_long_pressed_time` will allow to change `long_press_time` at runtime for each input device.

Another solution might be to move the define to the `lv_conf.h` which then do not increase API size but we can only change it for each device nor during execution (which is fine for me).

### Rabbit holes

None

### Testing

Write a unit test

### Teaching

Update the documentation

### Considerations

I will love it 😊

Jvsjvs, 2024-08-05:
Makes sense, the default "long press time" is also rather short.

liamHowatt, 2024-08-06:
Related #6528

@avilleret would you be comfortable submitting a PR?

avilleret, 2024-08-07:
@liamHowatt sure I can implement a `lv_obj_set_long_pressed_time` method

avilleret, 2024-08-09:
It appears that in 8.4 branch it is pretty straight forward.
I'm doing something like : 

```static lv_indev_drv_t indev_drv_tp;
lv_indev_drv_init(&indev_drv_tp);
indev_drv_tp.type = LV_INDEV_TYPE_POINTER;
indev_drv_tp.read_cb = touchpad_read;
indev_drv_tp.long_press_time = 2000;
indev_touchpad = lv_indev_drv_register(&indev_drv_tp);```

But `lv_indev_drv_t` is now obfuscated in master, so a PR will follow

closed 2024-08-13 by FASTSHIFT