lvgl/lvgl #6640: GESTURE bug

IAMMX, 2024-08-05:
### LVGL version

master

### What happened?

Only a very fast slide will trigger GESTURE, normal sliding cannot trigger GESTURE.

But my test is normal in v8

### How to reproduce?

```c
static void my_event(lv_event_t * e)
{
  lv_obj_t * screen = lv_event_get_current_target(e);
  lv_dir_t dir = lv_indev_get_gesture_dir(lv_indev_active());
  printf("Gesture:\t");
  switch(dir) {
    case LV_DIR_LEFT:
      printf("Left\n");
      break;
    case LV_DIR_RIGHT:
      printf("Right\n");
      break;
    case LV_DIR_TOP:
      printf("Top\n");
      break;
    case LV_DIR_BOTTOM:
      printf("Bottom\n");
      break;
  }
}

void gesture_test(void)
{
  lv_obj_remove_flag(lv_screen_active(), LV_OBJ_FLAG_GESTURE_BUBBLE);
  lv_obj_center(lv_screen_active());
  lv_obj_add_event_cb(lv_screen_active(), my_event, LV_EVENT_GESTURE, NULL);
}
```

becseya, 2024-08-05:
The threshold which determines if a gesture is registered is set by `LV_INDEV_DEF_GESTURE_MIN_VELOCITY`. You should try lowering it.

Since the "velocity" is not actually corrected for time passed, it might also be necessary to increase the update period of the input device (`LV_DEF_REFR_PERIOD` in even polling mode).

IAMMX, 2024-08-06:
@becseya 
1. Decreasing LV_INDEV_DEF_GESTURE_MIN_VELOCITY didn't have a noticeable effect, but modifying LV_INDEV_DEF_GESTURE_LIMIT did achieve the desired result. However, these configurations are not present in lv_conf.h, making them hard to modify.
2. I used the default configurations in lv_conf.h, but the v8 version feels very responsive, while the v9 version requires a much more forceful swipe (for example, when testing a pull-down notification shade on a smartphone).
3.  All tests were conducted using an SDL simulator.

kisvegabor, 2024-08-07:
> Decreasing LV_INDEV_DEF_GESTURE_MIN_VELOCITY didn't have a noticeable effect, but modifying LV_INDEV_DEF_GESTURE_LIMIT did achieve the desired result. However, these configurations are not present in lv_conf.h, making them hard to modify.

We have two open issues to add API for these parameters, so I hope these will be resolved soon: 
- https://github.com/lvgl/lvgl/issues/6633
- https://github.com/lvgl/lvgl/issues/6528

> I used the default configurations in lv_conf.h, but the v8 version feels very responsive, while the v9 version requires a much more forceful swipe (for example, when testing a pull-down notification shade on a smartphone).

I have no immediate idea about what could be difference in v8 and v9. This part havn't changes much as much as remember. :slightly_frowning_face:

IAMMX, 2024-08-07:
> I have no immediate idea about what could be difference in v8 and v9. This part havn't changes much as much as remember. 

Would you mind testing it? I tried it and found that the sliding speed difference between the two is quite significant. (I don't know if it's because of the difference between v8's sdl and v9's sdl)

kisvegabor, 2024-08-17:
I couldn't immerse in it yet, but something just came to my mind. Is there any difference with these modifications?

```diff
diff --git a/src/drivers/sdl/lv_sdl_mouse.c b/src/drivers/sdl/lv_sdl_mouse.c
index 98b4f695f..f00e1030d 100644
--- a/src/drivers/sdl/lv_sdl_mouse.c
+++ b/src/drivers/sdl/lv_sdl_mouse.c
@@ -61,7 +61,7 @@ lv_indev_t * lv_sdl_mouse_create(void)
     lv_indev_set_read_cb(indev, sdl_mouse_read);
     lv_indev_set_driver_data(indev, dsc);
 
-    lv_indev_set_mode(indev, LV_INDEV_MODE_EVENT);
+//    lv_indev_set_mode(indev, LV_INDEV_MODE_EVENT);
     lv_indev_add_event_cb(indev, release_indev_cb, LV_EVENT_DELETE, indev);
 
     return indev;
@@ -198,7 +198,7 @@ void lv_sdl_mouse_handler(SDL_Event * event)
 #endif /*LV_SDL_MOUSEWHEEL_MODE == LV_SDL_MOUSEWHEEL_MODE_CROWN*/
             break;
     }
-    lv_indev_read(indev);
+//    lv_indev_read(indev);
 }
 

```

IAMMX, 2024-08-19:
> I couldn't immerse in it yet, but something just came to my mind. Is there any difference with these modifications?

@kisvegabor It works normally now that it's changed to this.

kisvegabor, 2024-08-21:
Great! 

In this case the problem is a little bit deeper. Now there is no time stamp is used to calculate the and size of the "velocity vectors" of touches. LVGL just sees something like "there was an `x=10, y=20` compared to the last read". It doesn't consider if the last ready was 1ms or 1000ms ago so it doesn't know the real speed either.

We should fix in LVGL at some point.

closed 2024-09-14 by lvgl-bot