lvgl/lvgl #6643: Add a define to lv_conf.h to skip SD Card initialization

Jvsjvs, 2024-08-05:
### Problem to solve

Currently when you define "LV_USE_FS_ARDUINO_SD" as 1 such that you can use the Arduino SD Card library (e.g. to read PNG files to be decoded with LodePNG), LVGL by default (re)initializes the SD Card (see line 28 in the file lv_init.c).

I believe it is more convenient NOT to have LVGL initialize the Arduino SD Card but have the user of LVGL do this himself. This allows for example to already use the SD Card for other purposes when LVGL is not yet initialized.

This can easily be solved in a backwards compatible way such that both models can be used.

### Success criteria

#define LV_FS_ARDUINO_SD_SKIP_SD_INIT 1
Test that the SD Card is indeed not (re)initialized.

### Solution outline

To solve this is in a backwards compatible way
 I would suggest:
1) Add: "#define LV_FS_ARDUINO_SD_SKIP_SD_INIT 0" to the lv_config.h file and
2) Change line 28 of lv_init.c to:
    #if LV_FS_ARDUINO_SD_SKIP_SD_INIT == 0
        fs_init();
    #endif

This way everything works the same as today, but the end user can define: "LV_FS_ARDUINO_SD_SKIP_SD_INIT" as 1 and take care of the initialization outside of LVGL.

The default Arduino SD Card section in lv_conf.h then becomes:
--------------------------------------
/*API for Arduino Sd. */
#define LV_USE_FS_ARDUINO_SD 0
#if LV_USE_FS_ARDUINO_SD
    #define LV_FS_ARDUINO_SD_LETTER '\0'     /*Set an upper cased letter on which the drive will accessible (e.g. 'A')*/
    #define LV_FS_ARDUINO_SD_CS_PIN 0     /*Set the pin connected to the chip select line of the SD card */
    #define LV_FS_ARDUINO_SD_FREQUENCY 40000000     /*Set the frequency used by the chip of the SD CARD */
    **#define LV_FS_ARDUINO_SD_SKIP_SD_INIT 0**
#endif
--------------------------------------

### Rabbit holes

None as it is fully backwards compatible

### Testing

1)
#define LV_FS_ARDUINO_SD_SKIP_SD_INIT 1
Test that the SD Card is indeed not initialized.

2)
#define LV_FS_ARDUINO_SD_SKIP_SD_INIT 0
Test that the SD Card is initialized

3)
#define LV_FS_ARDUINO_SD_SKIP_SD_INIT 1
initialize the SD Card outside LVGL and use LodePNG to decode a PNG file from SD Card by seeting the image source: lv_img_set_src(img, "S:folder1/my_img.png"

### Teaching

1) Add it to the documentation
2) The lv_conf.h file will have a simple self explanatory line

kisvegabor, 2024-08-07:
I've just realized that Arduino SD is added after v9.1, so it's not released yet. 

In light if that I'd simply remove the initialization and the related configs and let it to be done by the user. (We do the same with FatFS)

kisvegabor, 2024-08-22:
@liamHowatt, we should address before the release.

PR #6725 (2024-08-23): fix(fs): remove Arduino SD initialization
Fixes #6643

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig).
- Run `scripts/code-format.py` (`astyle v3.4.12` need [... cut]
closed 2024-08-23 by kisvegabor