lvgl/lvgl #6647: NEON sw/blend assembler generates superfluous symbols

tobiasjakobi-compleo, 2024-08-06:
### LVGL version

v9.1.0-370-g7e3be59dd

### What happened?

Hello,

I'm currently migrating our software stack from LVGL v8.4.0 to v9.1.0. Since our platform is NEON capable (the SoC is a i.MX7D), I thought: Why not enable the new asm code that accelerates blending?!

I should mention that we build LVGL as shared library through the Yocto buildsystem. The shared library goes into the operating system, where it is then used by our application. Turns out that we now run into a linker error when linking application against the shared library. Error is the usual undefined symbols.

I tracked the issue down to the NEON assembler code. I manually assembled the code via `arm-tdx-linux-gnueabi-as -a -am lv_blend_neon.S` and checked the debug output of the assembler. The undefined symbols already pop up here.

The log states this:
```console
UNDEFINED SYMBOLS
color
ld
st
normal
rgb565
rgb888
xrgb8888
argb8888
```

The symbols looked familiar. It turns out that these symbols are all the string constants that are passed through the various macros to control which code is generated.

I haven't figured out why this happens, but it seems to be related to the use of `.altmacro`. I have rewritten the code using the conventional macro syntax here: https://github.com/tobiasjakobi-compleo/lvgl/tree/neon-macro-syntax

This generates the same assembler output, i.e. if you pipe the code through `as` and then do a `objdump -D` again, then the resulting output is identical before and after my modifications.
But now these undefined symbols are gone.

I'm not sure if this is the correct way forward. But at least things work now for me.

With best wishes,
Tobias

tobiasjakobi-compleo, 2024-08-06:
Here's the output from the assembler.
[neon.tar.gz](https://github.com/user-attachments/files/16508338/neon.tar.gz)

kisvegabor, 2024-08-07:
Thank you for reporting this issue. @PeterBee97 could you take a look?

tobiasjakobi-compleo, 2024-08-22:
> We need some feedback on this issue.
> 
> Now we mark this as "stale" because there was no activity here for 14 days.
> 
> Remove the "stale" label or comment else this will be closed in 7 days.

Well, I guess I need to bump this then...

tobiasjakobi-compleo, 2024-08-23:
> Let's continue in #6682

Uhm, that PR has nothing to do with the problem described in this issue.

kisvegabor, 2024-08-23:
Oh, sorry

closed 2024-09-19 by lvgl-bot