lvgl/lvgl #6650: How to use LV_EVENT_DRAW_TASK_ADDED to refactor the text for scale?

lhdjply, 2024-08-07:
### Introduce the problem

cc @C47D @liamHowatt @kisvegabor 

### Introduce the problem

Currently, lv_scale_set_text_src is not very flexible, it can only use a fixed string array. Therefore, I want to use it through LV_EVENT_DRAW_TASK_ADDED. So I made the following modifications to the source code

```diff
diff --git a/src/widgets/scale/lv_scale.c b/src/widgets/scale/lv_scale.c
index b6de17e90..7c1b7c351 100644
--- a/src/widgets/scale/lv_scale.c
+++ b/src/widgets/scale/lv_scale.c
@@ -568,6 +568,9 @@ static void scale_draw_indicator(lv_obj_t * obj, lv_event_t * event)      

         const int32_t tick_value = lv_map(tick_idx, 0U, total_tick_count - 1, scale->range_min, scale->range_max);

+        label_dsc.base.id1 = tick_idx;
+        label_dsc.base.id2 = tick_value; 
+
         /* Overwrite label and tick properties if tick value is within section range */      
         lv_scale_section_t * section;
         LV_LL_READ_BACK(&scale->section_ll, section) {
```

### Test example

```c
static void draw_event_cb(lv_event_t * e)
{
    lv_obj_t * obj = lv_event_get_target(e);
    lv_draw_task_t * draw_task = lv_event_get_draw_task(e);
    lv_draw_dsc_base_t * base_dsc = lv_draw_task_get_draw_dsc(draw_task);
    lv_draw_label_dsc_t * label_draw_dsc = lv_draw_task_get_label_dsc(draw_task);
    if(label_draw_dsc) {
      lv_color_t color_idx[7]={
        lv_palette_main(LV_PALETTE_RED),
        lv_palette_main(LV_PALETTE_ORANGE),
        lv_palette_main(LV_PALETTE_YELLOW),
        lv_palette_main(LV_PALETTE_GREEN),
        lv_palette_main(LV_PALETTE_CYAN),
        lv_palette_main(LV_PALETTE_BLUE),
        lv_palette_main(LV_PALETTE_PURPLE),
      };
      uint8_t major_tick=lv_scale_get_major_tick_every(obj);
      label_draw_dsc->color=color_idx[base_dsc->id1/major_tick];
      lv_snprintf((char*)label_draw_dsc->text,
                  sizeof(label_draw_dsc->text),
                  "%.1f",
                  base_dsc->id2*1.0f);
    }
}

void lv_example_scale_test(void)
{
    lv_obj_t * scale = lv_scale_create(lv_screen_active());
    lv_obj_set_size(scale, lv_pct(80), 100);
    lv_scale_set_mode(scale, LV_SCALE_MODE_HORIZONTAL_BOTTOM);
    lv_obj_center(scale);

    lv_scale_set_label_show(scale, true);

    lv_scale_set_total_tick_count(scale, 31);
    lv_scale_set_major_tick_every(scale, 5);

    lv_obj_set_style_length(scale, 5, LV_PART_ITEMS);
    lv_obj_set_style_length(scale, 10, LV_PART_INDICATOR);
    lv_scale_set_range(scale, 10, 40);

    lv_obj_add_event_cb(scale, draw_event_cb, LV_EVENT_DRAW_TASK_ADDED, NULL);
    lv_obj_add_flag(scale, LV_OBJ_FLAG_SEND_DRAW_TASK_EVENTS);
}
```
### Generated results

[image]

### Discuss

The generated result is not very perfect, the text becomes two lines,I don't sure how to modify it?

C47D, 2024-08-09:
Hi,

I will try to figure out how it can be displayed in a single line, if you try an smaller font, it is drawn also in 2 lines?

lhdjply, 2024-08-10:
Yes, it's still two lines. The reason should be that it is modified like this, and the displayed area is still the width of the original label.

C47D, 2024-08-16:
Hi, I wasn't able to get lvgl compiling but I just did. I'll try to figure out how to help you until Gabor reaches out.

C47D, 2024-08-16:
@lhdjply 
Hi, I was able to do it by increasing the draw task coordinates, in this example it gets increased by 20, I don't know how can we get it right, there's `scale_get_label_coords` but I don't know if it can be used outside of `lv_scale.c`

```c
static void draw_event_cb(lv_event_t * e)
{
    lv_obj_t * obj = lv_event_get_target(e);
    lv_draw_task_t * draw_task = lv_event_get_draw_task(e);
    lv_draw_dsc_base_t * base_dsc = lv_draw_task_get_draw_dsc(draw_task);
    lv_draw_label_dsc_t * label_draw_dsc = lv_draw_task_get_label_dsc(draw_task);
    if(base_dsc->part == LV_PART_INDICATOR) {
        if(label_draw_dsc) {
          const lv_color_t color_idx[7]={
            lv_palette_main(LV_PALETTE_RED),
            lv_palette_main(LV_PALETTE_ORANGE),
            lv_palette_main(LV_PALETTE_YELLOW),
            lv_palette_main(LV_PALETTE_GREEN),
            lv_palette_main(LV_PALETTE_CYAN),
            lv_palette_main(LV_PALETTE_BLUE),
            lv_palette_main(LV_PALETTE_PURPLE),
          };
          uint8_t major_tick=lv_scale_get_major_tick_every(obj);
          label_draw_dsc->color=color_idx[base_dsc->id1/major_tick];
          lv_snprintf((char*)label_draw_dsc->text,
                      sizeof(label_draw_dsc->text),
                      "%.1f",
                      base_dsc->id2*1.0f);
          LV_LOG_USER("Draw task added event added: %s", label_draw_dsc->text);
          draw_task->area.x2 += 20; // <---- HERE
        }
    }
}
```
[image]

lhdjply, 2024-08-17:
Great, thank you for your answer. Do you think it's necessary to write a PR for this feature?

C47D, 2024-08-17:
Hi,
I don't know to be honest, what do you think @kisvegabor ?

kisvegabor, 2024-08-17:
Sorry for joining lately. 

Yes a PR would be very nice to set
```c
label_dsc.base.id1 = tick_idx;
label_dsc.base.id2 = tick_value; 
```

Could you also add an example for this. The same code as you have used above would be perfect. 

FYI, if would be very precise about the area size you can do this
```c
#include "lvgl/src/lvgl_private.h" //To expose the fields of lv_draw_task_t

static void draw_event_cb(lv_event_t * e)
{
    lv_obj_t * obj = lv_event_get_target(e);
    lv_draw_task_t * draw_task = lv_event_get_draw_task(e);
    lv_draw_dsc_base_t * base_dsc = lv_draw_task_get_draw_dsc(draw_task);
    lv_draw_label_dsc_t * label_draw_dsc = lv_draw_task_get_label_dsc(draw_task);
    if(base_dsc->part == LV_PART_INDICATOR) {
        if(label_draw_dsc) {
          const lv_color_t color_idx[7]={
            lv_palette_main(LV_PALETTE_RED),
            lv_palette_main(LV_PALETTE_ORANGE),
            lv_palette_main(LV_PALETTE_YELLOW),
            lv_palette_main(LV_PALETTE_GREEN),
            lv_palette_main(LV_PALETTE_CYAN),
            lv_palette_main(LV_PALETTE_BLUE),
            lv_palette_main(LV_PALETTE_PURPLE),
          };
          uint8_t major_tick=lv_scale_get_major_tick_every(obj);
          label_draw_dsc->color=color_idx[base_dsc->id1/major_tick];
          lv_snprintf((char*)label_draw_dsc->text,
                      sizeof(label_draw_dsc->text),
                      "%.1f",
                      base_dsc->id2*1.0f);
          lv_point_t size;
          lv_text_get_size(&size, label_draw_dsc->text, label_draw_dsc->font, 0, 0, 1000, LV_TEXT_FLAG_NONE);
          int32_t new_w = size.x;
          int32_t old_w = lv_area_get_width(&draw_task->area);

          // Distribute the new size equally on both sides
          draw_task->area.x1 -= (new_w - old_w) / 2;
          draw_task->area.x2 +=  ((new_w - old_w) + 1) / 2; // +1 for rounding

        }
    }
}
```

lhdjply, 2024-08-19:
Continue #6699

closed 2024-08-19 by lhdjply