lvgl/lvgl #6657: Version V9.1, RLE decompression error occurred.

Chenyuanadd, 2024-08-08:
### LVGL version

V9.1.0

### What happened?

/*Decode bin images to RAM*/
#define LV_BIN_DECODER_RAM_LOAD 1
/*RLE decompress library*/
#define LV_USE_RLE 1
lv_image_dsc_t t_main_bg = {
  .header.cf = LV_COLOR_FORMAT_ARGB8888,
  .header.magic = LV_IMAGE_HEADER_MAGIC,
  .header.w = 800u,
  .header.h = 480u,
  .header.flags = LV_IMAGE_FLAGS_COMPRESSED,
  .data_size = 616412,
  .data = (uint8_t *)UI_MAIN_BG,
};

Executing lv_image_set_src(icon, &t_main_bg); results in the image not being displayed.
After troubleshooting, it was found that the issue lies in the static lv_result_t decode_compressed(lv_image_decoder_t * decoder, lv_image_decoder_dsc_t * dsc) and static lv_result_t decode_compressed(lv_image_decoder_t * decoder, lv_image_decoder_dsc_t * dsc) functions within lv_bin_decoder.c. The sections with lines are my modifications. After making these changes, the display works correctly. The problem was caused because the RLE data starts from the 12th bit, not from the 0th bit.
[image]
[image]

Chenyuanadd, 2024-08-08:
I read the code in LVGLImage.py. The .bin file generated using the RLE compression command has the first 24 bits representing some V9 and image information. The next 12 bits indicate whether RLE or LZ4 compression is used, as well as the compressed information. It is possible that the bit positions in the code are incorrect.

kisvegabor, 2024-08-08:
Thank you for opening this issue and investigating the problem.

@XuNeo 
Is this case not covered in the CI?

Chenyuanadd, 2024-08-08:
Thank you for your prompt response. I'm glad to contribute to the project. If there's any additional information or testing you need from my side, please let me know. I'm happy to assist in any way possible to help resolve this issue.

XuNeo, 2024-08-08:
Should already be covered in CI.

https://github.com/lvgl/lvgl/blob/6264265715aa40a659c279ee61ba7d7461731fc1/tests/src/test_cases/draw/test_image_formats.c#L159-L173

> The problem was caused because the RLE data starts from the 12th bit, not from the 0th bit.

I believe you mean 12th byte. `compressed` is the header of `lv_image_dsc_t->data`, and compressed->data is the actual compressed data.

`compressed->data = image->data + len;`

Please try to reproduce your issue on LVGL simulator and upload the test image data if still an issue.

Chenyuanadd, 2024-08-09:
I apologize for the confusion, but I didn't quite understand what you wanted me to test. Currently, the code submitted in the previous image is running correctly.

In the LVGLImage.py file, the first 24 bits of the RLE-compressed data represent:

The first 12 bits are:

Magic Number (0x19): Indicates a specific LVGL version (version 9), used to identify the data format.
Color Format (self.cf.value): Represents the color format, usually defining how the pixel colors are encoded.
Flags (self.flags): A 16-bit flag field used to store various options or states.
Width (self.w): A 16-bit width value, indicating the width of the image in pixels.
Height (self.h): A 16-bit height value, indicating the height of the image in pixels.
Stride (self.stride): A 16-bit stride value, representing the number of bytes per row of image data.
The remaining 12 bits are:

01 indicates RLE compression.
I remember the amount of data before and after compression is as follows:
image->data + 12 is because it needs to copy the position of the RLE information, and then compare the size of the RLE with the size of the image.
compressed_len – 12 is because compressed->compressed_size does not include the first 12 bits of image information.
const uint8_t * input = compressed->data + 12; is because we still need to start from the RLE information.

Chenyuanadd, 2024-08-09:
Yes, sorry, I may not have been very clear. I read the code in LVGLImage.py, and I see that bits 0-11 represent version information and some image data, while bits 12-23 represent compression information.

In the source code:
lv_memcpy(compressed, image->data, len);
compressed->data = image->data + len;
if(compressed->compressed_size != compressed_len) {
    LV_LOG_WARN("Compressed size mismatch: %" LV_PRIu32" != %" LV_PRIu32, compressed->compressed_size, compressed_len);
    return LV_RESULT_INVALID;
}
The way lv_memcpy(compressed, image->data, len); is used actually copies bits 0-11, so the length check is obviously failing. Similarly, the line const uint8_t * input = compressed->data; has an issue with the starting position of the data.

Therefore, after my modification:
lv_memcpy(compressed, image->data + 12, len);
compressed->data = image->data + len;
if(compressed->compressed_size != compressed_len - 12) {
    LV_LOG_WARN("Compressed size mismatch: %" LV_PRIu32" != %" LV_PRIu32, compressed->compressed_size, compressed_len);
    return LV_RESULT_INVALID;
}
const uint8_t * input = compressed->data + 12;

With these changes, it runs normally. I hope this helps you better analyze the issue.

XuNeo, 2024-08-19:
I wasn't able to identify what might be wrong

@Chenyuanadd  Could you please send you RLE image data that has this issue?

Chenyuanadd, 2024-08-20:
@XuNeo Of course! I generated the bin file using LVGLImage.py with the command ./script/LVGLImage.py --ofmt BIN --cf ARGB8888 --compress RLE cogwheel.png. I'll send you the bin file now.

Chenyuanadd, 2024-08-20:
@XuNeo Of course! I generated the bin file using LVGLImage.py with the command ./script/LVGLImage.py --ofmt BIN --cf ARGB8888 --compress RLE cogwheel.png. I just tried uploading the bin file, but it couldn't be uploaded, so I'll send you a screenshot of the first 24 bits of the RLE data instead.
[image]

Chenyuanadd, 2024-08-20:
If you need any additional information or testing, please let me know. I'm happy to help in any way possible to resolve this issue.

closed 2024-08-22 by Chenyuanadd