lvgl/lvgl #6659: Kconfig file format error?

vwheeler63, 2024-08-08:
### LVGL version

v9.1.1

### What happened?

I was reading an article about Kconfig that showed some of the Linux Kernel's `conf` program input parsing source code (which reads the Kconfig file), and I stumbled across that THAT source code I was looking at ONLY wanted to see leading tab characters ('\t') in the input for "child" items.  The code I saw left no room for space characters.  However, I noticed that in LVGL's ./Kconfig file on line 1692 (the line after "config LV_USE_LIBINPUT" beginning with "bool") has 4 leading spaces in it instead of tabs.  While I have not been able to determine if this is a real issue or not with the context in which Kconfig would be used (appears to be including LVGL in a Linux Kernel?), I am reporting this in case it is because it appears to be unintentional.

### How to reproduce?

View ./Kconfig with an editor that can show the difference between leading spaces and tabs, and go to line 1692 (or if that line changed, search for LV_USE_LIBINPUT and look at the line immediate after it beginning with the word "bool").

kisvegabor, 2024-08-17:
Thank you for pointing it out! Seemingly it doesn't cause any issues, but it's be better to fix it for sure. 

LVGL is not integrated into the Linux kernel, but many RTOS-es and framewroks are using Kconfig too (Zephyr, NuttX, RT-Thread, ESP-IDF).

vwheeler63, 2024-08-17:
@kisvegabor   Ah!  Submitted as PR #6695.

closed 2024-09-02 by vwheeler63

kisvegabor, 2024-09-02:
Fixed by https://github.com/lvgl/lvgl/pull/6695