lvgl/lvgl #6660: Roller Crashes If Only Given One String

DovahBrownies, 2024-08-08:
### LVGL version

9.1

### What happened?

The roller widget crashes if only a single string is provided as an option to choose from. Issues doesn't seem to exist in version 8.4.

Suspected culprit code:
https://github.com/lvgl/lvgl/blob/master/src/widgets/roller/lv_roller.c#L510-L515

[image]

mvladic, 2024-08-16:
It doesn't crash anymore. But, something is odd when there is one option compared when there is two or more options.

One option:

[image]

Two options:

[image]

vwheeler63, 2024-08-16:
Good morning, @mvladic 

Nice screen capture to illustrate!  I believe you will find that behavior goes away if you remove the SCROLLABLE flag from the parent object like this:

```
lv_obj_remove_flag(ptr_to_parent_obj, LV_OBJ_FLAG_SCROLLABLE);
```

mvladic, 2024-08-16:
Hm, it didn't help, same result.

vwheeler63, 2024-08-16:
@kisvegabor  Gabor?  What did we miss?

kisvegabor, 2024-08-17:
The roller is handled in the `LV_EVENT_PRESSING` so disabling scrolling won't help. 

What about ignoring event for 0 and 1 options like this?
```diff
diff --git a/src/widgets/roller/lv_roller.c b/src/widgets/roller/lv_roller.c
index 4c7893ed6..0eae6a351 100644
--- a/src/widgets/roller/lv_roller.c
+++ b/src/widgets/roller/lv_roller.c
@@ -349,10 +349,14 @@ static void lv_roller_event(const lv_obj_class_t * class_p, lv_event_t * e)
         refr_position(obj, LV_ANIM_OFF);
     }
     else if(code == LV_EVENT_PRESSED) {
+       if(roller->option_cnt <= 1) return;
+
         roller->moved = 0;
         lv_anim_delete(get_label(obj), set_y_anim);
     }
     else if(code == LV_EVENT_PRESSING) {
+       if(roller->option_cnt <= 1) return;
+
         lv_indev_t * indev = lv_indev_active();
         lv_point_t p;
         lv_indev_get_vect(indev, &p);
@@ -364,6 +368,8 @@ static void lv_roller_event(const lv_obj_class_t * class_p, lv_event_t * e)
         }
     }
     else if(code == LV_EVENT_RELEASED || code == LV_EVENT_PRESS_LOST) {
+       if(roller->option_cnt <= 1) return;
+
         release_handler(obj);
     }
     else if(code == LV_EVENT_FOCUSED) {
@@ -399,6 +405,8 @@ static void lv_roller_event(const lv_obj_class_t * class_p, lv_event_t * e)
         }
     }
     else if(code == LV_EVENT_KEY) {
+       if(roller->option_cnt <= 1) return;
+
         uint32_t c = lv_event_get_key(e);
         if(c == LV_KEY_RIGHT || c == LV_KEY_DOWN) {
             if(roller->sel_opt_id + 1 < roller->option_cnt) {
@@ -416,6 +424,8 @@ static void lv_roller_event(const lv_obj_class_t * class_p, lv_event_t * e)
         }
     }
     else if(code == LV_EVENT_ROTARY) {
+       if(roller->option_cnt <= 1) return;
+
         int32_t r = lv_event_get_rotary_diff(e);
         int32_t new_id = roller->sel_opt_id + r;
         new_id = LV_CLAMP(0, new_id, (int32_t)roller->option_cnt - 1);

```

vwheeler63, 2024-08-17:
Interestingly, this might also apply to one of my LVGL projects here....  :-)

liamHowatt, 2024-08-19:
> What about ignoring the events for 0 and 1 option count, like this?

LGTM.

Another possibility is letting the 1 option move. In that case, the fix would be to unfreeze the label in the selected area.

kisvegabor, 2024-08-21:
I think `return`ing is simpler and we can easily improve it later if needed :slightly_smiling_face:

PR #6717 (2024-08-23): fix(roller): do not move when there is only one option
### Description of the feature or fix

Fixes #6660

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig).
- [... cut]