lvgl/lvgl #6663: LV_MEM_SIZE Optimum Value

fatihozen3, 2024-08-09:
### Introduce the problem

Hi I use IMXRT 1176 MCU and LVGL v9.1.0 .

I have some screens and screen's have lots of objects. When I add big sized object to screen I get below error according to LV_MEM_SIZE and object's size.

[image]

I can pass this error by increasing LV_MEM_SIZE but is this the optimum solution ? Will I increase mem size when I add new object to the screen every time ? What is the optimum value according to my system ? Is there any upper limit for this mem size ?

jbd1986, 2024-08-13:
The default size is QUITE low, but your target device plays a big role here.  We have 1GB of RAM on our system, so I've set LV_MEM_SIZE to 64MB (64 * 1024 * 1024).  I have no issues with that large of a max size.

kisvegabor, 2024-08-24:
The default value is enough to run `lv_demo_widgets` which is considered a simple real-life-like UI.

You can set `LV_MEM_SIZE` to a larger value and enable `LV_USE_MEM_MONITOR`. This way you will see the current and max memory usage too.

closed 2024-09-18 by lvgl-bot