lvgl/lvgl #6670: no documentation non how to use the radial or conical gradients

kdschlosser, 2024-08-12:
### LVGL version

master branch

### What happened?

Read the title.

There is also an issue with the use of macros instead of an enumeration. This is going to make it unavailable to higher level language bindings.

```c
#define LV_GRAD_LEFT    LV_PCT(0)
#define LV_GRAD_RIGHT   LV_PCT(100)
#define LV_GRAD_TOP     LV_PCT(0)
#define LV_GRAD_BOTTOM  LV_PCT(100)
#define LV_GRAD_CENTER  LV_PCT(50)
```

I am also not sure why `lv_grad_dsc_t`, `lv_gradient_stop_t`, `lv_grad_extend_t` and `lv_grad_dir_t` are defined in `lv_style.h` and not in the headers where the gradient functions are declared.

kdschlosser, 2024-08-12:
There is also an enormous amount of code for rendering the gradients. It has got to be incredibly slow to do.

kisvegabor, 2024-08-12:
Some docs are really missing for ne features. I've just opened this issue to track it #6671

> I am also not sure why lv_grad_dsc_t, lv_gradient_stop_t, lv_grad_extend_t and lv_grad_dir_t are defined in lv_style.h and not in the headers where the gradient functions are declared.

Because these are generic, used by software rendering, GPUs, and many others.

kisvegabor, 2024-08-12:
Let's continue in #6671

closed 2024-08-12 by kisvegabor