lvgl/lvgl #6671: Add missing documentation

kisvegabor, 2024-08-12:
### Introduce the problem

Some new features are not documented properly.

- Gradients
- Vector fonts
- Vector API
- ThorVG
- Caching
- SW rendering
- Chip vendors
  - ST
  - NXP
  - ESP
- RTOS
  - FreeRTOS

vwheeler63, 2024-08-12:
@kisvegabor

Hi, Gabor!  I can help with this.  Esp. the FreeRTOS part.  There is a lot more in the list as well.  We can use at least part of the (upcoming) Doxygen "is not documented" baseline list to help.

As an aside, I also note that the documentation about threading is incomplete.  I have taken some notes about this, as I had to deal with it extensively in a project that was handed over to me in early 2022 that had LVGL calls being made from 6 different threads!  I think I can do a good job of supplementing that documentation.  I think there will be a lot more too.  Let me know your thoughts.

kisvegabor, 2024-08-13:
I've just noticed that @tore-espressif already sent a PR to update the ESP docs: #6574

vwheeler63, 2024-08-14:
Hi, Gabor!

I'm sorry, I created a misunderstanding:  I should be more careful about abbreviating.  By "Esp." I meant "especially" -- especially FreeRTOS.

I do keep hearing however about the ESP32 -- it must be a popular microcontroller.

clydebarrow, 2024-08-15:
Another undocumented (old) feature - color filters. And colour gradients.

kisvegabor, 2024-08-17:
@vwheeler63 
FreeRTOS only is awesome too :blush:

Microchip is not an official partner of LVGL yet and we don't have in house expertise for PICs now. It might be a good collaboration/work opportunity in the future! 

@clydebarrow 
 > Another undocumented (old) feature - color filters. And color gradients.

Added to the list, thank you!

vwheeler63, 2024-08-17:
> It might be a good collaboration/work opportunity in the future!

I like the sound of that.  :-)

lorenzo-arena, 2024-09-20:
About "Color filter" and "Gradients": what would be a good place for them? I see that there's already a section with a "TODO" for Color filter, would it be good to add gradients right after that?

vwheeler63, 2024-09-25:
> About "Color filter" and "Gradients": what would be a good place for them? I see that there's already a section with a "TODO" for Color filter, would it be good to add gradients right after that?

Seems logical.

apavelm, 2025-08-31:
Hi Gabor!

Let me add a documentation request here.

Here: https://docs.lvgl.io/9.3/details/integration/driver/display/gen_mipi.html
And as well as for all others display controller drivers, there is a mentioning "all you need is just to implement the following functions..."

```
/* Send short command to the LCD. This function shall wait until the transaction finishes. */
int32_t my_lcd_send_cmd(lv_display_t *disp, const uint8_t *cmd, size_t cmd_size, const uint8_t *param, size_t param_size)
{
        ...
}

/* Send large array of pixel data to the LCD. If necessary, this function has to do the byte-swapping. This function can do the transfer in the background. */
int32_t my_lcd_send_color(lv_display_t *disp, const uint8_t *cmd, size_t cmd_size, uint8_t *param, size_t param_size)
{
        ...
}
```

But without any details or samples, it's difficult to guess what exactly should be there. Considering ESP and LVGL version related changes in approaches, this is not clear at all.

apavelm, 2025-09-16:
Its difficult to get on what level it should be done: by using SPI direct command to display or by using sort of abstraction especially for that? I tried to find any simple implementation over the internet based on this, but didn't find anything. 
Everything what I found was based on previous version (flush/flush_cb). So, still no clue which way is better. On one hand - I want to stick with the latest version, but on the other hand I have no information why this should be better (may be it is faster or something?). There is no information about pros and cons on this as well.