lvgl/lvgl #6677: When deleting the object, the ESP32 crashes.

Donocean, 2024-08-13:
### LVGL version

v8.3.11

### What happened?

I'm tasked with dynamically adding all the content to a page upon entering it and deleting all objects when exiting. However, the ESP32 crashes when I delete the root object that contains all the children.
This doesn't always happen; sometimes it does, and sometimes it doesn't. I've addressed all possible issues that could cause this, but the ESP32 still occasionally crashes, and I don't know why.

The crash information from esp32 is as follows:
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4201456e  PS      : 0x00060630  A0      : 0x82017910  A1      : 0x3fce32c0  
0x4201456e: _lv_event_mark_deleted at /Users/don/lcd_offical_lib_test/managed_components/lvgl__lvgl/src/core/lv_event
.c:156

A2      : 0x3fce6a34  A3      : 0xffffc000  A4      : 0x00060420  A5      : 0x3fce6a34  
A6      : 0x00000021  A7      : 0x00000001  A8      : 0x00100000  A9      : 0x00060e20  
A10     : 0x3fce6ab4  A11     : 0x3fce3300  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x3fcc0ba0  SAR     : 0x00000015  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00100004  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff  
0x400556d5: strlen in ROM
0x400556e5: strlen in ROM

Backtrace: 0x4201456b:0x3fce32c0 0x4201790d:0x3fce32e0 0x4209dcaf:0x3fce3300 0x4209dcbd:0x3fce3320 0x4201c834:0x3fce3340 0x4201c7c9:0x3fce3360 0x4201c7c9:0x3fce3380 0x4201c900:0x3fce33a0 0x4201c929:0x3fce33c0 0x42022cc8:0x3fce33e0 0x42023f29:0x3fce3400 0x42023fd6:0x3fce3420 0x42009de8:0x3fce3440 0x40383099:0x3fce3460

0x4201456b: _lv_event_mark_deleted at /lvgl__lvgl/src/core/lv_event.c:155
0x4201790d: lv_obj_destructor at /lvgl__lvgl/src/core/lv_obj.c:454
0x4209dcaf: _lv_obj_destruct at lvgl__lvgl/src/core/lv_obj_class.c:136 (discriminator 1)
0x4209dcbd: _lv_obj_destruct at /lvgl__lvgl/src/core/lv_obj_class.c:143
0x4201c834: obj_del_core at /lvgl__lvgl/src/core/lv_obj_tree.c:385
0x4201c7c9: obj_del_core at /lvgl__lvgl/src/core/lv_obj_tree.c:362
0x4201c7c9: obj_del_core at /lvgl__lvgl/src/core/lv_obj_tree.c:362
0x4201c900: lv_obj_del at /lvgl__lvgl/src/core/lv_obj_tree.c:61
0x4201c929: lv_obj_del_async_cb at /lvgl__lvgl/src/core/lv_obj_tree.c:348
0x42022cc8: lv_async_timer_cb at /lvgl__lvgl/src/misc/lv_async.c:103
0x42023f29: lv_timer_exec at /lvgl__lvgl/src/misc/lv_timer.c:313 (discriminator 2)
0x42023fd6: lv_timer_handler at /lvgl__lvgl/src/misc/lv_timer.c:109
```

The code finally crashed in this function.
```c
void _lv_event_mark_deleted(lv_obj_t * obj)
{
    lv_event_t * e = event_head;

    while(e) {
        if(e->current_target == obj || e->target == obj) e->deleted = 1;
        e = e->prev;
    }
}

```

xh2000, 2024-08-16:
Same problem which troubles me a lot recently, looking forward to some helps.

vwheeler63, 2024-08-16:
Good morning!  I am not the Guru that Gabor is, but I thought I could help because I once had to solve similar issues.  One of the ways this can happen is if _any_ `lv_...()` calls are being made in another thread.  For instance if your delete call is made in an ISR context and rendering (happens by calls to `lv_timer_handler()` indirectly (or if any form of `lv_refr...()` is being called directly) is happening in in another thread, or vice versa.  Is it running under any kind of OS?

lewisxhe, 2024-08-17:
Same problem, but it happened on lvgl 9

Donocean, 2024-08-17:
> Good morning! @xh2000 @Donocean
> 
> I am not the Guru that Gabor is, but I thought I could help because I once had to solve similar issues. One of the ways this can happen is if _any_ `lv_...()` calls are being made in more than one thread. For instance if your delete call is made in an ISR context and rendering (happens indirectly by calls to `lv_timer_handler()`, or if any form of `lv_refr...()` is being called directly) is happening in in another thread, or vice versa. Is it running under any kind of OS?

The ESP32 is running on FreeRTOS, which is a real-time OS for embedded devices. I'm sure there is only one thread handling the LVGL tasks, and all the `lv_...()` calls are made by input device events, such as keypad events, through calls to `lv_timer_handler()`.

kisvegabor, 2024-08-17:
If it's not a threading issue:
1. Be sure that there are no warnings
2. Try deleting widgets with a very simple example
3. Try to simplify your code and see when it gets broken
4. Check for array out of bound errors
5. Be sure that `lv_style`s are static or global

Yamauchi-Kaz, 2024-08-28:
I am also encountering the same issue (crash in _lv_event_mark_deleted), which is causing problems for me. 
LVGL version: v8.3.11 
Environment: ESP32S3 

The crash occasionally occurs when deleting the old screen during a screen transition.

I tried extracting the GUI processing part from the ESP32S3 and testing it in Visual Studio 2022(Windows), but the same issue occurred.

Donocean, 2024-08-28:
> I am also encountering the same issue (crash in _lv_event_mark_deleted), which is causing problems for me. LVGL version: v8.3.11 Environment: ESP32S3
>
> The crash occasionally occurs when deleting the [...]

Could you paste your code here so we can figure out what's actually going on with the code or LVGL?

Yamauchi-Kaz, 2024-08-30:
I was able to resolve the issue by following the guidance on the following site: 
Reference site: https://forum.lvgl.io/t/lv-event-mark-deleted-crashes-sometimes/10323 

The issue was caused by running 'lv_task_handler()' in a different task from the one handling screen rendering (calling the lv_xxx functions).

closed 2024-09-03 by Donocean