lvgl/lvgl #6678: Binary Images don't get correct 'stride' automatically - lv_image_set_src

jbd1986, 2024-08-13:
### LVGL version

LVGL 9+

### What happened?

I'm loading up a uint8_t buffer with an XRGB8888 formatted image's data (in my case, I'm DMA'ing over a 320x240 image).
In the image descriptor, I provide the color format, width, and height.

After calling lv_image_set_src(obj, dsc), I will get a heavily staircased, incorrect version of my image.

If I set the stride in the image descriptor manually to (width * 4), then my image will get displayed correctly.  This was not an issue in LVGL 8.3, so I suspect a bug has been introduced, or a regression occurred.

kisvegabor, 2024-08-29:
> I'm loading up a uint8_t buffer with an XRGB8888 formatted image's data 

How do you create your this custom image? This is the cleanest way of doing this in v9:
```c
  static lv_draw_buf_t draw_buf;
  lv_draw_buf_init(&draw_buf, 320, 240, LV_COLOR_FORMAT_XRGB8888, 320*4, buf, buf_size);
  
  static lv_image_dsc_t img_dsc;
  lv_draw_buf_to_image(&draw_buf, &img_dsc);
  
  lv_image_set_src(img, &img_dsc);
```

jbd1986, 2024-08-29:
> ```c
> lv_draw_buf_init
> ```

That makes sense that it would work for you as you wrote it, because you are manually setting stride in your example above (320*4).

To my understanding, if stride was set to 0, it used to be calculated automatically via some function that would essentially do:

stride = width * (bpp / 8)

closed 2024-09-28 by lvgl-bot