lvgl/lvgl #6679: The Window size (client area) is incorrected when using Windows display driver.

sliang951753, 2024-08-14:
### LVGL version

v9.0.0

### What happened?

When using lv_windows_create_display() to create LVGL display, the Window's client area always smaller than the specified screen resolution. 
for example, lv_windows_create_display(L"LVGL Window", 1024, 600, 100, true, false); 
then, the created window is at the resolution of 1024*600, however, its client area is smaller and about to 1000 * 550. 

This is because the specified display resolution does not consider the window caption and client edge size. 

Here's a possible fix for this issue.
In lv_windows_display_thread_entrypoint() function, calling AdjustWindowRectEx before CreateWindowExW. 

`    RECT rc = {0, 0, data->hor_res, data->ver_res};
    AdjustWindowRectEx(&rc, window_style, FALSE, WS_EX_CLIENTEDGE);

    HWND window_handle = CreateWindowExW(
                             WS_EX_CLIENTEDGE,
                             L"LVGL.Window",
                             data->title,
                             window_style,
                             CW_USEDEFAULT,
                             0,
                             rc.right - rc.left,
                             rc.bottom - rc.top,
                             NULL,
                             NULL,
                             NULL,
                             data);`

kisvegabor, 2024-08-29:
Thank you for reporting it! 

@MouriNaruto could you take look at this issue?

vwheeler63, 2024-08-29:
> HWND window_handle = CreateWindowExW(
>                          WS_EX_CLIENTEDGE,
>                          L"LVGL.Window",
>                          data->title, [...]

I believe there are also 2 one-off errors in this call.  Specifically:

width should = rc.right - rc.left + 1 + (2 * edge_width);
height should = rc.bottom - rc.top + 1 + (2 * edge_width) + title_bar_height

I am a little rusty on `CreateWindowExW()`, but what I remember about Win32 is that (using Win32 data):

width = right - left
height = bottom - top

and I am 100% certain that:

right = 1 pixel PAST right edge, and
bottom = 1 pixel PAST bottom edge.

Kind regards,
Vic

MouriNaruto, 2024-08-29:
I will try to fix that.

Kenji Mouri

kisvegabor, 2024-09-16:
@MouriNaruto did you have time to look into this?

MouriNaruto, 2024-09-16:
I will look into this issue in the recent days.

Kenji Mouri

MouriNaruto, 2024-10-03:
not stale

kissa96, 2024-10-21:
How are we doing with this issue?

MouriNaruto, 2024-10-21:
[image]

Actually, it should not have the issue when people use the simulator mode in the current stage

Note: My screen is 125% DPI scaling, so I set the zoom_level to 80% for rendering the content without scaling (1.25 * 0.8 = 1). It shows the client area is the same as which I set for the lv_windows_create_display.

Kenji Mouri